blackmagic-debug/blackmagic #295: Unable to erase flash on STM32F303CCT6

hydra, 2017-10-04:
Hi again!

I was trying to test #293 on an STM32F303CCT6 cpu (tried several boards) but always get the error message as follows:

```
$ arm-none-eabi-gdb -ex "target extended-remote COM9" -ex "monitor swdp_scan" -ex "attach 1" -ex "load obj/main/cleanflight_SPRACINGF3EVO.elf"
GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 7.12.1.20170417-git
...
Remote debugging using COM9
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32F3
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x20000024 in ?? ()
Error erasing flash with vFlashErase packet
(gdb)
```

I also tried using `monitor erase_mass` before the `load` command as follows:

```
arm-none-eabi-gdb -ex "target extended-remote COM9" -ex "monitor swdp_scan" -ex "attach 1" -ex "monitor erase_mass" -ex "load obj/main/cleanflight_SPRACINGF3EVO.elf"
```
However, that doesn't help and the output is the same as above.

If I erase the flash on the target using j-flash, disconnect j-flash and then connect a blackmagicprobe (without cycling the power on the target) and then try again I get this:

```
$ arm-none-eabi-gdb -ex "target extended-remote COM9" -ex "monitor swdp_scan" -ex "attach 1" -ex "load obj/main/cleanflight_SPRACINGF3EVO.elf"
GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 7.12.1.20170417-git
...
Remote debugging using COM9
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32F3
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xfffffffe in ?? ()

Loading section .isr_vector, size 0x190 lma 0x8000000
Loading section .text, size 0x3c540 lma 0x8000190
Loading section .ARM, size 0x8 lma 0x803c6d0
Loading section .pg_registry, size 0x460 lma 0x803c6d8
Loading section .pg_resetdata, size 0xed lma 0x803cb38
Loading section .data, size 0x1ec0 lma 0x803cc25
Error writing data to flash
(gdb) 
```

Note isn't a new bug related to #293 - I just found it while testing the boards I have.  I recall having the same issue on the F303 when I first tried BlackMagic back in March 2017 recent activity has prompted me to re-test and report this issue, at the time I was working on F4 and F7 boards so it wasn't an issue for me, however I need a single debugging solution that works with F1, F3, F4 and F7 on OSX, Windows and Linux that I can recommend to Cleanflight developers (the project I work on) so this has a bit more priority for me now.

UweBonnes, 2017-10-04:
Your linker files now place on an odd byte address. Again the question to Gareth: Should we support random alignments?

hydra, 2017-10-04:
It's a fair question, I would say yes because:
a) other debuggers do, i have no issue with flashing via OpenOCD for instance.
b) the rest of the gdb server seems to be fine.  i.e. if i upload using another method and then attach to the target I can debug just fine.

UweBonnes, 2017-10-04:
And if you feel like this should be implemented, who should do it?

hydra, 2017-10-04:
whoever has time and inclination I guess.  I might not be grasping what is required to do this, if its even needed for my use-case nor the impact on the codebase.

hydra, 2017-10-05:
in the mean time i'll take a look at the alignment in the linker files, it would be good if blackmagic could eventually support this.

UweBonnes, 2017-10-05:
Please provide the failing elf file here.

hydra, 2017-10-05:
[cleanflight_SPRACINGF3EVO.zip](https://github.com/blacksphere/blackmagic/files/1360237/cleanflight_SPRACINGF3EVO.zip)

attached.

gsmcmullin, 2017-10-09:
Is this fixed by #293 or #297 ? (both are already merged in master)

We should support arbitrary alignments where it is possible.

UweBonnes, 2017-10-09:
Should be fixed by #299

closed 2020-12-09 by UweBonnes