blackmagic-debug/blackmagic #307: STM32L0 model not detected when connecting under reset

schodet, 2017-11-23:
I have a STM32L072 doing WFI. I need to connect under reset. When I do it the microcontroller model is not detected (only ARM Cortex-M). I need to disable reset and connect again:

```
(gdb) mon connect_srst enable
(gdb) mon swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      ARM Cortex-M
(gdb) attach 1
Attaching to program: test.elf, Remote target
warning: while parsing target memory map (at line 1): Required element <memory> is missing
0x0801ddcc in ?? ()
(gdb) mon connect_srst disable
(gdb) mon swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32L0x
(gdb) attach 1
Attaching to program: test.elf, Remote target
0x0801ddcc in ?? ()
(gdb)
```

UweBonnes, 2018-04-26:
This should be fixed with the recent commit of #334. Please retest and report.

gsmcmullin, 2018-04-26:
@schodet Please reopen the issue if you still have the problem.

gsmcmullin, 2018-04-27:
Ok, this is an STM32L0 for which we didn't split the probe/attach, so no surprises it didn't get fixed.

The fact that we can't even read `DBGMCU_IDCODE` when in reset is a concern. That suggests that there is no way to identify this device while it's in reset.

UweBonnes, 2018-04-29:
Trying to reproduce on the L053-DISCO with master. It is even worse!
The running  program hangs mostly in WFI but awakes for a 1 kHz timer tick.
There is no way to connect.!
Neither with "mon connect_srst enable" nor by pressing the reset buttom. I must plug a jumper between boot0 and VCC  and reset to enter the bootloader.  Now I can connect.

This is different from Nicolas experience, as for him at least the Cortex-M core appears.

I can reproduce Nicolas behaviour with eth STM32L073RZ.

schodet, 2018-04-30:
The chip is using stop mode and it wakes up every second. I do not have a reset button on this board.

UweBonnes, 2018-04-30:
In your situation , where scan detects the Cortex-M, we probably should add an option inside the Cortex-M to reprobe. So you would scan first,  attach to the cortex and do some kind of rescan. While halted and attached to the Cortex-M core, I think the STM32L07 will be responsive. 

How to attach to the L05 I don't have any idea yet.

UweBonnes, 2018-04-30:
Further investigation: Stm32CubeProgrammer can attach to both devices without reset line connected. Spying on the devices with saleae shows that a lot of things work similar when attach fails due to a programm using WFI or when attach succeeds, du to starting the system bootloader. xxx_good.csv is a spy on BMP sanning the device with bootloader running and xxx_bad.csv is canning with a user program running, using WFI but also 1000 wakeups/s.
[l053_bad.csv.gz](https://github.com/blacksphere/blackmagic/files/1962309/l053_bad.csv.gz)
[l053_good.csv.gz](https://github.com/blacksphere/blackmagic/files/1962310/l053_good.csv.gz)
[l073_bad.csv.gz](https://github.com/blacksphere/blackmagic/files/1962311/l073_bad.csv.gz)
[l073_good.csv.gz](https://github.com/blacksphere/blackmagic/files/1962312/l073_good.csv.gz)

UweBonnes, 2018-04-30:
sanning, canning -> scanning ("mon s")
du -> due

UweBonnes, 2018-05-01:
With appended hack, I can connect  BMP without the SRST line to STM32L0 devices mostly in WFI.  The hack does:
- after failing ROM Table read, use DP->idcode to call the Cortex-M probe function
- halt the target before probe.

Halting the target before probe should be made an option anyways. I am not sure if connect_srst should be kept as a separate option. Assert SRST after probe before attach seems wrong to me anyways. If the devices is unwillingly like in WFI, probe will fail before. Maybe some option "probe_mode none|halted|srst", executed before probe, would be better.

Comments or code welcome!
[hack.log](https://github.com/blacksphere/blackmagic/files/1963488/hack.log)

schodet, 2018-05-01:
By mostly in WFI you mean that you catch it when it wakes up?

UweBonnes, 2018-05-01:
Well, as I told above it is mostly in WFI but wakes up 1000 times per second. But in the SWD traces given above , I see no unnormal delay to the IDCODE request, So I guess that the devices is responsive even in WFI. Only reading the ROM Table fails. Reading the STM32 IDCode later however needs the device halted.

What about trying out the hack yourself?

schodet, 2018-05-01:
I will, but I am not at work today.

schodet, 2018-05-02:
The hack works!

Here is the test:

```
(gdb) mon swdp_scan
Target voltage: 3.4V
Available Targets:
No. Att Driver
 1      STM32L0x
(gdb) attach 1
Attaching to Remote target
0x080057c8 in ?? ()
(gdb) detach
Detaching from program: , Remote target
(gdb) mon swdp_scan
Target voltage: 3.4V
...long.time...
Available Targets:
Available Targets:
No. Att Driver
 1      STM32L0x
 2      STM32L0x
(gdb) attach 1
Attaching to Remote target
0x08000cf6 in ?? ()
(gdb)
```

(I suppose the second scan repetition is "normal" as this is a hack)

UweBonnes, 2018-05-02:
Yes, it is a hack. But it show that the device is rudumentary responsive even in WFI, but our probe needs rework.
..long.time... probably mean you only wake up from time to time and not at a high rate like my 1 kHz?

schodet, 2018-05-02:
Yes, I only wakes up once each second.

tve, 2018-05-16:
:1 here, I have an L082 and the regular BMP build only recognizes "ARM Cortex-M", with the hack I'm a happy camper. Helping to produce a proper fix is beyond my understanding, but I'd be happy to test any fixes.

konsgn, 2018-05-28:
I am having what I think is a related issue with STM32F030x6. On fist scan, debug shows:

 AP   0: IDR=04770021 CFG=00000000 BASE=e00ff003 CSW=03000040
0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space)
-> cortexm_probe
No STM32CortexM3 found looking for m0
t->idcode:00000000
SAM4L: Probe function called
0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace)
0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit)

(I added the debug lines of no stm32cortex m3 and the following t->idcode to see what happens at the relevant scan point)

After attaching to the cortex-m target, running by pressing c and then disconnecting/ reconnecting the cable the mon-swdp command works as expected and It finds the chip with the  t->idcode:00000444

UweBonnes, 2018-05-29:
Could people with the problems reported here please try.

https://github.com/UweBonnes/blackmagic/tree/probe_halted

In short, this is what the patch does:
- If the ROM Table read fails but we have a CortexM identified by the idcode, nonetheless call cortexm-probe().
- Halt the target before probing. E.g.on STM32L0 and F7, we can not read DBGMCU_IDCODE while in WFI
- Try hard to halt a target in WFI. STM32F7 needs the command repeated.
- Release SRST before probing. E.g. reading DBGMCU_IDCODE fails on L0/F7 when in reset.

konsgn, 2018-05-30:
@UweBonnes , A couple of notes for my situation testing your branch:
1) Debugging seems to no longer work for piping to the second tty. I can't see what low level stuff is happening. 
2) For my set up, the detection of the cortexm0 is still happening, however, I am no longer able to attach to the target.

UweBonnes, 2018-05-30:
Can you send me the elf or at least the bin File. I will try an whatever F0 I find.

UweBonnes, 2018-05-30:
Did you miss the "DEBUG=1" make option?

Regarding 2: I have flashed a F0 Discovery (STM32F51R8) with a sample code that uses WFI and waking up at 1000 Hz system tick. Even without the probe_halted  patch, I can connect without  problems. Is the problem perhaps on your side? can you access the device by other means, e.g. other debug probe or bootloader and erase? Can you attach to the erased device? Is your device perhaps read protected?

konsgn, 2018-05-30:
Hmmm, I didn't miss it but perhaps I didn't make clean the environment. Debug is working now with your branch, Sorry about the false alarm.

1) Perhaps I should open a separate issue? I don't think my failure to detect relates to WFI. 
 
2) I am working with a custom layout of the STM32F030x6, Boot0 shorted to gnd. Also yup, Jlink lite works without issue on the chip. It seems that in the Adiv5.c  It is capable of reading the various registers related to whatever DP is. However when it attempts to read using the target_read_mem32 it gets back all 0's. 

PS: A note on my earlier setup: with the mainline code, only initial swdp_scans fail. Following an attach to the cortex target and continue, subsequent scans all reliably locate the chip and allow me to work with it. I think the issue is somehow reset related.

UweBonnes, 2018-05-31:
Probably you used "mon connect_srst enable". I think this option has a flawed logic. SRST gets asserted as soon as the user enables it. Probe happens_under_ reset, and for most if not all devices only the cortex core debug registers are accessible, but not the device specific registers indicating the device. Only with attach, SRST is de-asserted.

konsgn, 2018-05-31:
Haha, Great insight, my rst line seems to be pulled low regardless of connect-srst commands or hard-srst attempts.

I will poke around at that.

Thank you!

konsgn, 2018-05-31:
Gah! the issue was with line : 
	gpio_set_mode(SRST_PORT, GPIO_MODE_OUTPUT_50_MHZ,
			(((platform_hwversion() == 0) ||
			  (platform_hwversion() >= 3))
			 ? GPIO_CNF_OUTPUT_PUSHPULL
			 : GPIO_CNF_OUTPUT_OPENDRAIN),
			SRST_PIN);

My diy version of the mini v2.0 used a buffer that doesn't understand the float state of the reset line hence it held reset output low always.

I fixed it for myself by setting  (platform_hwversion() >= 3)) to  (platform_hwversion() >= 2)), which i think should be pulled upstream since the v2 mini has a translator as well, but since i'm not working on original hardware I will abstain from that pull request. 

This sure explains a lot of what I was seeing. 

Thank you!

schodet, 2018-06-08:
Hello,

I just tested your `probe_halted`, it works really well!

With a target waking up 1000 times per second it connect and detect the L0 without problem.

With a target waking up every second, it takes some times and finally connects (with some display problems):

```
(gdb) mon swdp_scan
Target voltage: 3.4V
Ignoring packet error, continuing...
Available Targets:
Available Targets:
Available Targets:
Available Targets:
No. Att Driver
 1      STM32L0x
(gdb)
```

Now the same target with `connect_srst`, it connects immediately without any problem.

Thanks!

UweBonnes, 2018-06-22:
20180622: Git head now to halt the debice before probe and also allows to release SRST before probe. This should fix the problem. Pleas test and close the Issue.

schodet, 2018-06-25:
This works!

ofauchon, 2020-09-20:
Hi, 

I have the same problem when I try to flash a board with STM32L072CZT6
My bluepill is connected to the board with :  VCC, GND , SWDIO, SWDCK, and 
I have to press the hardware reset  button while trying to flash ... but it fails with : 

arm-none-eabi-gdb -nx --batch -ex "target extended-remote /dev/ttyACM1" -ex "monitor swdp_scan" -ex "attach 1" -ex load -ex compare-sections -ex kill main.elf 
Target voltage: 2.74V
Available Targets:
No. Att Driver
 1      ARM Cortex-M M0+
warning: while parsing target memory map (at line 1): Required element <memory> is missing
0x0800048a in (error).Error ()
Loading section .text, size 0x8648 lma 0x8000000
Loading section .tinygo_stacksizes, size 0x10 lma 0x8008648
Loading section .data, size 0x7a4 lma 0x8008658
Start address 0x08002ff8, load size 36348
Transfer rate: 124 KB/sec, 956 bytes/write.
Section .text, range 0x8000000 -- 0x8008648: MIS-MATCHED!
Section .tinygo_stacksizes, range 0x8008648 -- 0x8008658: MIS-MATCHED!
Section .data, range 0x8008658 -- 0x8008dfc: MIS-MATCHED!
warning: One or more sections of the target image does not match
the loaded file

Any ideas ? 

Thanks

UweBonnes, 2020-09-20:
Try my cortexm0romtable branch. Scheduled after the pending 1.6.2 release.

UweBonnes, 2021-03-26:
Fixes with #846

closed 2021-03-26 by UweBonnes