blackmagic-debug/blackmagic #308: Black magic probe stuck on bootloader mode

deangelign, 2017-11-27:
Hi everyone, I bought a black magic probe a couple weeks ago. Last week I tried to upgrade the firmware using  "bmp_upgrade-20130819.exe" download from http://www.blacksphere.co.nz/downloads/bmp_driver_20130819.zip. 
While bmp_upgrade-20130819 was running, it crashed, and gave me the following message. 

https://github.com/deangelign/deangelign.github.io/blob/master/black_magic_probe_firmware_upgrade_error.png

Now my black magic probe is stuck on bootloader mode (leds are flashing sequentially). I am a windows 7 user and my board is a Black Magic Probe V2.1. Can anyone help me?

0xeau, 2017-11-27:
Can you try to flash the firmware on Linux box? Ubuntu 16.04 lts would be good choice.

UweBonnes, 2017-11-28:
As the bootlaoder is still available and also alive, you have many options:
- dfu-utils
- scripts/stm32_mem.py using python/usb
- fold the bmp firmware in a dfu file and use stm dfuse

However do not use things ages old, as something called xxx_20130819_xxx

deangelign, 2017-11-28:
Hi, I tried to flash using ubuntu (14.04 instead of 16.04), but unfortunately the error still there. When I type `dfu-util -l` to list the capable devices, I get the following message

```
$ dfu-util -l  
dfu-util 0.9 

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Cannot open DFU device 1d50:6017 
```

when I try to flash I get

```
$ sudo dfu-util -d 1d50:6018,:6017 -s 0x08002000:leave -D blackmagic-native-1.6.1.bin 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1d50:6017
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08002000, size = 74396
Download	[                         ]   0%            0 bytesstate(2) = dfuIDLE, status(0) = No error condition is present
dfu-util: Wrong state after command "ERASE_PAGE" download
```

So...have I bricked my board? Any ideas/suggestion?

I do not know if this can help, but I'm leaving you some information about linux distribution. 

```
$ uname -a
Linux DSB247 3.16.0-77-generic #99~14.04.1-Ubuntu SMP Tue Jun 28 19:17:10 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty
```

UweBonnes, 2017-11-28:
Rectify your permissions on USB devices

deangelign, 2017-11-28:
As root user the console outputs are:

```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [1d50:6017] ver=0100, devnum=3, cfg=1, intf=0, path="1-1.1", alt=0, name="@Internal Flash   /0x08000000/8*001Ka,120*001Kg", serial="7BB297AA"
```

and

```
dfu-util sudo dfu-util -d 1d50:6018,:6017 -s 0x08002000:leave -D blackmagic-native-1.6.1.bin 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1d50:6017
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08002000, size = 74396
Download	[                         ]   0%            0 bytesstate(2) = dfuIDLE, status(0) = No error condition is present
dfu-util: Wrong state after command "ERASE_PAGE" download
```

I don't know why, but for some reason I can read the firmware from device

```
dfu-util -U test.bin 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 1d50:6017
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
Limiting default upload to 16384 bytes
Upload	[=========================] 100%        16384 bytes
Upload done.
```

UweBonnes, 2017-11-28:
"I don't know why, but for some reason I can read the firmware from device" 

Only recent bootloader can read back. Recent bootloaders can also be replaced by another bootloader. Older bootloaders fail at least somewhere?

deangelign, 2017-12-01:
Hi, I'm sorry for the the delay. I didn't get it what you mean by "Older bootloaders". are you talking about the dfu-util version? if you are...I've tried the same steps for a different dfu-util version (dfu-util 0.5) and for a different version of firmware (blackmagic-native-v1.6.bin), but no progress reached.

```
$ dfu-util -d 1d50:6017 -s 0x08002000 -D blackmagic-native-v1.6.bin 
dfu-util 0.5

(C) 2005-2008 by Weston Schmidt, Harald Welte and OpenMoko Inc.
(C) 2010-2011 Tormod Volden (DfuSe support)
This program is Free Software and has ABSOLUTELY NO WARRANTY

dfu-util does currently only support DFU version 1.0

Filter on vendor = 0x1d50 product = 0x6017
Opening DFU USB device... ID 1d50:6017
Run-time device DFU version 011a
Found DFU: [1d50:6017] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash   /0x08000000/8*001Ka,120*001Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
No valid DFU suffix signature
Warning: File has no DFU suffix
DfuSe interface name: "Internal Flash   "
Error: Wrong state after command download
```

```
dfu-util -d 1d50:6017 -s 0x08000000 -D blackmagic-native-v1.6.bin 
dfu-util 0.5

(C) 2005-2008 by Weston Schmidt, Harald Welte and OpenMoko Inc.
(C) 2010-2011 Tormod Volden (DfuSe support)
This program is Free Software and has ABSOLUTELY NO WARRANTY

dfu-util does currently only support DFU version 1.0

Filter on vendor = 0x1d50 product = 0x6017
Opening DFU USB device... ID 1d50:6017
Run-time device DFU version 011a
Found DFU: [1d50:6017] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash   /0x08000000/8*001Ka,120*001Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
No valid DFU suffix signature
Warning: File has no DFU suffix
DfuSe interface name: "Internal Flash   "
Error: Page at 0x08000000 is not writeable
```

```
$ dfu-util -U output.bin

dfu-util 0.5

(C) 2005-2008 by Weston Schmidt, Harald Welte and OpenMoko Inc.
(C) 2010-2011 Tormod Volden (DfuSe support)
This program is Free Software and has ABSOLUTELY NO WARRANTY

dfu-util does currently only support DFU version 1.0

Opening DFU USB device... ID 1d50:6017
Run-time device DFU version 011a
Found DFU: [1d50:6017] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash   /0x08000000/8*001Ka,120*001Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
Limiting default upload to 16384 bytes
bytes_per_hash=1024
Starting upload: [###############] finished!
```
Also, I tried to flash following the steps described here
https://medium.com/@paramaggarwal/converting-an-stm32f103-board-to-a-black-magic-probe-c013cf2cc38c
and I got the following messages

```
Can't init. Ensure that BOOT0 is enabled and reset device 
....
__main__.CmdException: Can't read port or timeout
```

UweBonnes, 2017-12-02:
Older bootloaders: Look at what the kernel reports when the bootloader is invoked.
For one device I get:
 usb 1-1.1: Product: Black Magic (Upgrade) for STLink/Discovery, (Firmware v1.6.1-51-gb984aa7-dirty)
 usb 1-1.1: Manufacturer: Black Sphere Technologies
usb 1-1.1: SerialNumber: E1E5CEDC

Dfu upload was addes\d April 27 2017.

And dfu-util 0.5 is ages old...

deangelign, 2017-12-05:
Ok. Using `dmesg` command I get

```
 usb 1-1.2: new full-speed USB device number 3 using ehci-pci
 usb 1-1.2: New USB device found, idVendor=1d50, idProduct=6017
 usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
 usb 1-1.2: Product: Black Magic Probe (Upgrade)
 usb 1-1.2: Manufacturer: Black Sphere Technologies
 usb 1-1.2: SerialNumber: 7BB297AA
```

UweBonnes, 2017-12-05:
Obviously an explicit version string is an indication of a recent bootloader. I don't think you have real problems with the hardware, the problems are more with tools and system setup.  So I don't think readback is really required for you and you can live with the old bootloader.

deangelign, 2017-12-06:
Thank for your attention, but I am not able to flash it with a new firmware. Also, I can't use it for debugging. Therefore, the board is useless. Just remembering, It used to work perfectly a month ago. It seems that it bricked or half-bricked after I have tried to upgrade the firmware.

UweBonnes, 2017-12-06:
I did not see a try  with a recent dfu-util or with the python script scripts/stm32_mem.py. Using old software often results in problems. Try to recover with new software.

The dmesg output posted above look quite sane, so it is up to you.

oferranti, 2018-05-03:
Hi all,

I have tried to use the same BlackMagic the user Deangelign attempted to reflash.

I have been not able to recover it.

All the leds starts blinking sequentially even without pressing the "boot" button.

Maybe we have messed up the DFU bootloader. Is there a way to flash the DFU bootloader using JTAG,SWD or USART (with the ROM bootloader) ?

What´s the hardware setup to use the "bootprog.py" script ?

Now, some results:

With "dfu-util" we can see that the device enumerates:

```
eld@eld:~/dfu-util/src$ sudo dfu-util -v -l

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [1d50:6017] ver=0100, devnum=2, cfg=1, intf=0, path="1-1", alt=0, name="@Internal Flash   /0x08000000/8*001Ka,120*001Kg", serial="7BB297AA"

```

So I´ve tried to flash the DFU bootloader:

```
eld@eld:~/dfu-util/src$ sudo ./dfu-util -v -S 7BB297AA -D ../../blackmagic/src/blackmagic_dfu.bin -s 0x08000000
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1d50:6017
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Memory segment at 0x08000000   8 x 1024 =  8192 (r)
Memory segment at 0x08002000 120 x 1024 = 122880 (rew)
Downloading to address = 0x08000000, size = 7360
dfu-util: Last page at 0x08001cbf is not writeable
```

I have tried also with st32_mem.py but it freezes.

```
eldo@eldo:~/blackmagic/scripts$ sudo ./stm32_mem.py -a 0x08000000 ../src/blackmagic_dfu.bin -m

USB Device Firmware Upgrade - Host Utility -- version 1.2
Copyright (C) 2011  Black Sphere Technologies
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

Device ID:	 1d50:6017
Manufacturer:	 Black Sphere Technologies
Product:	 Black Magic Probe (Upgrade)
Serial:		 7BB297AA
```

Finally, if we try to flash the firmware, we also have some error messages:

```
eld@eld:~/dfu-util/src$ sudo ./dfu-util -v -S 7BB297AA -D ../../blackmagic/src/blackmagic.bin -s 0x08002000
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1d50:6017
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Memory segment at 0x08000000   8 x 1024 =  8192 (r)
Memory segment at 0x08002000 120 x 1024 = 122880 (rew)
Downloading to address = 0x08002000, size = 79792
Download	[                         ]   0%            0 bytesstate(2) = dfuIDLE, status(0) = No error condition is present
dfu-util: Wrong state after command "ERASE_PAGE" download
```

We have a second Blackmagic Probe and it works correctly on the same Linux environment setup. I´m able to flash the DFU bootloader and the BMP firmware into this second BMP unit.

UweBonnes, 2018-09-12:
Please specify the connections you use for JTAG/SWD in your write-up.

cvanelteren, 2019-03-06:
To add my 2cts I had all the problems described above. Not entirely sure what was the magic bullet but as a tip checking and reloading my rules in /etc/udev/rules.d made it work! I think the wrong rule was loaded for me.

kunosee, 2021-04-13:
Hello,

having the same problem with a stuck BMP on a Blue Pill.  Some findings with newer dfu-utils might be also interesting.
(My current environment: Git-bash under Windows)

Flashing the bootloader via another BMP as USB-serial adapter just works fine:
`stm32flash.exe -w blackmagic_dfu.bin COM11`

Rebooted Blue Pill comes back as `Black Magic Probe DFU v1.7.1-146-ga6a8606`

Flashing the BMP with _dfu-util V0.9_ works without any failure in verbose mode. 
`dfu-util.exe -v -d 1d50:6018,:6017 -s 0x08002000 -D blackmagic.bin`
Flashed several times, resetted the device, unplugged and re-plugged, connected to an Linux notebook, nada! Always stuck in DFU-mode.

Retrieving the code from the BMP with _dfu-util V0.9_
`dfu-util.exe  -d 1d50:6018,:6017 -s 0x08002000 -U garbage.bin`
results in a file sized **16384** byte, while the blackmagic.bin was **105300** bytes in size.

Also no errors in verbose mode.

I changed to _dfu-util V1.0_ and tried to upload the binary as before.  Watch the results:
```
dfu-util.exe  -d 1d50:6018,:6017,0000:0000 -s 0x08002000 -D blackmagic.bin
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Warning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1d50:6017
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
Error: File ID 0000:0000 does not match device (1d50:6018 or 1d50:6017)
```

1. DFU-UTILS reports now a serial, which looks ways smaller than the previous 0.9 !
2. DFU-UTILS denies uploading the file, when addressing via VID/PID like before.
3. The Upload worked, when I addressed the device via its serial no.
4. Perhaps the make process needs to be supplemented by inserting the VID/PID numbers in the file, so that dfu-utils will allow flashing via VID/PID addressing

Now I repeated retrieving back the code from the device, this time the result was **122880 bytes**, which was bigger than the **105300 bytes** of the binary.  A quick look in the file showed, that it was padded with 0xFFs at the end.  Cutting out 105300 bytes from this file with dd (`dd if=garbage.bin of=garbage-reduced.bin bs=105300 count=1`) gave me a 100% identical file.

Repeating the download with dfu-utils 0.9 now also resulted in the identical 122880 byte file.

So I can conclude: My upload works, the flash cells in the Blue Pill are intact, as I am able to retrieve the byte identical file. However the device stucks in DFU mode.  Updating my existing BMP with the identical binary, works as desired.

Any other things, I can try?

kunosee, 2021-04-14:
For my problem **I can report the solution**: I forgot, that I used the SWLINK bootloaders to the native firmware for my existing probe. 
That is the result frm starting with SWLINK firmware updating to native firmware, after I soldered 3 status LEDs to my probe.
Now I pulled PB12 to high with an 10k resistor, and the bootloader problem is solved.

Therefore the **question to the issue-owner**:  Measure the voltage on pin B15!  If that is close to 0V watch the push-button and the PCB-track, if there is an problem.

I will open another issue, to track dfu-util requirements.

UweBonnes, 2021-04-14:
If you have an Stlink compatible probe, consider using the STLink Bootloader. That way you can easily switch between the ST Firmware and the BMP firmware. Look at the readme.md in the platform directories.

esden, 2021-12-15:
I think this thread has diverged from the original problem report. As the conversation is not about the original authors issue I am closing this. If you have programming issues please either ask your questions on [Discord in the blackmagic channel](https://1bitsquared.com/pages/chat) or open a dedicated issue.

closed 2021-12-15 by esden