blackmagic-debug/blackmagic #320: NRF5 stuck in reset until scan is started

UweBonnes, 2018-02-26:
Hello,

in https://github.com/sandeepmistry/arduino-nRF5/issues/252some user reports that the NRF5 does not start until a scan command is issued. As described there, SDWIO and Reset share the same line. Startup code of  the different platforms sets SWDIO low when BMP initializes the board, obvious causing this issue.

https://github.com/UweBonnes/blackmagic/tree/pin-defaults cares for the right  SDWIO and TDI defaults. 

Awaiting user reports...

probonopd, 2018-03-03:
Just a heads-up, I am currently away from the devices with no way to test. Will follow your suggestions and give an update, most likely next week. Thanks for your help.

probonopd, 2018-03-11:
Thanks @UweBonnes but after power cycling, SDWIO soldered to PB14 still is low.

rpavlik, 2018-03-19:
FYI, I have an adafruit nRF52 feather and will presumably be using the BMP v2.1 on it shortly.

flatsiedatsie, 2018-04-29:
This sounds like this problem:

https://forum.mysensors.org/topic/6961/nrf5-bluetooth-action/1583

> nrf52.cpu curstate
reset

UweBonnes, 2018-04-29:
Well, BMP should keep SWDIO tristated until needed. While tristated, pull-up internal to target SWDIO and maybe external will do the right thing.

UweBonnes, 2018-05-02:
Goggle slogans you don't understand first. Than ask more exact.

UweBonnes, 2018-06-23:
People with this problem please try #361. This pull request keeps TMS tristated until th scan happens. Please report!

closed 2018-08-01 by gsmcmullin

joric, 2018-08-02:
Related problem. I use the following script to flash new firmware:

`arm-none-eabi-gdb.exe --quiet --batch -ex "target extended-remote \\.\%port%" -ex "mon swdp_scan" -ex "file %file:\=/%" -ex "att 1" %opt% -ex "load" -ex "run"`

Ctrl+C works fine (console closes, MCU keeps running), but -ex "quit" or -ex "reset" somehow hangs the MCU. How to fix? I want a script that flashes fimware and restarts MCU on exit.

UweBonnes, 2018-08-02:
Maybe "-ex run -ex detach" is executed to fast and detach happens while CPU is still in reset. Has gdb some mean of delay?

joric, 2018-08-02:
I can't use -ex run in a batch file because it's not interactive. It's just -ex load -ex kill. Don't know about delay.