blackmagic-debug/blackmagic #321: Set SWD-CLK Speed

cad435, 2018-03-03:
Hello,

Is it possible to specify the SWD-CLK speed on the BlackMagic Probe?
I searched the net but didn't find any usable info...

Either its not possible, I didn't know the right keywords for getting the right results or I messed something up in my understanding how things come together in terms of SWD-Debugging...

(I basically hav to slow down the CLK speed because the connection is very unstable and I can not lower the lenght of the cables and PCB-Traces)

hope you guys can help or give me a hint,
cad435

UweBonnes, 2018-03-03:
SWD is done by bit-banging, There is no parameter yet for slowing it down. But try with more ground connections between your BMP and the target.Unstable connectiond often got useable for me when mor ground connections were added.

cad435, 2018-03-04:
Ok, I tried adding more GND-Connections...

but thats also extremely unstable...

little time it works, most time i got:
`Exception: SWDP invalid ACK`
or
'Exception: SWDP Parity error'

any other suggestions?

cad435, 2018-03-04:
Ok, 100 Ohm series-resistance seems fixed it...
Maybe i should try the obvious things first...

closed 2018-03-04 by cad435

UweBonnes, 2018-03-04:
The GPIO settings may also help. However getting everything right for everybody is a hard thing...