blackmagic-debug/blackmagic #323: STM32L496 support

arthurv, 2018-03-18:
Hi,

I've been able to connect to an STM32L476 target and start debugging, however I cannot connect to a close relative chip STM32L496. Is it possible to add support for the L496?
```
(gdb) monitor swdp_scan
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32L4xx
(gdb) attach 1
Attaching to Remote target
Attaching to Remote target failed
```

UweBonnes, 2018-03-18:
Please try #310

arthurv, 2018-03-19:
@UweBonnes tried it, unfortunately I still get the same issue. Unplugging from my L496 target and plugging in to my L476 target confirms it's still not compatible with L496. Targets are electrically identical except for the uC.

UweBonnes, 2018-03-19:
Does the L49 hardware really work as expected? Can you connect to the L49 with some other debugger?

arthurv, 2018-03-20:
Apologies - it was the code running on the L496 which was preventing GDB from attaching to the target. I can confirm it works now.

closed 2018-03-21 by arthurv