blackmagic-debug/blackmagic #327: Design approach question

konsgn, 2018-03-27:
I feel like I'm missing something here.

I wanted to port this project to the device [CSR-SPI](https://github.com/raplin/CsrUsbSpiDeviceRE) But it seems that only some of the spi pins are broken out into the jtag/swdp interface.

When I looked at the stlinkv2 design, I noticed that it had a sort of similar breakout.

At first I thought that the layout was not at all considering of possible alternate uses, but seeing the tck's line up with the spi clocks made me think that was done on purpose. If so, I don't get it. This code seems to be mostly bitbanging so why that wireup?

Just to be clear I am not looking for alternatives to my porting project (I will be targeting a raw blue-pill board). Rather, I am wondering if there is a reason for the com lines to be tied to the pins that they are? Or if it was a random hookup, can it be considered to wire up the mosi/miso as tdo/tdi in further designs?

UweBonnes, 2018-03-27:
At the moment, I think the JTAG/SWD pins are choosen at random. Having SWO on a UART RX pin will allow fast asynchronous SWO against SWO on a timer input only allowing calculation intensive synchronous SWO.

Better pin assignment with more capable SPI peripherals may or may not allow faster Jtag/Swd with future drivers.

gsmcmullin, 2018-04-08:
The pin assignments were chosen based on what was easy to route on the first hardware revision.  The possibility of doing fancy footwork to use the SPI peripherals wasn't considered, nor was any possibility of capturing any trace info.

closed 2018-04-08 by gsmcmullin