blackmagic-debug/blackmagic #338: Compiling for stlink platform in Ubuntu 18.04

freynder, 2018-04-25:
After losing several hours of my time trying to troubleshooting and bricking an stlink device, I thought it might be helpful to inform you of the issues I experienced when compiling BMP for stlink in the upcoming Ubuntu release.

When compiling under Ubuntu 18.04, the resulting blackmagic image files (dfu and main) do not work on probes like blue pill and stlink. They are not recognized as USB devices.

I'm unsure of the exact cause but was able to reproduce it using docker instances using exactly the same commands to compile. The resulting file sizes are different as well:

Ubuntu 18.04: 
-rwxr-xr-x 1 root root 1110156 Apr 25 11:12 blackmagic
-rwxr-xr-x 1 root root   73544 Apr 25 11:12 blackmagic.bin
-rwxr-xr-x 1 root root  368220 Apr 25 11:12 blackmagic_dfu
-rwxr-xr-x 1 root root    7684 Apr 25 11:12 blackmagic_dfu.bin
-rw-r--r-- 1 root root   21672 Apr 25 11:12 blackmagic_dfu.hex

Ubuntu 17.10
-rwxr-xr-x 1 root root 949620 Apr 25 10:44 blackmagic
-rwxr-xr-x 1 root root  62288 Apr 25 10:44 blackmagic.bin
-rwxr-xr-x 1 root root 344924 Apr 25 10:44 blackmagic_dfu
-rwxr-xr-x 1 root root   7380 Apr 25 10:44 blackmagic_dfu.bin
-rw-r--r-- 1 root root  20817 Apr 25 10:44 blackmagic_dfu.hex

I hope this helps.

UweBonnes, 2018-04-25:
Please check your build setup.  The size of blackmagic.bin 
-rwxr-xr-x 1 root root 73544 Apr 25 11:12 blackmagic.bin
looks more like compiled for native or similar. Check if these files are the riight ones, e.g. with strings:
strings blackmagic_dfu.bin |tail -5
Black Sphere Technologies
Black Magic (DFU Upgrade) for STLink/Discovery, (Firmware v1.6.1-134-ga8e93b0)
Black Magic (Upgrade) for STLink/Discovery, (Firmware v1.6.1-134-ga8e93b0)
 @Internal Flash   /0x08000000/8*001Ka,000*001Kg
@Internal Flash   /0x08000000/8*001Kg

freynder, 2018-04-25:
Here is the strings output:
$ strings blackmagic_dfu.bin |tail -5
Black Sphere Technologies
Black Magic (DFU Upgrade) for STLink/Discovery, (Firmware v1.6.1-128-g455e0a7)
Black Magic (Upgrade) for STLink/Discovery, (Firmware v1.6.1-128-g455e0a7)
 @Internal Flash   /0x08000000/8*001Ka,000*001Kg
@Internal Flash   /0x08000000/8*001Kg

I already tried to eliminate any external factors by using docker instances with a minimal build setup. You can easily replicate the problem by:

1) starting docker container:
docker pull ubuntu:bionic
docker run ubuntu:bionic

2) in the docker instance:
apt update
apt upgrade
apt install git make gcc-arm-none-eabi python
git clone https://github.com/blacksphere/blackmagic.git
cd blackmagic/
make
cd src
make clean
make PROBE_HOST=stlink
ls -al blackmagic*

UweBonnes, 2018-04-25:
I installed docker. However the run command does not give me a different envionment:
 hertz:~> docker run ubuntu:bionic
 hertz:~> cat /etc/issue
 Welcome to openSUSE Leap 42.3 - Kernel \r (\l).
Any help welcome!

freynder, 2018-04-25:
Sorry, I forgot the -it option:
docker run -it ubuntu:bionic
should get you into a container shell

UweBonnes, 2018-04-25:
I can reproduce the error. I misstrust the compiler in that image/ Any alternatives?

freynder, 2018-04-25:
I experienced the issue first on the native ubuntu system before
reproducing with docker. Checksums of the compilation results are the same
so the docker image is representative.

semicolo, 2018-05-21:
Same issue for me on Ubuntu 18.04, a freshly built BMP firmware does not work on the STLink V2 clone I have (does not enumerate correctly). I tried a pre built firmware found here: https://github.com/BuFran/blackmagic-binaries and it works, but it's 5 years old (V 1.5).
That one is smaller, could it be that the micro hasn't got the full 128Ki flash?
There's a computer with an older Ubuntu at work, I'll try again wednesday.

lorenz, 2018-06-19:
The toolchain on Ubuntu 18.04 is indeed a problem. A firmware built from the same source in a Fedora container works, but the same build on Ubuntu 18.04 HardFaults pretty much immediately. Both builds are over 64K, so no difference in the handling of high flash.

UweBonnes, 2018-06-19:
Can some Ubuntu user please open a bug with Ubuntu? Or is there already some bug opened? Thanks!

UweBonnes, 2019-02-21:
Closing as probably caused by a bad Ubuntu arm-none toolchain.

closed 2019-02-21 by UweBonnes