blackmagic-debug/blackmagic #343: The 'c' and 's' packets ignore the address field

bobbatcomcastdotnet, 2018-05-02:
The 'c' (continue) and 's' (step) request packets ignore the address field in the packet. This means that continuing or stepping from a specific address is not supported.

gsmcmullin, 2018-05-04:
To resume from a specific address, GDB sets the registers with the 'G' packet, and then resumes with the 'c' packet.  This works perfectly.

bobbatcomcastdotnet, 2018-05-04:
@gsmcmullin ; I understand that your suggestion is a work around for the current implementation of the 'c' and 's' requests. The issue I opened points out a deviation of these two requests from the GDB Remote Serial Protocol specification. I opened this issue as advised by @UweBonnes since I believe the BMP firmware should implement requests according to the GDB RSP specification.

UweBonnes, 2018-05-04:
Bob: What about trying to implement yourself. Get a bmp git setup that you can compile. Test by recompiling master and check that it works. Then take a look at src/gdb_main.c how the 'g' and  'G' commands handle register, try to excerpt the address from pbuf and and that to th 'c' command.

esden, 2021-12-15:
@bobbatcomcastdotnet Are you using some other software than the original GDB? Is that why you are running into this issue? I am trying to get some context for this issue.

sidprice, 2022-06-26:
@dragonmux This looks low-hanging fruit here and I see value in adding the feature. Being able to continue or step from a specific address would be nice. If you would like I can take a look at it.

dragonmux, 2022-06-26:
Certainly, please be our guest.

sidprice, 2022-07-05:
@dragonmux It seems there is confusion here about the optional parameters with the 'c' and 's' commands. Although the BMP source code comments (in gdb_main.c) say the optional parameter is and address, the GDB manual for the 'c' and 's' commands do not say this.

In the case of 'c' the optional argument is the 'ignore-count', which defines the number of times the BMP should ignore the breakpoint at the current address.

Similarly, for the 's' command, the optional argument is 'count', and it defines the number of steps to be performed.

The confusion here is that the GDB RSP does say these 'c' and 's' commands have optional arguments to define the address of the continuation point or the address of the step to be executed. 

When a 'c' command is issued with GDB, BMP receives the 'c' RSP command as expected, however, I don't see how to have GDB send an address as an argument. The GDB help for the 'c' and 's' commands agrees with the manual.

What am I missing here?

dragonmux, 2022-07-05:
We would suggest that means the BMP source documentation is misleadingly wrong - we always defer to what GDB does in these situations. Indeed, the GDB man page backs up your analysis, so we suggest updating the comments accordingly and, if you can, implementing the missing functionality.

sidprice, 2022-07-05:
The conflict is between the GDB command reference and the RSP reference.

sidprice, 2022-07-05:
So, you suggest implementing what GDB sends and ignore what the RSP reference says?

dragonmux, 2022-07-05:
judging by this.. we have [the 'c' packet](https://docs.jade.fyi/gnu/gdb/gdb.html#index-c-packet) and [the 's' packet](https://docs.jade.fyi/gnu/gdb/gdb.html#index-s-packet) which match the BMP documentation.

This is indeed different from the front-end documented behaviour for continue and single-step, which suggests that either GDB is adding a pile of extra logic in this or the remote protocol reference is outdated.. this requires 'loosing the game' (ie diving into the GDB sources to see what it actually does). Stand by

sidprice, 2022-07-05:
A quick test with the 'c' command suggests that GDB is managing the skipping of breakpoint encounters. I have not tested 's'.

dragonmux, 2022-07-05:
Looks like at [least for step](https://github.com/bminor/binutils-gdb/blob/gdb-10.2-release/gdb/remote.c#L6336), the address parameter never gets generated..

We came to that conclusion after finding [remote_resume_with_hc](https://github.com/bminor/binutils-gdb/blob/gdb-10.2-release/gdb/remote.c#L6204) which is how we do resumptions and neither 'c' nor 's' can generate an address according to this code - perhaps it's a historical artefact

sidprice, 2022-07-05:
So, we should close this as "will not implement" and fix the bad comments in another PR?

dragonmux, 2022-07-05:
Yeah, think that's the best course of action we have with the findings we've both made