blackmagic-debug/blackmagic #344: STM32F030F4 problems

evark, 2018-05-04:
Blackmagic programs the protection option when loading an elf, and after that I can't remove the protection. The only way to erase the chip is by using STLink.
```
Remote debugging using /dev/ttyACM0
Target voltage: Present
Available Targets:
No. Att Driver
 1      STM32F03
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xfffffffe in ?? ()
(gdb) mon erase_mass
(gdb) mon option
Device is Read Protected
Use "monitor option erase" to unprotect, erasing device
(gdb) mon option erase
0x1FFFF800: 0x5AA5
0x1FFFF802: 0xFFFF
0x1FFFF804: 0xFFFF
0x1FFFF806: 0xFFFF
0x1FFFF808: 0xFFFF
0x1FFFF80A: 0xFFFF
0x1FFFF80C: 0xFFFF
0x1FFFF80E: 0xFFFF
(gdb) mon option      
Device is Read Protected
Use "monitor option erase" to unprotect, erasing device
(gdb) mon erase_mass     
(gdb) load test3_mxf0.elf
Error erasing flash with vFlashErase packet
```

UweBonnes, 2018-05-07:
The option arenot yet reloaded. If you test now, it will probably work, as you proably unpligged the power. To make ne option active, reset or un-and repower the device.

sinihappo, 2018-07-10:
I have the fix for this.  Should I send the patch or will you give me rights to make a branch a pull request?

Well, the patch is [here:](https://alo.fi/BMP-ed6fb7/BMP-patch-20180710-032707.patch)

UweBonnes, 2018-07-10:
Ah, the fix seems obvious. Can you produce a Pull request for Gareth or should I?

sinihappo, 2018-07-10:
I can produce a pull request if someone gives me rights to push a branch.  As it happens, I will be doing some real work using BMP in the near future so it is not totally impossible I will fix something else, too.

But you may do it yourself if you want, I don't mind.

Bseides, I have not studied different STM32s thoroughly but shouldn't the default case be to write this exactly same value for all devices?  I didn't dare to change the default behaviour, though.

UweBonnes, 2018-07-10:
You don't need any rights. Clone the master repo on your github account, create a branch, push the change to that branch and create the pull request on that branch.

STM32 devices are not orthogonal at all. There are many exceptions. You have to carefully read the datasheet and reference manual for every devices and implement what is needed for the device.

gsmcmullin, 2018-07-10:
Fixed by #370

closed 2018-07-10 by gsmcmullin