blackmagic-debug/blackmagic #348: Misleading message after scanning a target

esden, 2018-05-20:
https://github.com/blacksphere/blackmagic/blob/48d232807ee6096bfa839d91b1708252db0e1a59/src/target/stm32f4.c#L198

Scanning a target that is not an STM32F4 will lead to a misleading message. This driver is not just STM32F4 any more. Scanning STM32F7 or STM32F2 targets might lead to the following message:

```
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32F4
```

Even if the target is an F2 or an F7. This is quite misleading. The target type strings are being set correctly later during `attach`.

I think either the generic driver string should be changed to `STM32F2/F4/F7` or the correct target name associated with the ID should be set at scan and not later during attach.

Let me know what you think. :)

UweBonnes, 2018-06-04:
The problem came up as some STM32F4 register, needed for setting up the memory map,  can not be accessed without halting the target. At the moment, probe happens with the target not halted. So probe  and memory map setup was split. However L0 and F7 can not be probed at all without halting the target. So I strongly proposed to halt all target when probing. Then we can set up again the F4 and other dual bank memory maps with probe again and use the right device string.

esden, 2018-06-05:
The #349 PR fixes the issue. Closing.

closed 2018-06-05 by esden