blackmagic-debug/blackmagic #359: commit 471ce2547c42 breaks BMP on blue pill

arjunak234, 2018-06-21:
When running master on a blue pill i get the following error(with an Arduino Due as target):

>Target voltage: unknown
>Ignoring packet error, continuing...

It works if i revert that commit. Can someone with a bluepill please test it out?

RickKimball, 2018-06-21:
I grabbed the latest ae6f0eadc9748deced4e48074eb7d73064cc4e28 and it seems to work fine with ubuntu and gcc 7.2.1

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
```

arjunak234, 2018-06-21:
I tried using ubuntu 18.04 but the USB doesn't even enumerate. Am i flashing it right

>openocd -f /usr/share/openocd/scripts/board/st_nucleo_f103rb.cfg -c init -c 'reset halt' -c 'flash write_image erase src/blackmagic;reset halt;flash write_image erase src/blackmagic_dfu;reset;exit'

UweBonnes, 2019-02-21:
Closing as probably caused by broken Ubuntu arm-none gcc.

closed 2019-02-21 by UweBonnes

arjunak234, 2019-02-23:
>Closing as probably caused by broken Ubuntu arm-none gcc.

I use fedora, I tried it on Ubuntu because the comment above said he got it to work on Ubuntu.

UweBonnes, 2019-02-23:
Argh, lpc15xx_probe seems to be as intrusive as lpc17xx_probe. As STM is probed before lpc, stm32 has no problem. sam is probed later and hangs in lpc17xx_probe. Move lpc17xx_probe. to the end  and try again and if that succeeds, do a pull request. Oook at f39701c4c8977dfcaa1548573fa8aae7777d2891