blackmagic-debug/blackmagic #364: Use LVFS for distributing firmware updates?

hughsie, 2018-06-28:
Hi,

My name is Richard Hughes and I work for Red Hat. I'm the maintainer of fwupd, an open source firmware update system designed for Linux. We're updating over 350,000 devices every month safely and securely using the Linux Vendor Firmware Service. We've had a large number of hardware vendors join this service in the last few months, including HP, Dell, Logitech and Jabra.

Using the LVFS means you're able to offer firmware updates onto thousands of devices without having to use a command line update tool. The best bit is that it's a completely free service, and I'm happy to help you as much as you need. DFU is already well supported in fwupd so I don't think there would be any code to write.

If this sounds interesting, please see https://fwupd.org/ or ask me a question here.

There's a draft technical whitepaper here if you'd like more details https://people.freedesktop.org/~hughsient/temp/LVFS_Whitepaper.pdf about how the service works. Thanks for reading!

Richard

UweBonnes, 2018-07-03:
How do you think that some update service does fit  for a device that is plugged  as tool into different computers?

hughsie, 2018-07-03:
Well, as I understood it the firmware was updatable using DFU. The idea would be that you insert the hardware into a Linux PC, it tells you have updates available, installs them for you, then you remove the device.

dragonmux, 2023-02-12:
o/ hughsie. We had to do some research on what fwupd was and what LVFS was, and using this channel for helping people update their Black Magic Probe native hardware would be of potential interest to us. Sorry it's been 4 and a bit years since you opened this.

We do have a few questions - for example, can fwupd use product description strings to tell native hardware apart from the other platforms so we can target the update files to only native hardware? This is unfortunately important because different platforms need quite different firmware update files and the only way to tell them apart is the presence (or absence) of "(<platform>)" in the product string.

As an example, BMP (native) uses the product string "Black Magic Probe v1.9.0", while an ST-Link v2 would show as "Black Magic Probe (ST-Link/v2) v1.9.0"

hughsie, 2023-02-13:
> can fwupd use product description strings to tell native hardware apart from the other platforms

Yes, we might need to add a quirk for that specific device, but we certainly have other devices that use the product string, e.g. https://fwupd.github.io/libfwupdplugin/emmc-README.html#guid-generation

dragonmux, 2023-02-13:
👍🏼  excellent, then we'll have a talk with @esden about it and work with you to get the firmware listed as this makes a lot of sense for the typical user to ensure they get the latest release firmware if they're on old firmware

The only other thing of note is that the DFU implementation is specifically DFUse based rather than straight DFU, so needs rebooting into the firmware with the DFUse-specific `-s 0x08002000:leave` (as it would be given to dfu-util) - we assume that's not a problem as that's cropped up before in fwupd?

hughsie, 2023-02-13:
> specifically DFUse based rather than straight DFU

I think that should be fine; lots of STM32 devices are supported now

> DFUse-specific -s 0x08002000:leave

Hmm, that's a new one; normally we do the empty payload to manifest trick, i.e. https://github.com/fwupd/fwupd/blob/main/plugins/dfu/fu-dfu-target-stm.c#L29

I'm more than happy to implement that as a quirk if needed, but I do wonder if the fwupd attach function works too. e.g. if you do `fwupdtool --plugins dfu attach` if that puts the device from bootloader to runtime.

VilleWitt, 2025-10-15:
As a user, this would be a nice feature.

Just inserted 1.9 and saw there is a 2.0-rc2. fwupd would be nice for when 2.0 is official.

VilleWitt, 2025-10-16:
I used bmputil to get developing.

I came to the conclusion that fwupd is probably NOT the way to go for one simple reason: one has to select firmware depending on target. I had 4 options: ARM, RISC, ..., not just "the current thing". I don't see how fwupd is able to switch between "branches".

dragonmux, 2025-10-17:
So.. VID:PID will always be the same; the extra nugget you mention is only available after upgrade to v2.0.0 firmware and is the response to the architectures and families remote protocol requests: https://github.com/blackmagic-debug/blackmagic/blob/942817692d1b6a326c1a0a4b816a434e4d17b423/src/remote.c#L363-L439

this then pairs with a decoder key of sorts for which combination of those represents which of the release variants; finally, we'll also need to key on the product string for whether the probe is running stock native firmware or not because we do not want to shoot "upgrades" at, eg, ST-Link v2 firmware probes as that'll mess them up and brick.

hughsie, 2025-10-17:
I guess it really boils down to if the developers _want_ to use LVFS to distribute firmware -- I don't want to force anyone to do it of course. I don't have any Backward-compatible hardware and so someone else would need to make any fwupd changes needed for things like the VID/PID disambiguation.

dragonmux, 2025-10-17:
That would not solve the problem in question - the VID:PID being the same saying "this is Black Magic Debug firmware" is not a problem (additionally, getting more PIDs is a big pain and would break a lot more as this impacts BMDA, `bmputil`, `bmpflash`, the compuphase suite, etc when from a software perspective.. it's all the same product - same protocol, same interface). We can tell platforms apart from the string value accessed by following iProduct. The problem is that solutions that increase Flash usage rather than using existing mechanisms built into the firmware, like the remote protocol, are a very poor fit because we're already so Flash constrained. So, for example, encoding something about the architectures and targets supported by a build into the product string is a no-go. hughsie's suggestion of extending the BOS which we already use for Microsoft OS Descriptors would work except it does increase Flash usage too, and we already released v2.0.0, so it doesn't work for the existing firmware.

We need to think on this and get back to hughsie, though we do have one further question for him: can fwupd be constrained to only run checks when it sees the device appear? because if so, then it won't matter if it temporarily opens the serial interface, runs a couple of remote protocol commands, and closes it again. That won't hurt anything to do.

hughsie, 2025-10-17:
> can fwupd be constrained to only run checks when it sees the device appear

Yup, that's what it does right now -- it caches all the information about the device and then only re-opens the device when the flashing is about to start.