blackmagic-debug/blackmagic #365: STM32F76x Error erasing flash with vFlashErase packet

andrewleech, 2018-07-03:
Hi, I'm trying to debug a STM32F765 but it fails to load with `Error erasing flash with vFlashErase packet`
I'm using an stlink clone with firmware updated just now from: http://builds.blacksphere.co.nz/blackmagic
If it helps, current head is at  d4d24c2

```
target extended-remote COM4
-> Remote debugging using COM4

file firmware.elf
->done

set mem inaccessible-by-default off
->done

monitor swdp_scan
-> Target voltage: unknown
-> Available Targets:
-> No. Att Driver
->  1      STM32F76x
-> done

attach 1
->done

info mem
-> Using memory regions provided by the target.
-> Num Enb Low Addr   High Addr  Attrs 
-> 0   y  \t0x00000000 0x00004000 rw nocache 
-> 1   y  \t0x00200000 0x00210000 flash blocksize 0x4000 nocache 
-> 2   y  \t0x00210000 0x00220000 flash blocksize 0x10000 nocache 
-> 3   y  \t0x00220000 0x00400000 flash blocksize 0x20000 nocache 
-> 4   y  \t0x08000000 0x08010000 flash blocksize 0x4000 nocache 
-> 5   y  \t0x08010000 0x08020000 flash blocksize 0x10000 nocache 
-> 6   y  \t0x08020000 0x08200000 flash blocksize 0x20000 nocache 
-> 7   y  \t0x20000000 0x20020000 rw nocache 
-> 8   y  \t0x20020000 0x20080000 rw nocache 

load
-> error,msg="Error erasing flash with vFlashErase packet"
```

I previously had issues flashing with this part with a segger, the dual bank flash seemed to trip it up. Is it likely something similar here?

UweBonnes, 2018-07-03:
Check the device option bits. Erase option bits and check again. If there is a differences,perhaps you accidental write protected the flash. Mass erase the device. Does that work? Can you flash after option and flash mass erase?  If second flash does not work, check your programm that is does not do anything strange.

andrewleech, 2018-07-03:
As a test I switched to a different stlink converted to segger jlink and that works fine, no mass erase needed.

andrewleech, 2018-07-03:
```
>arm-none-eabi-size -x -A *.elf
firmware.elf  :
section               size         addr
.text             0x12d544    0x8020000
.fb_memory         0x60000   0x20020000
.dma_memory         0xf400   0x20000000
.data                 0xac   0x2000f400
.bss                0x1c70   0x2000f4ac
._heap              0xd000   0x2001111c
._stack             0x1000   0x2001e11c
.ARM.attributes       0x37          0x0
.debug_info       0x4d84b7          0x0
.debug_abbrev      0x45948          0x0
.debug_aranges      0xaf90          0x0
.debug_macro       0x7115b          0x0
.debug_line        0xf6ebd          0x0
.debug_str        0x16fe6b          0x0
.comment              0x7e          0x0
.debug_loc        0x121ad3          0x0
.debug_ranges      0x17ed0          0x0
.debug_frame       0x22974          0x0
Total             0xb0963e
```

UweBonnes, 2018-07-03:
I could try to flash your elf file to a nucleo-f767 if you make it available.

andrewleech, 2018-07-03:
I can share it tomorrow when I'm back in the office however it's based on this so should be completely compatible with the elf here:

https://github.com/openmv/openmv/tree/master/firmware/OPENMV3

I haven't tried flashing the elf directly from these guys though.

andrewleech, 2018-07-04:
I just double checked and yes, firmware.elf from the above git repo does have the same error for me.
jlink with the same elf works fine.

UweBonnes, 2018-07-04:
Will check with a F767 a.s.a.p.

UweBonnes, 2018-07-05:
Please test #366. Fixes obvious omission and fixxes the problem for me

andrewleech, 2018-07-05:
That looks great thanks. I'll test in when I'm back in the office in around 12 hours.

andrewleech, 2018-07-06:
Yes, I can confirm this does fix the problem, thanks!

gsmcmullin, 2018-07-07:
Fixed by #366

closed 2018-07-07 by gsmcmullin

MReed1959, 2020-02-12:
I've seen this a few times on a couple of different STM32 systems.
What I've found is that the debugger NRST is not working. I no idea why, everything is connected correctly and the NRST pin is not being held by an external device. The only solution is to go into the debugger setup and switch to "Software system reset". Then you don't even need the reset line.