blackmagic-debug/blackmagic #367: cortexm_forced_halt in cortexm_probe() fails

jdk, 2018-07-07:
I couldn't say if this is wide spread yet. But I have preliminary platform support for the samd2x and the following line in cortexm_probe() always fails and thus returns false:

```
        if (!cortexm_forced_halt(t))
                  return false;
```

The target is another samd21E17 part, so that might have something to do with it, but will further test to see if this problem exists for other targets. I will also rebase on top of the stable 1.6.1 firmware and see if there are any issues there.

Commenting out the above line, I can detect the correct chip, if this line is in there, I get a generic CortexM which I can't attach to.

UweBonnes, 2018-07-08:
Where does your setup abort in cortexm_forced_halt(t))? Can you either instrument that code and use the blackmagic debug output on the second ACM or use a FTDI debugger and single step? Does your SRST setup work as required?

gsmcmullin, 2018-07-08:
@jdk Are you trying to scan while the target is held in reset?

Assuming your device is similar to those already supported, this may be because of some problem the samd parts have with SRST:
https://github.com/blacksphere/blackmagic/blob/master/src/target/samd.c#L181-L190

Or the extra magic needed to get them out of reset:
https://github.com/blacksphere/blackmagic/blob/master/src/target/samd.c#L206-L211

jdk, 2018-07-10:
Sorry, I spent a good time trying to reproduce this, and I wasn't successful. I added back said code above in my original comment and it's been working fine. I apologize, and I'll close the issue and reopen if I'm able to reliably reproduce.

closed 2018-07-10 by jdk