blackmagic-debug/blackmagic #381: Flashing nRF52832 in __WFE() loop fails

chacal, 2018-08-22:
I have encountered a curious issue when flashing [EByte nRF52832](http://www.cdebyte.com/en/product-view-news.aspx?id=243) module using BMP from master (`c5c0783337..`) on STM32F108 "Bluepill" board.

Everything works just fine as long as the target system is running software that doesn't explicitly conserve power, but as soon as the target system is put into low power mode by issuing `__WFE()` in the main loop the `mon swdp_scan` command fails consistently and thus flashing new software is not possible anymore.

Working `main.c`:
```
#include <nrf.h>

int main(void) {
  for(;;) {
  }
}
```

Failing `main.c`:
```
#include <nrf.h>

int main(void) {
  for(;;) {
    __WFE();
  }
}
```

Interestingly the target can be "resurrected" by attaching it to a Raspberry Pi and running `mass_erase` command there using OpenOCD. After this, flashing using BMP again works as the target is not in the `__WFE()` loop anymore.

Any ideas what might cause this issue? Is there something in BMP firmware that I could try to tweak to get this working better?

mczerski, 2018-12-02:
@chacal I have exactly the same problem. Could You explain how did managed to recover the nrf52832 from this state using RPI ? Is there any software for the RPI that makes it a debugger ? I was trying to connect with openocd and st-link v2 from my pc but it is also failing ...

chacal, 2018-12-02:
@mczerski I think I followed this tutorial: https://learn.adafruit.com/programming-microcontrollers-using-openocd-on-raspberry-pi/overview

The idea is to use Raspberry Pi's GPIO pins directly with OpenOCD instead of some other hardware programmer.

tl;dr from the above link:

1. Install latest Raspbian on your Pi
1. Install dependencies: `apt install git autoconf libtool make pkg-config libusb-1.0-0 libusb-1.0-0-dev`
1. `git clone git://git.code.sf.net/p/openocd/code openocd # Clone openocd`
1. `cd openocd`
1. `./bootstrap`
1. `./configure --enable-sysfsgpio --enable-bcm2835gpio`
1. `make`
1. `sudo make install`

After compiling OpenOCD connect Pi and nRF52 as follows:

    Raspi  | nRF52
    ------ | ------
    3V3    | 3V3
    GND    | GND
    GPIO25 | SWIO
    GPIO11 | SWCLK

Then run this command on Pi's console:

    sudo openocd -f interface/raspberrypi-native.cfg -c "transport select swd" -f target/nrf52.cfg -c "init; nrf52.dap apreg 1 0x0c; nrf52.dap apreg 1 0x04 0x01; reset; nrf5 mass_erase; reset; shutdown;"

The output should be something like this:

```Open On-Chip Debugger 0.10.0+dev-00410-gf0767a31 (2018-05-09-19:36)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO config: tck = 11, tms = 25, tdi = 10, tdo = 9
BCM2835 GPIO nums: swclk = 11, swdio = 25
swd
adapter speed: 1000 kHz
cortex_m reset_config sysresetreq
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : JTAG and SWD modes enabled
Info : clock speed 1006 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
0x00000001
Error: nrf52.cpu -- clearing lockup after double fault
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0xffffffd8
Polling target nrf52.cpu failed, trying to reexamine
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : nRF52832-QFAA(build code: B0) 512kB Flash
Error: nrf52.cpu -- clearing lockup after double fault
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0xffffffd8
Polling target nrf52.cpu failed, trying to reexamine
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
shutdown command invoked
Warn : Flash driver of nrf52.flash does not support free_driver_priv()
Warn : Flash driver of nrf52.uicr does not support free_driver_priv()
```

After that the nRF52 is again programmable using STM32F108 "Bluepill".

mczerski, 2018-12-02:
ok, I got it. openocd can use plain gpios to connect over swd or jtag. And I can confirm that openocd is able to connect to nrf52832 with sysfsgpio interface.

mczerski, 2018-12-02:
@chacal thanks

UweBonnes, 2018-12-02:
You should try #399 and attach under reset. Please report back.

mczerski, 2018-12-03:
I don't know if I'm doing it ok. This is my gdb command:

> arm-none-eabi-gdb -ex "target extended-remote /dev/ttyACM0" -ex "monitor connect_srst enable" -ex "monitor swdp_scan" -ex "att 1"

and this is the output:

> Remote debugging using /dev/ttyACM0
> Target voltage: ABSENT!
> SW-DP scan failed!
> Attaching to Remote target
> Attaching to Remote target failed

UweBonnes, 2018-12-03:
Is SRST connected?

mczerski, 2018-12-03:
Well this is the whole problem that swdp scan fails. It fails with bmp and openocd + st link v2. Only openocd on rpi using gpio bitbanging is able to scan successfully. The wiring is correct and no srst.

UweBonnes, 2018-12-03:
Is there some SWCLK restriction for the NRF when in sleep? Perhaps the RPI is much slower? Can you perhaps try OpenOCD  with a much slower SWCLK frequency and see if that succeeds?

mczerski, 2018-12-03:
I don't know what I did but now the openocd + st link v2 works. Maybe it was working with openocd all the time but the wiring was bad ... BMP still does not work. After flashing with openocd program without __WFE call, BMP can connect to the board again. Openocd uses 950kHz clock.

UweBonnes, 2018-12-03:
Does "mon s" still show the error when running with BMP?

mczerski, 2018-12-03:
yes.
So to be sure that it is not the wiring problem I did this:
1. flashed the board with program that do not call __WFE
2. connected BMP to the board
3. flash the board with BMP with program that calls __WFE:
```
GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor swdp_scan
Target voltage: ABSENT!
Available Targets:
No. Att Driver
 1      Nordic nRF52
 2      Nordic nRF52 Access Port
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x000004e4 in ?? ()
(gdb) 
(gdb) load Release/nrf52_test.elf 
Loading section .isr_vector, size 0x1fe lma 0x0
Loading section .inits, size 0x28 lma 0x200
Loading section .text, size 0x2c0 lma 0x228
Loading section .data, size 0xc lma 0x4e8
Start address 0xe0, load size 1266
Transfer rate: 19 KB/sec, 316 bytes/write.
(gdb) detach 
Detaching from program: , Remote target
(gdb) quit
```
4 disconnect the BMP from the PC to power off the board (I power the board from BMP and until the board is not powered off I can connect with BMP)
5. connect the BMP to the PC and try to connect to the board:
```
GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor swdp_scan
Target voltage: ABSENT!
SW-DP scan failed!
```

UweBonnes, 2018-12-03:
Is there e reset buttom on the board? Run "mon s" with the reset buttom pressed.

mczerski, 2018-12-03:
unfortunately no, hower I can attach a switch to the reset but I need some tools for that. I will post when I do that.

UweBonnes, 2018-12-03:
https://devzone.nordicsemi.com/f/nordic-q-a/15204/swd-frequency does not give a limit for swdclk that BMP could reach. So the reason must be different
.

UweBonnes, 2018-12-03:
Try to find out when "SW-DP scan failed!" is issued during the initialization sequence. Either compile with "ENABLE_DEBUG=1", open the serial terminal on startuo and switch on debugging before doing the scan, or run with a FTDI MPSSE  adapter and look at the staring console.

mczerski, 2018-12-05:
using reset I can connect to the board. I press the reset button, launch "monitor swdp_scan" and then release reset button.
I'm unable to see the debug logs. I'm pretty sure that I enabled the debug mode because the binary of the bmp firmware is much larger than before. This is what I typed into the gdb:
```
 (gdb) monitor debug_bmp enable
Debug mode is enabled
```
and then I connected to ttyACM1 (the second tty that is created after bmp is connected to pc) but nothing is showing up.

UweBonnes, 2018-12-05:
After debug enable. do "mon s". That shoud print messages.

mczerski, 2018-12-05:
They should appear in gdb console?

UweBonnes, 2018-12-05:
No, the terminal console. But something must happen so that messgaes show up.. The something may be e.g. "mon s"

mczerski, 2018-12-06:
Still nothing shows up when i run scan

UweBonnes, 2018-12-06:
What platform do you use? Mayb there is something fishy with debug on that platform.

mczerski, 2018-12-06:
I'm using swlink platform on blue pill board.

UweBonnes, 2018-12-06:
Try
diff --git a/src/platforms/swlink/platform.c b/src/platforms/swlink/platform.c
index 42dfd0b..ff206f2 100644
--- a/src/platforms/swlink/platform.c
+++ b/src/platforms/swlink/platform.c
@@ -114,7 +114,6 @@ void platform_init(void)
        /* Don't enable UART if we're being debugged. */
        if (!(SCS_DEMCR & SCS_DEMCR_TRCENA))
                usbuart_init();
-       usbuart_init();
 }
 
recompile and reload and report back.

mczerski, 2018-12-06:
now I get something on the debug console but ... when the connection is made (target extended-remote /dev/ttyACM0) i get this:
```
*** Unsupported packet: vMustReplyEmpty
*** Unsupported packet: Hg0
*** Unsupported packet: qTStatus
```
then, when I run monitor swdp_scan I get only one empty line in the console ...

UweBonnes, 2018-12-06:
And now do the scan!

UweBonnes, 2018-12-06:
Sorry, had other things to do and look only short. Do the same on an erased chip ans see what is expected. 

The newline comes from adiv5_swdp_scan.
       if((ack != SWDP_ACK_OK) || swdptap_seq_in_parity(&dp->idcode, 32)) {
                DEBUG("\n");
                free(dp);
                return -1;
 
Probably both conditions should be checked indepentant and emit their own descriptive DEBUG message.

UweBonnes, 2018-12-09:
You should decipher, if the error is SWDP_ACK_WAIT or SWDP_ACK_FAULT or a combination. If it is SWDP_ACK_WAIT only, the SWD interface tells us simply to wait and try again...

mczerski, 2018-12-11:
the ack value is 0

UweBonnes, 2018-12-11:
Perhaps revert 46b681e050bbfb27acafab88a3b67bda292825ca that speeded up SWD. An ACK value of 0 is very strange.

cwmoreiras, 2020-04-30:
Has this been resolved? Experiencing this issue with nrF52840

UweBonnes, 2020-04-30:
This is thread miles long. So "Has _this_ been resolved"  is very generic.

cwmoreiras, 2020-04-30:
The thread is miles long, and the whole thing is about one issue, which is the fact that the BMP can't flash the nRF52 when it's in a low power state.

UweBonnes, 2020-05-01:
No cortex can be reliable debugged when the core is in a low power state. It is up to the target code to keep the debug units running even in sleep. For STM32, there are the DBGMCU_CR bits. Look if NRF has something similar and implement is src/nrf5. 
What is the exact sequence to take? Attach to the target, stop by ^C and then "load"? Does BMP still have control after the ^C.

It helps to be as expicite as possible.

PhracturedBlue, 2021-11-17:
Huge thanks to @chacal for the RPi solution to this.  One note is that the Debian Bullseye kernel doesn't have the /dev/gpiomem patches, and I couldn't access the GPIO with the stock kernel.  I compiled the raspbian kernel (for aarch64 Bullseye), and was able to use /dev/gpiomem.

What is needed to fix BMP for this issue?
The RPi can reprogram an nrf52 that is in deep sleep using only Vcc, Gnd, SWDIO, and SWCLK.  The BMP cannot program an nrf52 with these same 4 pins connected when it is in deep sleep due to 'mon swd' failing to detect the nrf52 board.

I am not sure that the chip detection routines are comparable between BMP and OpenOCD, but would it be useful to connect a logic-analyzer and dump the traffic of each case?

UweBonnes, 2021-11-17:
Please run recent BMP as hosted on the failing setup and post the output. The step before using a LA is running hosted with -v31, but here it is only the second step.

PhracturedBlue, 2021-11-18:
The program:
```
#include <zephyr.h>
#include <pm/pm.h>

void main(void)
{
    NRF_POWER->SYSTEMOFF = 1;
    while(1) {}
}
```
After flashing with BMP, using the BMP to try to scan via SWD:
```sudo ./blackmagic -t
BMP hosted v1.7.1-284-g7307f08
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 E1D2B9C2 Black Sphere Technologies
 Black Magic Probe (SWLINK) v1.7.1-284-g7307f08
Running in Test Mode
Target voltage:  Volt
Speed set to  3.2727 MHz for SWD
Exception: SWDP Parity error
Trying old JTAG to SWD sequence
Exception: SWDP Parity error
No usable DP found
Scan SWD failed, trying JTAG!
Speed set to  3.2727 MHz for JTAG
No target found```

Trying again using gdb directly:
```
gdb-multiarch -ex 'target extended-remote /dev/ttyACM0' -ex 'mon swdp_scan'
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Remote debugging using /dev/ttyACM0
SW-DP scan failed!
Failed
```
Here is what the logic-analyzer captured:
```
Time	Type	R/W	AP/DP	Register	Request byte	ACK	WData	WData details
0.00000041	Line reset							
0.00005575	Line reset							
0.00007541	Operation	read	DebugPort	IDCODE	'165'	OK	'4294965891'	DESIGNER='2691', PARTNO='65535', Version='15'
0.00011341	Line reset							
```

Using SWD on RPi via OpenOCD:
```
sudo openocd -f interface/raspberrypi-native.cfg -c "transport select swd" -f target/nrf52.cfg -c "init; nrf52.dap apreg 1 0x0c; nrf52.dap apreg 1 0x04 0x01; reset"
Open On-Chip Debugger 0.11.0+dev-00464-g5a0b4889d (2021-11-16-18:30)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
trst_only separate trst_push_pull

swd
Info : BCM2835 GPIO JTAG/SWD bitbang driver
[... 24 lines cut ...]
0.00048325	Operation	write	DebugPort	CTRL/STAT	'169'	OK	'1342177312'	CSYSPWRUPACK=0, CSYSPWRUPREQ=1, CDBGPWRUPACK=0, CDBGPWRUPREQ=1, CDBGRSTACK=0, CDBGRSTREQ=0, TRNCNT='0', MASKLANE='0', WDATAERR=0, READOK=0, STICKYERR=1, STICKYCMP=0, TRNMODE=Normal, STICKYORUN=0, ORUNDETECT=0
0.00051583	Operation	read	DebugPort	CTRL/STAT	'141'	OK	'4026531840'	CSYSPWRUPACK=1, CSYSPWRUPREQ=1, CDBGPWRUPACK=1, CDBGPWRUPREQ=1, CDBGRSTACK=0, CDBGRSTREQ=0, TRNCNT='0', MASKLANE='0', WDATAERR=0, READOK=0, STICKYERR=0, STICKYCMP=0, TRNMODE=Normal, STICKYORUN=0, ORUNDETECT=0
0.00055133	Operation	write	DebugPort	CTRL/STAT	'169'	OK	'1342177280'	CSYSPWRUPACK=0, CSYSPWRUPREQ=1, CDBGPWRUPACK=0, CDBGPWRUPREQ=1, CDBGRSTACK=0, CDBGRSTREQ=0, TRNCNT='0', MASKLANE='0', WDATAERR=0, READOK=0, STICKYERR=0, STICKYCMP=0, TRNMODE=Normal, STICKYORUN=0, ORUNDETECT=0
0.00058350	Operation	read	DebugPort	CTRL/STAT	'141'	OK	'4026531840'	CSYSPWRUPACK=1, CSYSPWRUPREQ=1, CDBGPWRUPACK=1, CDBGPWRUPREQ=1, CDBGRSTACK=0, CDBGRSTREQ=0, TRNCNT='0', MASKLANE='0', WDATAERR=0, READOK=0, STICKYERR=0, STICKYCMP=0, TRNMODE=Normal, STICKYORUN=0, ORUNDETECT=0
0.00062441	Operation	read	DebugPort	CTRL/STAT	'141'	OK	'4026531840'	CSYSPWRUPACK=1, CSYSPWRUPREQ=1, CDBGPWRUPACK=1, CDBGPWRUPREQ=1, CDBGRSTACK=0, CDBGRSTREQ=0, TRNCNT='0', MASKLANE='0', WDATAERR=0, READOK=0, STICKYERR=0, STICKYCMP=0, TRNMODE=Normal, STICKYORUN=0, ORUNDETECT=0
0.00066475	Operation	read	DebugPort	CTRL/STAT	'141'	OK	'4026531840'	CSYSPWRUPACK=1, CSYSPWRUPREQ=1, CDBGPWRUPACK=1, CDBGPWRUPREQ=1, CDBGRSTACK=0, CDBGRSTREQ=0, TRNCNT='0', MASKLANE='0', WDATAERR=0, READOK=0, STICKYERR=0, STICKYCMP=0, TRNMODE=Normal, STICKYORUN=0, ORUNDETECT=0
0.00069991	Operation	write	DebugPort	CTRL/STAT	'169'	OK	'1342177281'	CSYSPWRUPACK=0, CSYSPWRUPREQ=1, CDBGPWRUPACK=0, CDBGPWRUPREQ=1, CDBGRSTACK=0, CDBGRSTREQ=0, TRNCNT='0', MASKLANE='0', WDATAERR=0, READOK=0, STICKYERR=0, STICKYCMP=0, TRNMODE=Normal, STICKYORUN=0, ORUNDETECT=1
0.00073225	Operation	read	DebugPort	CTRL/STAT	'141'	OK	'4026531841'	CSYSPWRUPACK=1, CSYSPWRUPREQ=1, CDBGPWRUPACK=1, CDBGPWRUPREQ=1, CDBGRSTACK=0, CDBGRSTREQ=0, TRNCNT='0', MASKLANE='0', WDATAERR=0, READOK=0, STICKYERR=0, STICKYCMP=0, TRNMODE=Normal, STICKYORUN=0, ORUNDETECT=1
0.00078366	Operation	write	DebugPort	SELECT	'177'	OK	'240'	APSEL='0', APBANKSEL='15', PRESCALER='0'
0.00081616	Operation	read	AccessPort	IDR	'159'	OK	'0'	Revision='0', JEP-106 continuation='0', JEP-106 identity='0', Class=This AP is not a Memory Acces Port, AP Identfication='0'
0.00101958	Operation	read	DebugPort	RDBUFF	'189'	OK	'611778577'	
0.00106125	Operation	read	AccessPort	CFG	'175'	OK	'611778577'	Big-endian
0.00110241	Operation	read	DebugPort	RDBUFF	'189'	OK	'0'	
0.00114400	Operation	write	DebugPort	SELECT	'177'	OK	'0'	APSEL='0', APBANKSEL='0', PRESCALER='0'
0.00117683	Operation	write	AccessPort	CSW	'163'	OK	'2717909024'	DbgSwEnable=1, Prot=", SPIDEN=0, Mode='0', TrInProg=0, DeviceEn=0, AddrInc=Increment packed, Size=Byte (8 bits)
0.00121533	Operation	write	AccessPort	TAR	'139'	OK	'0'	
0.00125316	Operation	read	AccessPort	CSW	'135'	OK	'0'	DbgSwEnable=0, Prot='0', SPIDEN=0, Mode='0', TrInProg=0, DeviceEn=0, AddrInc=Auto-increment off, Size=Byte (8 bits)
0.00129325	Operation	read	DebugPort	RDBUFF	'189'	OK	'587202656'	
0.00133775	Operation	write	AccessPort	CSW	'163'	OK	'2717909026'	DbgSwEnable=1, Prot=", SPIDEN=0, Mode='0', TrInProg=0, DeviceEn=0, AddrInc=Increment packed, Size=Word (32 bits)
...
```

And to reiterate, Once I wipe the flash via the RPi, BMP is able to communicate with the board just fine.

I also have full capture logs from the logic analyzer if they are useful.

UweBonnes, 2021-11-18:
Please try now with "connect under reset" (hosted -t -C). I suspect that the second AP gets visible and you can mass erase the device on the second AP.

UweBonnes, 2021-11-18:
Can the LA display HEX numbers? The decimal numbers above are not intuitive

PhracturedBlue, 2021-11-18:
I have updated the traces above to be in hex.

Using `blackmagic -t -C` did the trick (mostly).  When I run 'use connect under reset', the nrf52 is not immediately detected, but subsequent `blackmagic -t` or gdb `mon swdp_scan` calls work as desired.

```
blackmagic -t
BMP hosted v1.7.1-284-g7307f08
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 E1D2B9C2 Black Sphere Technologies
 Black Magic Probe (SWLINK) v1.7.1-284-g7307f08
Running in Test Mode
Target voltage:  Volt
Speed set to  3.2727 MHz for SWD
Exception: SWDP Parity error
Trying old JTAG to SWD sequence
Exception: SWDP Parity error
No usable DP found
Scan SWD failed, trying JTAG!
Speed set to  3.2727 MHz for JTAG
No target found

blackmagic -t -C
BMP hosted v1.7.1-284-g7307f08
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 E1D2B9C2 Black Sphere Technologies
 Black Magic Probe (SWLINK) v1.7.1-284-g7307f08
Connecting under reset
Running in Test Mode
Target voltage:  Volt
Speed set to  3.2727 MHz for SWD
Exception: SWDP Parity error
Trying old JTAG to SWD sequence
Exception: SWDP Parity error
No usable DP found
Scan SWD failed, trying JTAG!
Speed set to  3.2727 MHz for JTAG
No target found

blackmagic -t
BMP hosted v1.7.1-284-g7307f08
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 E1D2B9C2 Black Sphere Technologies
 Black Magic Probe (SWLINK) v1.7.1-284-g7307f08
Running in Test Mode
Target voltage:  Volt
Speed set to  3.2727 MHz for SWD
Exception: SWDP Parity error
Trying old JTAG to SWD sequence
DPIDR 0x2ba01477 (v1 rev2)
RESET_SEQ failed
AP   0: IDR=24770011 CFG=00000000 BASE=e00ff003 CSW=23000040 (AHB-AP var1 rev2
Halt via DHCSR: success 00030003 after 7ms
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 244 Partno   6
0 0xe000e000: Generic IP component - Cortex-M4 SCS (System Control Space) (PIDR = 0x04000bb00c  DEVTYPE = 0x00 ARCHID = 0x0000)-> cortexm_probe
CPUID 0x410fc241 (M4 var 0 rev 1)
1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x04003bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x04002bb003  DEVTYPE = 0x00 ARCHID = 0x0000)
3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x04003bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
4 0xe0040000: Debug component - Cortex-M4 TPIU (Trace Port Interface Unit) (PIDR = 0x04000bb9a1  DEVTYPE = 0x11 ARCHID = 0x0000)
5 0xe0041000: Debug component - Cortex-M4 ETM (Embedded Trace) (PIDR = 0x04000bb925  DEVTYPE = 0x13 ARCHID = 0x0000)
ROM: Table END
AP   1: IDR=02880000 CFG=00000000 BASE=00000000 CSW=00000000 (AHB-AP var0 rev0
***  1      Nordic nRF52 M4
***  2      Nordic nRF52 Access Port 
RAM   Start: 0x20000000 length = 0x10000
Flash Start: 0x00000000 length = 0x80000 blocksize 0x1000
Flash Start: 0x10001000 length = 0x1000 blocksize 0x1000
```

However, it is not 100% reliable.
While the behavior above is usually reproducible, sometimes after the `blackmagic -t -C` it takes several tries of `blackmagic -t` before I can see the device.  If I run `blackmagic -t -C` more than once in a row, then the device never becomes available (even using `blackmagic -t`) and needs a hard reset.  Once the device has been detected (as above), then `blackmagic -C -t` can also detect the device.

So the behavior is not very intuitive.  If this is expected behavior, it would be nice if there was a place in the Wiki for it.  Even better would be if SWD initialization could be as reliable so that a single command could detect the device.

Either way, thanks for your help.  it is good to know that the BMP can recover devices stuck in sleep mode on its own

UweBonnes, 2021-11-18:
Please post the lines of "blackmagic -t -v31 >& /tmp/log" up to" Exception: SWDP Parity error."

PhracturedBlue, 2021-11-18:
Well...while it was mostly reliable when I tried using `blackmagic -C -t` earlier today (~90% success rate), when I tried it now to capture the logs, it took > 15 attempts before I was able to successfully attach to the device using the above method, so I guess it is less reliable than I 1st thought.  There is no change in my setup since then.

Here is the log (the log up to this point is identical whether the connection was successful or failed)
```
BMP hosted v1.7.1-284-g7307f08
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 E1D2B9C2 Black Sphere Technologies
 Black Magic Probe (SWLINK) v1.7.1-284-g7307f08
+#!GA#
       K(SWLINK) v1.7.1-284-g7307f08
Remote is (SWLINK) v1.7.1-284-g7307f08
!GZ0#
       K0
Running in Test Mode
 !GV#
       K
Target voltage:  Volt
!GF003d0900#
       K0
!Gf#
       K17f03100
!Gf#
       K17f03100
Speed set to  3.2727 MHz for SWD
remote_swdptap_init
!SS#
       K0
swdptap_seq_out        32 ticks: ffffffff
!So20ffffffff#
       K0
swdptap_seq_out        32 ticks: ffffffff
!So20ffffffff#
       K0
swdptap_seq_out        32 ticks: 6209f392
!So206209f392#
       K0
swdptap_seq_out        32 ticks: 86852d95
!So2086852d95#
       K0
swdptap_seq_out        32 ticks: e3ddafe9
!So20e3ddafe9#
       K0
swdptap_seq_out        32 ticks: 19bc0ea2
!So2019bc0ea2#
       K0
swdptap_seq_out        12 ticks: 000001a0
!So0c1a0#
       K0
swdptap_seq_out        32 ticks: ffffffff
!So20ffffffff#
       K0
swdptap_seq_out        32 ticks: 0fffffff
!So20fffffff#
       K0
swdptap_seq_out         8 ticks: 000000a5
!So08a5#
       K0
!Si03#
       K1
swdptap_seq_in          3 ticks: 00000001
!SI20#
       P3000067
swdptap_seq_in_parity  32 ticks: 03000067 ERR
Exception: SWDP Parity error
```

UweBonnes, 2021-11-18:
I noticed that that the DP behaves different after powerup or a successfull run. So un- and repower the NRF on each run.  Please try https://github.com/UweBonnes/blackmagic/tree/fixes

PhracturedBlue, 2021-11-18:
That seems to have done the trick.

I have done many cycles of detecting and reprogramming and so far in 100% of tests, `blackmagic -t` and gdb `mon swdp_scan` can consistently detect the nrf52 when it is in deep-sleep.  No need for the '-C' switch anymore

UweBonnes, 2021-11-18:
It the original problem fixed? Can this issue be closed?

PhracturedBlue, 2021-11-18:
While I didn't open the issue, I am using the same nrf board in question, and the fix you applied did resolve the original issue for me.

UweBonnes, 2021-11-19:
Issue(s) seem to be resolved. Closing. Is similar issues with similar devices are still pending, please open a new issue for clarity.

closed 2021-11-19 by UweBonnes