blackmagic-debug/blackmagic #382: Board revision detection goes wrong for genuine ST-Link V2-1

rikvdh, 2018-08-22:
Hi,

I have a ST-Link V2-1 where I've installed (latest master) of blackmagic. But I've had to do some changes in order to make it work.

When I opened my ST-link to flash the bootloader at first it wouldn't detect the target. Also the on-board LED showed some odd behavior. By looking at the version output in GDB and the code it detected that my ST-link was an V1 instead of a V2.

Looking at the board it was quite obvious. On PC13/14 where the board-revision detection was. Only the second resistor was assembled. So the fourth (non-implemented) option was actually present. PC13 has a not-assembled resistor. The eventual detection becomes than`10` which than resolves to `rev 0` in `stlink_common.c`.

I shall try to make a proper fix for this and send a pull-request.

rikvdh, 2018-09-02:
Yes I did, I want to make a pull request because it requires some rewriting in the ST-Link detection part. I've also managed to make target voltage work for this platform. But I wanted some other changes first in the stlink platform since the code there is (in my opinion) a bit difficult to read sometimes.

See https://github.com/blacksphere/blackmagic/pull/384 and https://github.com/blacksphere/blackmagic/pull/383

esden, 2021-12-15:
@rikvdh I see that one of the patches is merged and the other one is closed. Is this issue still present? If not I would like to close this issue as it is becoming a bit stale.

dragonmux, 2022-06-27:
We ran the latest Black Magic Firmware on a ST-Link v2.1 (which is part of our Nucleo-F441RE) and it all worked..

We're thinking this isn't a problem any more, could @rikvdh please confirm so we can either close this issue or make a PR that fixes any remaining issues.

dragonmux, 2022-08-01:
On account of the lack of answer and seeing the most recent testing of the code in question showed no issues, closing as solved. If this isn't true, please re-open this issue or create a new one detailing the problem you're having.

closed 2022-08-01 by dragonmux