blackmagic-debug/blackmagic #385: attach not working under reset

anyn99, 2018-09-02:
I have a custom board with an STM32L432KB.
There was some trouble when trying to attach to it, scanning worked fine most of the time.
I did some, well quite a lot of debugging, and it seems that the issue is related to my board.
It has only two layers and the SWDCLK and SDWIO traces are routed all over the place - they are simple button inputs when not debugging. The BMP code as it is, is just to fast for this! :-)

Would it be possible to add some that can be set in the monitor menu?

I did something like this: 
`void delay(unsigned int ticks)
{
  volatile unsigned int i;
  for (i = 0; i < ticks; i++)
  {
    asm("nop");
  }
}
`
And then at every clock out in /platforms/common/swdptap.c
`        gpio_set(SWCLK_PORT, SWCLK_PIN);
        delay(DEFAULT_DELAY);
	gpio_clear(SWCLK_PORT, SWCLK_PIN);
	delay(DEFAULT_DELAY);`
>

UweBonnes, 2018-09-03:
Are you using git head? Git Head has no sequence to 
bla()
gpio_set()
gpio_clear()
bla()

Either gpio_set() is already doubled or there are other commands in between. So the time the signals are high is already extended as before e54a826745ae (loop unrolling). Even before with the short high times, BMP worked for many boards.

Adding a function call in the unrolled loops with result in substantial longer programming times.

Did you look at the signals on your board with a scope? Can you compare against against e.g. a BMP modified Nucleo Board?

anyn99, 2018-09-03:
yes i use the current head, but modified it until it works for me, sorry for being unclear about that.
Here some more info. I actually don't use original BMP, but ST-LINK v2 "Baite" clone.
Sorry that i can't compare to a different BMP probe....

I started using the code as is, with gpio_set() doubled in swdptap.c, scanning worked fine, but when attaching the probe failed everytime at the first adiv5_swdp_low_access. After the register write (8 bits) the ack was always 111 (0x7) instead of 100 or 010 or at least 001. So i think it was a timing issue on the bus.
Maybe i will try to doube the gpio_clear()s as well and see if it works.

UweBonnes, 2018-09-03:
Did you try "mon assert_srst attach"

anyn99, 2018-09-03:
Yes i did, but as mentioned already, this is bot sufficient since it asserts just before attaching. During attach, there is a call to cortexm_force_halt() which deasserts srst!

UweBonnes, 2018-09-03:
It is good to keep all relevant information in one thread. You probably mentioned the fact in another thread...

Perhaps "Request halt on reset" should be sent to the Cortex-M before scanning when "mon assert_srst attach" is active? Can you try?

anyn99, 2018-09-03:
Sorry, didn't mean to be rude. I just wrote that i “connect under reset“, not that i do it with “mon ass attach“, my mistake.

Actually your suggestion sounds like a very good solution! I will try that and report back!

anyn99, 2018-09-03:
Fixed!
In the end it was little more than moving one line of code....
About to create a pull request...

closed 2019-01-16 by anyn99