blackmagic-debug/blackmagic #389: EFM32G222F64 detection in efm32_probe doesn't work

rikvdh, 2018-09-05:
I found out that the reading of the `efm32_v1_read_eui64` works, but after shifting and masking there is something completely different than `EFM32_V1_DI_EUI_SILABS`. The resulting value from the shift and mask (`((efm32_v1_read_eui64(t) >> 40) & 0xFFFFFF)`)of the EUI64 is: `0x0024bcdc`.

Also the format of this register is not documented by Silabs, so I think it is a bad idea to rely on it.

rikvdh, 2018-09-05:
I found some knowledge base article on the unique-id registers from early 2017.
https://www.silabs.com/community/mcu/32-bit/knowledge-base.entry.html/2017/11/01/efm32_and_efr32_seri-uVaX

When looking through the public IEEE listing for OUI's (http://standards-oui.ieee.org/oui/oui.txt) I find several entries for Silabs including the current one. Also the energy micro variant is there. But my variant (see below) is not listed.

My part is the EFM32G222F64G-E-QFP48R (via digikey ordered, so I don't expect counterfeits) https://www.digikey.nl/product-detail/en/silicon-labs/EFM32G222F64G-E-QFP48R/336-4245-1-ND/7805193

```
EFM32 EUI64 0x24bcdc 055a 02d28e
EFM32 EUI48 0x574e2c 1cffff
```

UweBonnes, 2018-09-05:
Can you use the  Simplicity Commander “device info” command on your device? What does it tell?

rikvdh, 2018-09-05:
I can't use it, it needs a Segger J-Link which I don't have. I shall try to find someone in my area which can lend it to me.

UweBonnes, 2018-09-05:
The EFM demo boards have a Segger J-Link on-board. It can also be used for external devices. Do you perhaps have such a device?

rikvdh, 2018-09-05:
I don't have such board. I shall arrange something to test this with. Perhaps someone of my collegues has a J-Link.

N-Storm, 2018-09-24:
I have a custom EFM32**T**G222F32 board. And I can't even swdp_scan it:
```
Remote debugging using /dev/ttyACM0
(gdb) monitor swdp_scan
Target voltage: unknown
Ignoring packet error, continuing...
Ignoring packet error, continuing...
^C
(gdb) monitor targets
Available Targets:
No. Att Driver
No usable targets found.
```

My BMP are running on "Blue Pill" and firmware are build from recent git master branch:
```
kernel: [636015.175336] usb 5-4: Product: Black Magic Probe (STLINK), (Firmware v1.6.1-214-g29386ae)
kernel: [636015.175337] usb 5-4: Manufacturer: Black Sphere Technologies
```

I've had a JLink and worked fine with this board before. Any suggestions?

rikvdh, 2018-09-24:
@N-Storm This is not related to this. Please use the latest Blackmagic firmware, the LP17xx probe gets stuck. See https://github.com/blacksphere/blackmagic/issues/388 and https://github.com/blacksphere/blackmagic/pull/390

N-Storm, 2018-09-25:
My datecode are 1501. My chip was ordered from DigiKey as well (i.e. it can't be fake). Here it is:
[image]

richardeoin, 2018-10-20:
I've proposed a solution for this in pull #402

Silabs are using some additional OUIs we don't know about. Reported in issue #389
These should use a DI version 1 layout, so assume version 1 layout for OUIs we don't
know. However do print a notice about this on DEBUG() as suggested by @UweBonnes

The IDCODE value is sufficient to make a positive identification of an EFM32 device.
See AN0062 Section 2.2. Therefore accepting any OUI is reasonable behaviour.
Additionally the part familiy is checked, and the target rejected if not in the
`efm32_devices` table. This commit makes that rejection explicit, although it does
not change the logical behaviour here.

Note that the important registers (part number, part family, flash size) are at the
same addresses in both layouts anyhow. Currently only `efm32_cmd_serial` and
`efm32_cmd_efm_info` functions use registers that differ between DI versions.

gsmcmullin, 2019-01-15:
Should be fixed by #402

closed 2019-01-15 by gsmcmullin