blackmagic-debug/blackmagic #394: Build blackmagic

filiporlo, 2018-09-26:
Hi!
I'm trying to build blackmagic I clone this repo and then:
```
git submodule init
git submodule update
make
```
and then shows some errors:
```
In file included from ../../../include/libopencm3/cm3/nvic.h:133,
                 from ../../../include/libopencm3/cm3/vector.h:41,
                 from ../../cm3/vector.c:22:
../../../include/libopencm3/dispatch/nvic.h:6:11: fatal error: libopencm3/stm32/f0/nvic.h: No such file or directory
 # include <libopencm3/stm32/f0/nvic.h>
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [../../Makefile.include:46: vector.o] Error 1
make[1]: *** [Makefile:64: lib/stm32/f0] Error 2
make: *** [Makefile:8: all] Error 2
```

filiporlo, 2018-09-26:
Strange I do everything same on vmware on ubuntu and it works(Hurra)
Next problem is that if I use this precompiled files from this webpage https://embdev.net/articles/STM_Discovery_as_Black_Magic_Probe
Blackmagicprobe is seen in windows.
When I compiled from git clone no errors, windows tell 
A request for the USB device descriptor failed.

UweBonnes, 2018-09-26:
What PROBE_PLATFORM do you compile? I did not test PROBE_PLATFORM=f4_discovery for ages. Perhaps bitrot...

filiporlo, 2018-09-26:
My board is one of cheap stlink V2 with 128kb STM32F103 so my command is 'make PROBE_HOST=stlink'

UweBonnes, 2018-09-26:
Some changes needed to recognize recent ST-Nucleo STLINK boards probably made the "cheap stlink V2" aka "baite" clone unrecognized. An overhaul is needed. Try https://github.com/UweBonnes/blackmagic/tree/baite and report back.

filiporlo, 2018-09-27:
Still not working and windows says Unknown USB device.

seppestas, 2018-10-03:
I have a similar error when attempting to build from master (29386aee140e5e99a958727358f60980418b4c88) on mac OS:

```
...
  CC      platforms/native/platform.c
In file included from ../libopencm3/include/libopencm3/cm3/nvic.h:133:0,
                 from platforms/native/platform.c:33:
../libopencm3/include/libopencm3/dispatch/nvic.h:8:11: fatal error: libopencm3/stm32/f1/nvic.h: No such file or directory
 # include <libopencm3/stm32/f1/nvic.h>
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

UweBonnes, 2018-10-03:
Did you init and update the submodule? Did you run make from the base directory?

UweBonnes, 2018-10-03:
Don't tell only what is not working. Tell also what you did yourself to track down the error!

seppestas, 2018-10-08:
Woops, I built from the src folder iof from the base directory. Still I would expect this to work as well since the src folder has its own Makefile.

@filiporlo maybe this is your problem as well.

Having a standalone Makefile that is dependent on external libraries is not very good practice IMHO. At least it could check if the libraries it depends on are there and give a useful error message when they are not... Would it be interesting to create a separate issue for this?

kousu, 2018-10-15:
I'm having similar issues on OS X.

```
git clone 
cd blackmagic
make # -> this works, and produces src/{blackmagic,blackmagic_dfu}{,.bin}
```

but I want to install on an ST-link programmer (actually a cheap clone product) as in [this blog post](http://blog.linuxbits.io/2016/02/15/cheap-chinese-st-link-v-2-programmer-converted-to-black-magic-probe-debugger/), so I did 

```
make clean && make PROBE_HOST=stlink
```

as the wiki ([indirectly](https://github.com/blacksphere/blackmagic/wiki/Debugger-Hardware#st-link) suggests [via](https://medium.com/@paramaggarwal/converting-an-stm32f103-board-to-a-black-magic-probe-c013cf2cc38c)). That worked for me.

For fun I tried `make all_platforms` and had some failures. I zeroed in on them: hydrabus and libftdi are broken:

```
anaesthetic-mac:src kousu$ make clean && make PROBE_HOST=hydrabus
  CLEAN
  GIT     include/version.h
  CC      target/adiv5.c
  CC      target/adiv5_jtagdp.c
  CC      target/adiv5_swdp.c
  CC      command.c
  CC      target/cortexa.c
  CC      target/cortexm.c
  CC      crc32.c
  CC      target/efm32.c
  CC      exception.c
  CC      platforms/stm32/gdb_if.c
  CC      gdb_main.c
  CC      gdb_hostio.c
  CC      gdb_packet.c
  CC      hex_utils.c
  CC      target/jtag_scan.c
  CC      platforms/stm32/jtagtap.c
  CC      target/jtagtap_generic.c
  CC      target/lmi.c
  CC      target/lpc_common.c
  CC      target/lpc11xx.c
  CC      target/lpc17xx.c
  CC      target/lpc15xx.c
  CC      target/lpc43xx.c
  CC      target/kinetis.c
  CC      main.c
  CC      morse.c
  CC      target/msp432.c
  CC      target/nrf51.c
  CC      platforms/hydrabus/platform.c
  CC      target/sam3x.c
  CC      target/sam4l.c
  CC      target/samd.c
  CC      target/stm32f1.c
  CC      target/stm32f4.c
  CC      target/stm32h7.c
  CC      target/stm32l0.c
  CC      target/stm32l4.c
  CC      platforms/common/swdptap.c
  CC      target/swdptap_generic.c
  CC      target/target.c
  CC      platforms/common/cdcacm.c
  CC      platforms/stm32/traceswo.c
  CC      platforms/stm32/usbuart.c
  CC      platforms/stm32/serialno.c
  CC      platforms/common/timing.c
  CC      platforms/stm32/timing_stm32.c
  LD      blackmagic
  OBJCOPY blackmagic.bin
  OBJCOPY blackmagic.hex
Creating blackmagic.dfu
  File "../scripts/dfu-convert.py", line 24
    print 'File: "%s"' % file
                     ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print('File: "%s"' % file)?
make: *** [blackmagic.dfu] Error 1
```

this is because I am on a fresh install and `brew install python` now means **python 3**. This one can be glued over by this patch:

```
diff --git a/src/platforms/hydrabus/Makefile.inc b/src/platforms/hydrabus/Makefile.inc
index 36973e0..660c8aa 100644
--- a/src/platforms/hydrabus/Makefile.inc
+++ b/src/platforms/hydrabus/Makefile.inc
@@ -26,7 +26,7 @@ all: blackmagic.bin blackmagic.hex blackmagic.dfu
 
 blackmagic.dfu: blackmagic.hex
        @echo Creating $@
-       @python ../scripts/dfu-convert.py -i $< $@
+       @python2 ../scripts/dfu-convert.py -i $< $@
 
 host_clean:
        -$(Q)$(RM) blackmagic.bin blackmagic.hex blackmagic.dfu
```

**plus** you need to `sudo pip2 install intelhex`.

but there's other uses of python in the repo that are bitrotting. They should probably all be updated to python3.

```
anaesthetic-mac:src kousu$ make clean && make PROBE_HOST=libftdi
  CLEAN
  GIT     include/version.h
  CC      target/adiv5.c
In file included from target/adiv5.c:24:
In file included from include/general.h:33:
platforms/libftdi/platform.h:24:10: fatal error: 'ftdi.h' file not found
#include <ftdi.h>
         ^~~~~~~~
1 error generated.
make: *** [adiv5.o] Error 1
```

This one can be fixed by `brew install libftdi` plus this patch:

```
diff --git a/src/platforms/libftdi/Makefile.inc b/src/platforms/libftdi/Makefile.inc
index 5b7b17f..dec24e1 100644
--- a/src/platforms/libftdi/Makefile.inc
+++ b/src/platforms/libftdi/Makefile.inc
@@ -1,4 +1,4 @@
-CFLAGS += -DLIBFTDI
-LDFLAGS += -lftdi -lusb
+CFLAGS += -I/usr/local/include/libftdi1 -DLIBFTDI
+LDFLAGS += -lftdi1 -lusb-1.0
 
 SRC +=         timing.c        \
diff --git a/src/platforms/libftdi/swdptap.c b/src/platforms/libftdi/swdptap.c
index 150083b..917b8b2 100644
--- a/src/platforms/libftdi/swdptap.c
+++ b/src/platforms/libftdi/swdptap.c
@@ -186,7 +186,7 @@ uint32_t swdptap_seq_in(int ticks)
 void swdptap_seq_out(uint32_t MS, int ticks)
 {
        uint8_t cmd[15];
-       uint index = 0;
+       uint32_t index = 0;
        swdptap_turnaround(0);
        while (ticks) {
                cmd[index++] = MPSSE_TMS_SHIFT;
@@ -209,7 +209,7 @@ void swdptap_seq_out_parity(uint32_t MS, int ticks)
        uint8_t parity = 0;
        int steps = ticks;
        uint8_t cmd[18];
-       uint index = 0;
+       uint32_t index = 0;
        uint32_t data = MS;
        swdptap_turnaround(0);
        while (steps) {
```

but the lines

```
+CFLAGS += -I/usr/local/include/libftdi1 -DLIBFTDI
+LDFLAGS += -lftdi1 -lusb-1.0
```

are darwin specific and I'm not sure what the proper fix is.

UweBonnes, 2018-10-15:
I would love to do it the way you describe it. But git head uses travis CI, and (to my knowledge) the github CI environment does not know about ftdi1.

+#ifdef __has_include
+# if __has_include(<libftdi1/ftdi.h>)
+#  include <libftdi1/ftdi.h>
+#  define HAS_LIBFTDI1
+# else
+#  include <ftdi.h>
+# endif
+#else
+# include <ftdi.h>
+#endif
 
should help. Stay tuned...

UweBonnes, 2018-10-15:
To my knowlwdedge, "python: will call python2 and "python3" call python3.

crackwitz, 2019-01-03:
I have the same problem. it boils down to the specified submodule version of libopencm3 _not_ having _any_ nvic.h in `include/libopencm3/stm32/`, but requiring it in `include/libopencm3/dispatch/nvic.h`

there is no way to build this. the files just aren't there.

please advise.

crackwitz, 2019-01-03:
My bad. the libopencm3 Makefile uses `find` and on windows, that's been shadowed by windows' own `find`, not the POSIX one. I had to make sure that the POSIX one is found first in `PATH` env var. then the build succeeds...

UweBonnes, 2019-01-03:
Any idea how to improve the build process? 
BMP uses some old libopencm3 snapshot. Does a recent libopencm3 perhaps have a solution for that problem?

crackwitz, 2019-01-03:
I am noone to ask about this. libopencm3 still says "The API of the library is NOT yet considered stable". Trying to build blackmagic with freshest libopencm3 gives me a calling signature error for usbd_register_control_callback.

Building on Windows is often the path less traveled. Pitfalls such as find/find are unfortunate, but there's little to be done about that. Even the builds I did manage didn't do anything except light the LED on my Maple Mini clone (STM32F103) and run in circles (I watched the processor state with an ST-LINK). Maybe the code doesn't like the board, or Windows doesn't like the USB stuff...

UweBonnes, 2019-01-03:
I did not mean to build with recent libopencm3. You should only _look_ if libopencm3 uses another approach now for find. Then port only that solution.

hierophect, 2019-05-07:
Just ran into this. I'm not used to Windows path issues, what did you switch around to make sure the windows find was behind the POSIX one? 

This seems like a very serious issue. Does this hit every windows user?

crackwitz, 2019-05-07:
yes, I `set PATH=X:\wherever-the-desired-find-resides;%PATH%` in the terminal I needed it for.

hierophect, 2019-05-07:
Cripes I'm not good at windows path shenanigans, I don't even know where my correct `find` would be. I guess I'll just compile this on my mac and transfer the files over.

kousu, 2019-05-08:
> To my knowlwdedge, "python: will call python2 and "python3" call python3.

It depends on the system. On ArchLinux and OS X where you've `brew install python && brew link python`, python = python3. On Debian and on Ubuntu LTS python = python2. But python2 is being killed at the end of this year so very very soon everyone is going to have to ship python3 by default.

crackwitz, 2019-05-08:
> Cripes I'm not good at windows path shenanigans, I don't even know where my correct `find` would be.

that may depend on what kind of setup you have. I have msys2. try `which -a find` to list all instances that can be found. it will probably give you unix paths. use `cygpath --windows /usr/bin/find` (or equivalent) to get the windows-style path.

UweBonnes, 2019-05-08:
Please people, BMP is a community effort. Bringing the helper tools to python 3  would be a nice contribution...

crackwitz, 2019-05-08:
so... make everything in [/scripts](/blacksphere/blackmagic/tree/master/scripts) execute properly with python3? eh, I'm good enough with python, so I'll see what I can do, but no promises ;)

hierophect, 2019-05-08:
thanks @crackwitz for the find tip.

Didn't libopencm3 fix this issue back in 2017? Going by this issue:
https://github.com/libopencm3/libopencm3/issues/828

I don't encounter this issue when compiling libopencm3 for other projects.

rz2222, 2019-05-27:
Hello. I have had the same issue (on Windows 7 64 bit). Here is what helped me:
1.	Had the issue with the “find”. Resolved according to note from crackwitz (https://github.com/blacksphere/blackmagic/issues/394#issuecomment-451149182)
2.	The compiler still missed a lot of “.h” files. Then I have noticed the “readme.md” file in “…/libopencm3” folder. It says:
_Download and install:
 - msys - http://sourceforge.net/projects/mingw/files/MSYS/Base/msys-core/msys-1.0.11/MSYS-1.0.11.exe
 - Python - http://www.python.org/ftp/python/2.7/python-2.7.msi (any 2.7 release)
 - arm-none-eabi/arm-elf toolchain (for example this one https://launchpad.net/gcc-arm-embedded)_
(Advise: When nothing helps, read the manual ;-))) ).
So I added the path to Python 2.7 to the system PATH. (I had it on my system already present), installed msys and performed the make from an msys command line window. My PATH starts now as follows: 
“PATH=C:\Python27;C:\Program Files (x86)\GnuWin32\bin;C:\arm\bin;C:\msys\1.0\bin;…”
From left to right: Python 2.7; Make; arm-none-eabi-gcc compiler and msys.
After these changes the compilation progressed much further, but again failed with missing “version.h” file.
3.	I have copied a “version.h” from Roger Clark’s repository with content:
#define FIRMWARE_VERSION ""
And, voala, success :).
Now, I have: “blackmagic_dfu.bin”,  “blackmagic.bin” and “blackmagic_dfu.hex”. The compilation failed at “blackmagic_dfu.hex” with the following error:
_make  OBJCOPY blackmagic_dfu.hex
c:\arm\bin\arm-none-eabi-objcopy.exe: blackmagic_dfu.hex 64-bit address 0x4b4fa3
08000000 out of range for Intel Hex file
c:\arm\bin\arm-none-eabi-objcopy.exe:blackmagic_dfu.hex: bad value
make: *** [blackmagic_dfu.hex] Error 1_ 
Never the less, I was able to upload the bin files to ST-Link clone and it seems it is working. I am still not able to debug my Maple Mini clone. I think it is due to the missing RST signal.
So my next plan is to install SWO and RST lines on the ST-Link clone replacing SWIM and RST (which is in fact an SWIM_RST).
I hope this will help to some of you.

Btw., does anybody know why I have got this error and how to get rid of it?

UweBonnes, 2019-05-27:
Do you have git installed? Git is needed for creating version.h.

SW

rz2222, 2019-05-27:
Yes, I do, and it is on path:
_$ git --version
git version 2.21.0.windows.1_

UweBonnes, 2019-05-27:
So investigate why make does not build version.h

sidprice, 2019-05-27:
This process for Windows build is a work in progress, however you may find this blog useful.
http://www.sidprice.com/2018/05/23/cortex-m-debugging-probe/

hierophect, 2019-05-28:
@rz2222 Your hex issue is due to a major bug in the arm-gcc 2018-q4 release. You'll need to revert to the q2 release if you want it to go away.

kousu, 2019-05-28:
@hierophect are you talking about the same bug as https://github.com/Homebrew/homebrew-cask/issues/56706 / https://bugs.launchpad.net/gcc-arm-embedded/+bug/1809919 ?

hierophect, 2019-05-28:
I believe so? I think it's a pretty well known issue. I've seen it referenced by Photon users and various other communities.

hierophect, 2019-05-28:
@kousu No, it's this one: https://bugs.launchpad.net/gcc-arm-embedded/+bug/1810274

It's pretty well known, and has caused issues in the Nordic/Photon/Mbed/Zephyr communities, etc
https://devzone.nordicsemi.com/f/nordic-q-a/43354/linker-error-address-out-of-range-for-intel-hex-files

kousu, 2019-05-28:
Interesting. I first ran into it when my toolchain disappeared from brew. I haven't had problems with the new version yet. Does anyone know what combination of options triggers it? The bug report I linked doesn't have a lot of detail and ARM hasn't responded to it yet.
-- 
kousu

hierophect, 2019-05-28:
I'm not sure what the trigger is. So far, it has occurred in every single compilation for me using that version, across Mbed and Cube MX projects on STM32. I don't use .hex files so it wasn't really a big deal, I reverted mostly just to get the ugly error out of the way.

rz2222, 2019-05-29:
@UweBonnes: Thanks for the hint. I am not from Unix/Linux world, so for me every step command is a Google question. At the moment I can live with the bin files I have got.

rz2222, 2019-05-29:
@sidprice: Thanks for the hint. Very useful. I installed cigwin and tried to make a compilation on a freshly cloned BMP repository. Unfortunately without success. Make returned a FIND error:
_zajo@CELSIUSH730 ~/blackmagic/libopencm3
$ make
FIND: Parameter format not correct
File not found - *.[ch]
  BUILD   lib/stm32/f0
/usr/bin/sh: -c: line 0: syntax error near unexpected token `('
/usr/bin/sh: -c: line 0: `C:/Program Files (x86)/GnuWin32/bin/make --directory=lib/stm32/f0 SRCLIBDIR=\C:/cygwin64/home/zajo/blackmagic/libopencm3/lib\\'
make: *** [lib/stm32/f0] Error 1_

sidprice, 2019-05-29:
@rz2222 , please read the blog carefully, it says that running that command WILL result in an error, however, do it anyway, then go on to the next step.

rz2222, 2019-05-29:
@sidprice: Thanks again for your patience. Well, the nvic.h file was not created in the …\blackmagic\libopencm3\include\libopencm3\stm32\xxx folders. So I assume the make failed differently in my case. (It also may be, because I could not execute the mount command successfully).  I have attached the output. 
[ErrorListing.txt](https://github.com/blacksphere/blackmagic/files/3233665/ErrorListing.txt)

sidprice, 2019-05-29:
If the mount failed why did you proceed before fixing it. Each step is required.

rz2222, 2019-05-29:
I do not understand, what shall I do with mount. What shall I replace the <project root> with. I cloned the BMP repository directly to cigwind drive in the cigwin windows. Is the mount needed anyway?

sidprice, 2019-05-29:
Are you running these commands in the CygWin Terminal? Mount is a built-in command of CygWin.

rz2222, 2019-05-29:
Yes.

sidprice, 2019-05-29:
OK, looking at your error file again it appears that you probably don't need to do the mount. The source is seen in the right place.
Please post the output when you run the "make clean" and "make" commands in the "blackmagic" directory. Maybe we can find why the nvic.h files are not being generated.

rz2222, 2019-05-29:
The output of the "make clean" and "make" commands in the "blackmagic" directory are at the very beginning of the attached "ErrorListing.txt" file: https://github.com/blacksphere/blackmagic/issues/394#issuecomment-497030408

sidprice, 2019-05-29:
OK, thanks. So, the build of BMP will not work until you get those commands to work.

execute make --version and post the results

sidprice, 2019-05-29:
And also "python --version"

rz2222, 2019-05-29:
zajo@CELSIUSH730 ~/blackmagic/src
$ make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-pc-mingw32

$ python --version
Python 2.7.11

sidprice, 2019-05-29:
I am using GNU Make 4.2.1, but I don't think that is the issue. You could try updating.

I think you are going to have to debug why the make clean fails, I have to leave my office so can't help more. make clean should give the following output:

$ make clean
  CLNHDR  include/libopencm3/efm32/g/irq.json
  CLNHDR  include/libopencm3/efm32/gg/irq.json
  CLNHDR  include/libopencm3/efm32/lg/irq.json
  CLNHDR  include/libopencm3/efm32/tg/irq.json
  CLNHDR  include/libopencm3/lm3s/irq.json
  CLNHDR  include/libopencm3/lpc13xx/irq.json
  CLNHDR  include/libopencm3/lpc17xx/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m0/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m4/irq.json
  CLNHDR  include/libopencm3/sam/3a/irq.json
  CLNHDR  include/libopencm3/sam/3n/irq.json
  CLNHDR  include/libopencm3/sam/3s/irq.json
  CLNHDR  include/libopencm3/sam/3u/irq.json
  CLNHDR  include/libopencm3/sam/3x/irq.json
  CLNHDR  include/libopencm3/sam/d/irq.json
  CLNHDR  include/libopencm3/stm32/f0/irq.json
  CLNHDR  include/libopencm3/stm32/f1/irq.json
  CLNHDR  include/libopencm3/stm32/f2/irq.json
  CLNHDR  include/libopencm3/stm32/f3/irq.json
  CLNHDR  include/libopencm3/stm32/f4/irq.json
  CLNHDR  include/libopencm3/stm32/f7/irq.json
  CLNHDR  include/libopencm3/stm32/l0/irq.json
  CLNHDR  include/libopencm3/stm32/l1/irq.json
  CLNHDR  include/libopencm3/stm32/l4/irq.json
  CLNHDR  include/libopencm3/vf6xx/irq.json
  CLEAN   lib/stm32/f0
  CLEAN   lib/stm32/f1
  CLEAN   lib/stm32/f2
  CLEAN   lib/stm32/f3
  CLEAN   lib/stm32/f4
  CLEAN   lib/stm32/f7
  CLEAN   lib/stm32/l0
  CLEAN   lib/stm32/l1
  CLEAN   lib/stm32/l4
  CLEAN   lib/lpc13xx
  CLEAN   lib/lpc17xx
  CLEAN   lib/lpc43xx/m4
  CLEAN   lib/lpc43xx/m0
  CLEAN   lib/lm3s
  CLEAN   lib/lm4f
  CLEAN   lib/efm32/tg
  CLEAN   lib/efm32/g
  CLEAN   lib/efm32/lg
  CLEAN   lib/efm32/gg
  CLEAN   lib/sam/3a
  CLEAN   lib/sam/3n
  CLEAN   lib/sam/3s
  CLEAN   lib/sam/3u
  CLEAN   lib/sam/3x
  CLEAN   lib/sam/d
  CLEAN   lib/vf6xx
  CLEAN   doc

Make has some debug options you can use to try to find the issue, type make --help for a list.

suibin0127, 2020-03-24:
i have some issue to build the BMP with Git Bash, details as following.
It looks some EOF miss matching  on "usr/bin/sh", it is a exe file
I think it is very common issue but sorry for that i am new to this build tool.
Dont know how to deal with this SH.exe.  Anybody can help me on this issue? thanks!

Administrator@SC-201912221028 MINGW64 /d/Git/blackmagic (master)
$ make PROBE_HOST=swlink
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f7/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/l4/irq.json
  GENHDR  include/libopencm3/stm32/g0/irq.json
  GENHDR  include/libopencm3/gd32/f1x0/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc17xx/irq.json
  GENHDR  include/libopencm3/lpc43xx/m4/irq.json
  GENHDR  include/libopencm3/lpc43xx/m0/irq.json
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/msp432/e4/irq.json
  GENHDR  include/libopencm3/efm32/tg/irq.json
  GENHDR  include/libopencm3/efm32/g/irq.json
  GENHDR  include/libopencm3/efm32/lg/irq.json
  GENHDR  include/libopencm3/efm32/gg/irq.json
  GENHDR  include/libopencm3/efm32/hg/irq.json
  GENHDR  include/libopencm3/efm32/wg/irq.json
  GENHDR  include/libopencm3/efm32/ezr32wg/irq.json
  GENHDR  include/libopencm3/sam/3a/irq.json
  GENHDR  include/libopencm3/sam/3n/irq.json
  GENHDR  include/libopencm3/sam/3s/irq.json
  GENHDR  include/libopencm3/sam/3u/irq.json
  GENHDR  include/libopencm3/sam/3x/irq.json
  GENHDR  include/libopencm3/sam/4l/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
  GENHDR  include/libopencm3/swm050/irq.json
  BUILD   lib/stm32/f0
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 2: syntax error: unexpected end of file
make[1]: *** [Makefile:71: lib/stm32/f0] Error 1
make: *** [Makefile:24: all] Error 2

Administrator@SC-201912221028 MINGW64 /d/Git/blackmagic (master)

UweBonnes, 2020-03-24:
Check that the git directory is clean. Then "make PROBE_HOST=swlink clean && make PROBE_HOST=swlink "

Please comment also on the reset circuit on the target board.

suibin0127, 2020-03-24:
> Check that the git directory is clean. Then "make PROBE_HOST=swlink clean && make PROBE_HOST=swlink "
> 
> Please comment also on the reset circuit on the target board.
reset circuit have sent to you by email last night, not sure you got it, i have attach 2 pictures
if not, let me resent to you again.

I dont know what is clean means, it was a clone git repos in local , i didnt change anything, just update the cortexm.c from #601 updates.

I also did the make clean, but clean also have this issue

Administrator@SC-201912221028 MINGW64 /d/Git/blackmagic (master)
$ make PROBE_HOST=swlink clean && make
  CLNHDR  include/libopencm3/stm32/f0/irq.json
  CLNHDR  include/libopencm3/stm32/f1/irq.json
  CLNHDR  include/libopencm3/stm32/f2/irq.json
  CLNHDR  include/libopencm3/stm32/f3/irq.json
  CLNHDR  include/libopencm3/stm32/f4/irq.json
  CLNHDR  include/libopencm3/stm32/f7/irq.json
  CLNHDR  include/libopencm3/stm32/l0/irq.json
  CLNHDR  include/libopencm3/stm32/l1/irq.json
  CLNHDR  include/libopencm3/stm32/l4/irq.json
  CLNHDR  include/libopencm3/stm32/g0/irq.json
  CLNHDR  include/libopencm3/gd32/f1x0/irq.json
  CLNHDR  include/libopencm3/lpc13xx/irq.json
  CLNHDR  include/libopencm3/lpc17xx/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m4/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m0/irq.json
  CLNHDR  include/libopencm3/lm3s/irq.json
  CLNHDR  include/libopencm3/msp432/e4/irq.json
  CLNHDR  include/libopencm3/efm32/tg/irq.json
  CLNHDR  include/libopencm3/efm32/g/irq.json
  CLNHDR  include/libopencm3/efm32/lg/irq.json
  CLNHDR  include/libopencm3/efm32/gg/irq.json
  CLNHDR  include/libopencm3/efm32/hg/irq.json
  CLNHDR  include/libopencm3/efm32/wg/irq.json
  CLNHDR  include/libopencm3/efm32/ezr32wg/irq.json
  CLNHDR  include/libopencm3/sam/3a/irq.json
  CLNHDR  include/libopencm3/sam/3n/irq.json
  CLNHDR  include/libopencm3/sam/3s/irq.json
  CLNHDR  include/libopencm3/sam/3u/irq.json
  CLNHDR  include/libopencm3/sam/3x/irq.json
  CLNHDR  include/libopencm3/sam/4l/irq.json
  CLNHDR  include/libopencm3/sam/d/irq.json
  CLNHDR  include/libopencm3/vf6xx/irq.json
  CLNHDR  include/libopencm3/swm050/irq.json
/usr/bin/sh: -c: line 2: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 4: syntax error: unexpected end of file
make[1]: *** [Makefile:87: lib/stm32/f0.clean] Error 1
make: *** [Makefile:34: clean] Error 2

Administrator@SC-201912221028 MINGW64 /d/Git/blackmagic (master)

suibin0127, 2020-03-24:
#414  the target circuit picture as below
[https://github.com/suibin0127/bmp/commit/0699acba85ed6f8669a3d20e09e487f64b737158](url)

suibin0127, 2020-03-24:
I can confirm it is quit clean now with git status. how to clear this issue?

Administrator@SC-201912221028 MINGW64 /d/Git/blackmagic (master)
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

Administrator@SC-201912221028 MINGW64 /d/Git/blackmagic (master)
$ make PROBE_HOST=swlink clean && make PROBE_HOST=swlink
  CLNHDR  include/libopencm3/stm32/f0/irq.json
  CLNHDR  include/libopencm3/stm32/f1/irq.json
  CLNHDR  include/libopencm3/stm32/f2/irq.json
  CLNHDR  include/libopencm3/stm32/f3/irq.json
  CLNHDR  include/libopencm3/stm32/f4/irq.json
  CLNHDR  include/libopencm3/stm32/f7/irq.json
  CLNHDR  include/libopencm3/stm32/l0/irq.json
  CLNHDR  include/libopencm3/stm32/l1/irq.json
  CLNHDR  include/libopencm3/stm32/l4/irq.json
  CLNHDR  include/libopencm3/stm32/g0/irq.json
  CLNHDR  include/libopencm3/gd32/f1x0/irq.json
  CLNHDR  include/libopencm3/lpc13xx/irq.json
  CLNHDR  include/libopencm3/lpc17xx/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m4/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m0/irq.json
  CLNHDR  include/libopencm3/lm3s/irq.json
  CLNHDR  include/libopencm3/msp432/e4/irq.json
  CLNHDR  include/libopencm3/efm32/tg/irq.json
  CLNHDR  include/libopencm3/efm32/g/irq.json
  CLNHDR  include/libopencm3/efm32/lg/irq.json
  CLNHDR  include/libopencm3/efm32/gg/irq.json
  CLNHDR  include/libopencm3/efm32/hg/irq.json
  CLNHDR  include/libopencm3/efm32/wg/irq.json
  CLNHDR  include/libopencm3/efm32/ezr32wg/irq.json
  CLNHDR  include/libopencm3/sam/3a/irq.json
  CLNHDR  include/libopencm3/sam/3n/irq.json
  CLNHDR  include/libopencm3/sam/3s/irq.json
  CLNHDR  include/libopencm3/sam/3u/irq.json
  CLNHDR  include/libopencm3/sam/3x/irq.json
  CLNHDR  include/libopencm3/sam/4l/irq.json
  CLNHDR  include/libopencm3/sam/d/irq.json
  CLNHDR  include/libopencm3/vf6xx/irq.json
  CLNHDR  include/libopencm3/swm050/irq.json
/usr/bin/sh: -c: line 2: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 4: syntax error: unexpected end of file
make[1]: *** [Makefile:87: lib/stm32/f0.clean] Error 1
make: *** [Makefile:34: clean] Error 2

UweBonnes, 2020-03-24:
It does not help if you repeat non relevant lines all over. Cut down and you ellipsis ("...") to not flood this channel. The error seems to be in libopencm3. Can you build libopencm3 stand alone? Can you build libopencm3 if you check it out somewhere else?

suibin0127, 2020-03-24:
i rebuilt libopencm3, it build fail same, but have the code **(3eff201a...)**
not sure any clue from this ?

Administrator@SC-201912221028 MINGW64 /d/Git/blackmagic/libopencm3 ((3eff201a...))
$ make
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f7/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/l4/irq.json
  GENHDR  include/libopencm3/stm32/g0/irq.json
  GENHDR  include/libopencm3/gd32/f1x0/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc17xx/irq.json
  GENHDR  include/libopencm3/lpc43xx/m4/irq.json
  GENHDR  include/libopencm3/lpc43xx/m0/irq.json
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/msp432/e4/irq.json
  GENHDR  include/libopencm3/efm32/tg/irq.json
  GENHDR  include/libopencm3/efm32/g/irq.json
  GENHDR  include/libopencm3/efm32/lg/irq.json
  GENHDR  include/libopencm3/efm32/gg/irq.json
  GENHDR  include/libopencm3/efm32/hg/irq.json
  GENHDR  include/libopencm3/efm32/wg/irq.json
  GENHDR  include/libopencm3/efm32/ezr32wg/irq.json
  GENHDR  include/libopencm3/sam/3a/irq.json
  GENHDR  include/libopencm3/sam/3n/irq.json
  GENHDR  include/libopencm3/sam/3s/irq.json
  GENHDR  include/libopencm3/sam/3u/irq.json
  GENHDR  include/libopencm3/sam/3x/irq.json
  GENHDR  include/libopencm3/sam/4l/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
  GENHDR  include/libopencm3/swm050/irq.json
  BUILD   lib/stm32/f0
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 2: syntax error: unexpected end of file
make: *** [Makefile:71: lib/stm32/f0] Error 1

Administrator@SC-201912221028 MINGW64 /d/Git/blackmagic/libopencm3 ((3eff201a...))

UweBonnes, 2020-03-24:
It does not help if you repeat non relevant lines all over. Cut down and you ellipsis ("...") to not flood this channel.  Please, handle accordingly!

What happens if you clone libopencm3 outside of BMP? Can you perhaps bisect? Anything to find on the libopencm3 github account?

suibin0127, 2020-03-24:
1. i copy libopencm3 from BMP folder to another directory ,and make file with Verbose.
I saw the infomation was fail at building library for STM32 families, from F0 to others one by one,
if i comment any of them.

  **BUILD   lib/stm32/f1
C:/Program Files/Git/mingw64/bin/make.exe --directory=lib/stm32/f1 SRCLIBDIR="D:/project/libopencm3/lib\" PREFIX="arm-none-eabi-" || \
        echo "Failure building: lib/stm32/f1: code: $?" > .stamp_failure_lib_stm32_f1
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 2: syntax error: unexpected end of file
make: *** [Makefile:71: lib/stm32/f1] Error 1**

2. i clone one new libopencm3 from github, and make file for it.
this time, the error is different, see below, looks like the windows path cut off by some space for 
GNU compiling tool ?  it generate lots of stamp failures for those STM32 lib with the failure code 127

**Administrator@SC-201912221028 MINGW64 /d/Git/libopencm3 (master)
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

Administrator@SC-201912221028 MINGW64 /d/Git/libopencm3 (master)
$ make V1=1
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f7/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/l4/irq.json
  GENHDR  include/libopencm3/stm32/g0/irq.json
  GENHDR  include/libopencm3/stm32/h7/irq.json
  GENHDR  include/libopencm3/gd32/f1x0/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc17xx/irq.json
  GENHDR  include/libopencm3/lpc43xx/m4/irq.json
  GENHDR  include/libopencm3/lpc43xx/m0/irq.json
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/msp432/e4/irq.json
  GENHDR  include/libopencm3/efm32/tg/irq.json
  GENHDR  include/libopencm3/efm32/g/irq.json
  GENHDR  include/libopencm3/efm32/lg/irq.json
  GENHDR  include/libopencm3/efm32/gg/irq.json
  GENHDR  include/libopencm3/efm32/hg/irq.json
  GENHDR  include/libopencm3/efm32/wg/irq.json
  GENHDR  include/libopencm3/efm32/ezr32wg/irq.json
  GENHDR  include/libopencm3/sam/3a/irq.json
  GENHDR  include/libopencm3/sam/3n/irq.json
  GENHDR  include/libopencm3/sam/3s/irq.json
  GENHDR  include/libopencm3/sam/3u/irq.json
  GENHDR  include/libopencm3/sam/3x/irq.json
  GENHDR  include/libopencm3/sam/4l/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
  GENHDR  include/libopencm3/swm050/irq.json
  GENHDR  include/libopencm3/pac55xx/irq.json
  BUILD   lib/stm32/f0
/usr/bin/sh: C:/Program: No such file or directory
  BUILD   lib/stm32/f1
/usr/bin/sh: C:/Program: No such file or directory
  BUILD   lib/stm32/f2
/usr/bin/sh: C:/Program: No such file or directory
  BUILD   lib/stm32/f3
/usr/bin/sh: C:/Program: No such file or directory
  BUILD   lib/stm32/f4
/usr/bin/sh: C:/Program: No such file or directory
  BUILD   lib/stm32/f7
[... 75 lines cut ...]
Failure building: lib/msp432/e4: code: 127
Failure building: lib/pac55xx: code: 127
Failure building: lib/sam/3a: code: 127
Failure building: lib/sam/3n: code: 127
Failure building: lib/sam/3s: code: 127
Failure building: lib/sam/3u: code: 127
Failure building: lib/sam/3x: code: 127
Failure building: lib/sam/4l: code: 127
Failure building: lib/sam/d: code: 127
Failure building: lib/stm32/f0: code: 127
Failure building: lib/stm32/f1: code: 127
Failure building: lib/stm32/f2: code: 127
Failure building: lib/stm32/f3: code: 127
Failure building: lib/stm32/f4: code: 127
Failure building: lib/stm32/f7: code: 127
Failure building: lib/stm32/g0: code: 127
Failure building: lib/stm32/h7: code: 127
Failure building: lib/stm32/l0: code: 127
Failure building: lib/stm32/l1: code: 127
Failure building: lib/stm32/l4: code: 127
Failure building: lib/swm050: code: 127
Failure building: lib/vf6xx: code: 127
make: *** [Makefile:75: lib] Error 1

Administrator@SC-201912221028 MINGW64 /d/Git/libopencm3 (master)**

UweBonnes, 2020-03-24:
Again: It does not help if you repeat non relevant lines all over. Cut down and you ellipsis ("...") to not flood this channel. Please, handle accordingly!

I meant you should clone https://github.com/libopencm3/libopencm3.

Have a look also at https://github.com/libopencm3/libopencm3/issues/1099

[15 more replies]

closed 2020-03-25 by UweBonnes