blackmagic-debug/blackmagic #404: Impossible to use BMP, SW-DP scan failed

benjaminpeere, 2018-10-25:
Hello,
I am having trouble into using Black magic probe with my stm32f103c8.
First of all I tried this tutorial : https://medium.com/@paramaggarwal/converting-an-stm32f103-board-to-a-black-magic-probe-c013cf2cc38c
to create my own black magic probe, but it didn't work.
I found some binaries files in this git : https://github.com/rogerclarkmelbourne/blackmagic_archive_version/tree/master/binaries
I have succeed to upload one in my stm32 this way : 
[image]
and it seems to work, I can use gdb, until I make this command : monitor swdp_scan
[image]
It's been 2 days now that I am on this error, I absolutely Don't understand what I am doing wrong.
If you have any idea you would help me a lot!
Thank you!

UweBonnes, 2018-10-25:
Why can you use GIT head? Read the README.md in src/platforms/swlink and compile for swlink.

benjaminpeere, 2018-10-25:
How do I compile for swlink?

esden, 2021-12-15:
Closing this issue as it is stale.

closed 2021-12-15 by esden