blackmagic-debug/blackmagic #406: TARGET LOST while flashing LPC824 minimal schematic

Squantor, 2018-11-10:
Hello all,

I am encountering a problem while trying to loading an program in a LPC824 microcontroller. I am trying to make this as exhaustive as possible so my apologies in advance for the long post.

### Hardware used
Official black magic probe V2.0 with a [LPC824xpresso board](https://www.embeddedartists.com/products/lpc824-max/), I have disabled the programmer on the xpresso board. 

I have also tried using this setup with a LPC-link2 and this works fine (upload and debugging), so my assumption is that hardware wise the SWD interface is okay. 

I have also used a clone blackmagic probe V2.1, just to rule out that the levelshifters and pull ups interfere. Same issue here.

### Software used
I am using head from this date (2019-11-20 GMT+1)
```
$ git rev-list --count HEAD
972
```
This has been compiled for the general target but have now flashed a version where ENABLE_DEBUG=1, output will be shown below.

### Problem description and output
When flashing a [trivial program](https://github.com/Squantor/templates_programming/tree/master/template_lpc824_c_cpp) blinky style program, I get the following result:
```
$ arm-none-eabi-gdb
GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC82x
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x00000596 in ?? ()
(gdb) set mem inaccessible-by-default off
(gdb) set {int}0x40048000 = 2
(gdb) load bin/debug/template_lpc824_c_cpp.axf
Error erasing flash with vFlashErase packet
(gdb) mon morse
TARGET LOST.
(gdb) 
```
The debug output via ```/dev/ttyACM1```:
```
*** Unsupported packet: vMustReplyEmpty
*** Unsupported packet: Hg0
*** Unsupported packet: qTStatus
 AP   0: IDR=04770031 CFG=00000000 BASE=f0000003 CSW=03000040
0x14000000: 0x0 <- does not match preamble (0xB105000D)
0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space)
-> cortexm_probe
0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace)
0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit)
*** Unsupported packet: qC
*** Unsupported packet: Hg0
*** Unsupported packet: qfThreadInfo
*** Unsupported packet: qL1200000000000000000
m packet: addr = 596, len = 4
m packet: addr = 596, len = 2
X packet: addr = 40048000, len = 0
X packet: addr = 40048000, len = 4
m packet: addr = 596, len = 4
m packet: addr = 596, len = 2
Flash Erase 00000000 00001000
Exception: SWDP invalid ACK
```
I am awaiting a minimum LPC824 board from [OSHPARK](https://oshpark.com/shared_projects/V5bGnU7j) so I test this board make sure it is not the hardware and see on a logic analyser what is happening.

If anybody is interested to help solve this issue, I am willing to ship one such board, fully assembled, to them, free of charge.

UweBonnes, 2018-11-10:
Do you have an FTDI MPSSE dongle? The libftdi platform uses the exact same programming algos like the rest of BMP. But you can run on the PC and single step and print debugging messages direct on the console. This helps much when hunting down problems with BMP.  I can help you in setting up some FTDI board you might eventually have. Otherwise I don't mind you sending me a board but with no promise I can find time to help you. And the 0.05 inch pitch of the debug connector makes adapting to my existing dongles a nightmare. The STLINK 4 pin 0.1 inch header layout VCC/SWCLK/GND/SWDIO does not need more space than you header, is probably cheaper in the BOM and is easier to connect. Maybe some of Esdens adapter might help.

Squantor, 2018-11-10:
Hi Uwe, 
I have these FT2232H minimodules, are these suitable? What are the connections from FTDI pin to get SWD working (JTAG cant be used to debug on the LPC8XX series). If you have a schematic of sorts that would even be better ;-).

Well, actually, that is a good idea! I will whip up a quick breakout for that if I have time today, or either tomorrow and put it with the board. I already made something similar for the [Bluepill](https://oshpark.com/shared_projects/kK12FxE3), but that one has a different pinout. Do you have a link to Esdens breakout?

In the mean time, if the FT2232H minimodules are suitable I will make the PC based BMP for debugging, I was planning to get serious in BMP development, best start this way.

UweBonnes, 2018-11-10:
Yes, FT2232H is suitable.  There is no external buffer on the  FTDI/CS(TMS)  line, so we can read and write on that line. Connect FTDI/CK(JTCK) to SWCLK and FTDI/CS(TMS) to SWDIO. Use "-c  "flossjtag" as cable. Using TDI/TDO for SWDIO would speed up things about 20 % (https://github.com/UweBonnes/blackmagic/commits/ftdi) but I will do a pull request only after my other pull request get fullfilled. I have not heard from Gareth for at least 3 weeks and Esden is also busy.

UweBonnes, 2018-11-10:
Does LPC824 programming need(!) SRST? Only my FTDI branch can handle SRST in a sensible way yet.

Look at https://1bitsquared.com/products/black-magic-probe for the adapter cables esden provided to me. I have no need for more.

For my hardware I normally use the Turtelizer http://www.ethernut.de/en/hardware/turtelizer/hardware.html JTAG connection on a 10-pin 2 row 0.1 connector.

Squantor, 2018-11-11:
@UweBonnes I have cobbled something together with a FT2232H module per your instructions and built the black magic probe software, but I am getting the following error:
```
Black Magic Probe (v1.6.1-224-g14bedcc-dirty)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

unable to open ftdi device: -3 (device not found)
```
I guess that this is more a problem of not finding the device, but when I plug the device in I do get this message in dmesg on Xubuntu 18.04:
```
[35103.762934] usb 1-4.1.1: new high-speed USB device number 19 using xhci_hcd
[35103.863460] usb 1-4.1.1: New USB device found, idVendor=0403, idProduct=6010
[35103.863467] usb 1-4.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[35103.863471] usb 1-4.1.1: Product: Dual RS232-HS
[35103.863475] usb 1-4.1.1: Manufacturer: FTDI
[35103.864405] ftdi_sio 1-4.1.1:1.0: FTDI USB Serial Device converter detected
[35103.864470] usb 1-4.1.1: Detected FT2232H
[35103.865541] usb 1-4.1.1: FTDI USB Serial Device converter now attached to ttyUSB0
[35103.865834] ftdi_sio 1-4.1.1:1.1: FTDI USB Serial Device converter detected
[35103.865901] usb 1-4.1.1: Detected FT2232H
[35103.866234] usb 1-4.1.1: FTDI USB Serial Device converter now attached to ttyUSB1
```
So the hardware is being detected by the OS and two serial ports appear. 
Okay, to tell you the thruth, these FT2232H modules where found in the electronic trash at work. I will try to get on monday some new, known good modules to make sure this is not an hardware issue. But do you have any ideas? I can open the ttyUSB0's just fine so I guess it is not an permission thing.

UweBonnes, 2018-11-11:
Please post the command line!

UweBonnes, 2018-11-11:
It looks like you missed the cable description.

Squantor, 2018-11-11:
Hi, the commandline is:
```
./blackmagic -c "flossjtag"
```
But I have a hardware issue. These modules have been reprogrammed to have a different VID/PID, now trying to reset those to original values so it gets detected properly. I could also reprogram the VID/PID in the black magic probe code. They seem to be programmed to imitate a Broadcom Wiced JTAG interface.

UweBonnes, 2018-11-11:
You can easily add a new cable in platform.c by duplicating the flossjtag entry and changing the entries accordingly.

Squantor, 2018-11-11:
Too bad, still the same problem:
```
$ ./blackmagic -c "flossjtag"

Black Magic Probe (v1.6.1-224-g14bedcc-dirty)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

unable to open ftdi device: -3 (device not found)
Aborted (core dumped)
```
lsusb output:
```
Bus 001 Device 037: ID 0403:6010 Future Technology Devices International, Ltd FT2232C Dual USB-UART/FIFO IC
```
dmesg output:
```
[41100.819380] usb 1-4.4: new high-speed USB device number 37 using xhci_hcd
[41100.920046] usb 1-4.4: New USB device found, idVendor=0403, idProduct=6010
[41100.920052] usb 1-4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[41100.920057] usb 1-4.4: Product: Dual RS232-HS
[41100.920061] usb 1-4.4: Manufacturer: FTDI
[41100.922693] ftdi_sio 1-4.4:1.0: FTDI USB Serial Device converter detected
[41100.922757] usb 1-4.4: Detected FT2232H
[41100.922969] usb 1-4.4: FTDI USB Serial Device converter now attached to ttyUSB0
[41100.923184] ftdi_sio 1-4.4:1.1: FTDI USB Serial Device converter detected
[41100.923234] usb 1-4.4: Detected FT2232H
[41100.925585] usb 1-4.4: FTDI USB Serial Device converter now attached to ttyUSB1
```
Going to sleep soon will continue when I have time. Maybe we are polluting this thread too much with this debugging stuff, shall we continue on gitter?

Squantor, 2018-12-23:
Hello All. 

My apologies for not checking in for such a long while. I was waiting for the PCB's from Oshpark to arrive so I can retest with a minimal LPC824 circuit. The problems that I encounter are on the lpc824 max board, using the external programming connector. This connector does work with a LPClink2 so there is still something funny going on, but I feel it is more electrical in nature then related to the BMP.

I have copied over the LPC824 schematic from the lpc824 max and stripped all the things not needed so I have truly a minimal circuit that has a LPC824 and some passives. Using the BMP v2.0 and the latest trunk I am able to access, program, verify and debugging the lpc824. This is with the probe supplying the power.

moshee, 2019-01-01:
I'm having a similar issue here with LPC824. Everything worked with my LPC-Link, but I had to switch to the BMP (v2.1) after I somehow managed to kill the former. I am unable to load binaries using the BMP, even with the latest firmware, with or without tpwr enabled.

GDB also seems to hang quite often when connecting to the BMP, still not entirely sure why or when—except that it doesn't hang when the probe isn't plugged into the target board. 

```
GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 7.12.1.20170215-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote /dev/cu.usbmodem7BB359AA
Remote debugging using /dev/cu.usbmodem7BB359AA
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout

<unplugged and replugged BMP here>

(gdb) target extended-remote /dev/cu.usbmodem7BB359AA
Remote debugging using /dev/cu.usbmodem7BB359AA
(gdb) mon v
Black Magic Probe (Firmware 14bedcc) (Hardware Version 3)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) mon tpwr enable
(gdb) mon s
Target voltage: 3.2V
Available Targets:
No. Att Driver
 1      LPC82x
(gdb) att 1
Attaching to program: /Users/moshee/projects/lpctest/.build/lpctest.elf, Remote target
__WFI () at /usr/local/include/lpc8xx/core_cmInstr.h:338
338	  __ASM volatile ("wfi");
(gdb) load
Error erasing flash with vFlashErase packet
(gdb) c
Continuing.

Program terminated with signal SIGLOST, Resource lost.
The program no longer exists.
```

UweBonnes, 2019-01-01:
Did you read other reports? Di you try staging?

moshee, 2019-01-01:
Sorry, without more context I'm not sure what either of those things mean. I looked through the GitHub issues and found that this matched my problem exactly, and it's still open. I don't know what staging refers to, is there another branch? I'm using latest firmware on git master from the build server.

UweBonnes, 2019-01-01:
With staging I mean https://github.com/UweBonnes/blackmagic/tree/staging.
However I think your problem is either with wiring, a hardware that draws excessive current from the BMP or a write protected device.

UweBonnes, 2019-01-02:
Another problem could be that your software remaps the JTAG/SWD pins.

esden, 2021-12-15:
Considering that the original authors issue was resolved or at least likely a hardware/signal integrity issue, I am closing this. If you have similar problems feel free to open a new issue and refer to this one.

closed 2021-12-15 by esden