blackmagic-debug/blackmagic #410: st link v2 stm32f030f4px

kingontheroad, 2018-11-18:
Hello

Iam interrest in one of my st link to flash with bmp , but support the stm32f030f4px bmp?

i have try to flash without succes , with stutility

kingontheroad, 2018-11-18:
Nice , iam sorry 

I have a laptop with windows10,  ccg is new for me to new.
So i have first st ultility used, erase ,and flashing everything goes fine .

But when i put the st in usb port , windows say i don,t regonise this usb device

esden, 2021-12-15:
@kingontheroad you should check your drivers and make sure it is not an issue on your Windows configuration. Considering that this issue is very old and stale I am closing this. If you still have trouble feel free to reopen this.

closed 2021-12-15 by esden