blackmagic-debug/blackmagic #413: hang after swd upload on swlink

jcw, 2018-11-25:
I'm running into an issue with the swlink build (latest). Context:

* h/w is blue pill
* 6 wires to target: gnd, 5v, swdio, swclk, rx/tx (b6/b7)
* target is either another blue pill, or an stm32f407 board
* using macos and platformio (both latest)

Upload (via platformio, set to blackmagic) works.
Serial console debug als works fine.

But after each upload, I have to _unpower_ the bmp + target to get the uploaded code to start. A plain reset of either bmp or target is not enough - it seems there is some sort of electrical latchup?

Any tips in how to chase this down further would be helpful.

tve, 2018-11-25:
Have you tried connecting with GDB and issuing a 'run' command?

jcw, 2018-11-25:
Aha, interesting! - just doing `mon swdp` already starts (the code).

jcw, 2018-11-25:
Behaviour is still as described above. But one more hint perhaps - when I upload again, the original code runs until the first interrupt, and then it looks like the new upload re-flashes and hangs again.

jcw, 2018-11-25:
Apologies to follow up all the time, but has something changed in the way uploads to BMP work?

This is currently what's happening with uploads: https://github.com/platformio/platform-ststm32/blob/develop/builder/main.py#L123-L132 - i.e. `mon swdp, attach 1, load, kill`.

I tried adding "run" (which starts ok, but never returns), or "detach", but it's not resolving this issue.

UweBonnes, 2018-11-25:
> But after each upload, I have to unpower the bmp + target to get the uploaded code to start. A plain reset >of either bmp or target is not enough - it seems there is some sort of electrical latchup?
This is ficed in #399. Pull request was sent Oct. 4

ivankravets, 2018-11-26:
@UweBonnes what is the right solution for this issue? See [our source code for BMP](https://github.com/platformio/platform-ststm32/blob/develop/builder/main.py#L123-L132).

P.S: @PlatformIO founder here.

UweBonnes, 2018-11-26:
Look around on my github account. Apply #399 and if you want to support more libftdi adapter also apply the ftdi branch. Only Garteth and Piotr have write access to the master repo. I have not heard from Gareth since about 6 weeks and about 1 week from Piotr. Piotr also seems very busy and has not applied any patch ever. I hope to hear from them soon. If they keep unresponsive, at some point I will consider forking.

ivankravets, 2018-11-29:
@UweBonnes I mean, could we make a patch on our side which will work for the non-patched BMP firmware? See @jcw solution. Is it OK?

UweBonnes, 2018-11-29:
Ah, jcw already proposed a workaround above. But gettinng BMP right would be the right solution.

RickKimball, 2018-12-02:
I came here to check for any outstanding issues as I encountered the same problem. I am using the latest code on a TI  tm4c123g launchpad. I resolved the problem doing this:
```
arm-none-eabi-gdb -ex 'set confirm off' -ex 'target extended-remote /dev/ttyACM0' -ex 'mon swd' -ex 'attach 1' -ex load -ex 'mon vector disable reset' -ex 'set *(unsigned *)(0xe000e000+0x0d00+0x00c) = ((0x05fa << 16) | 1<<2)' -ex quit blink.elf 
```

ivankravets, 2018-12-20:
So, I've finally found a time to try my BMP in pair with Blue PIll. I tested plain uploading, debugging and everything works great! My config for the macOS
```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = blackmagic
upload_port = /dev/cu.usbmodem7BB16EC1
debug_tool = blackmagic
debug_port = /dev/cu.usbmodem7BB16EC1
```

Maybe we have different versions for BMP? What is your? My is
```
(gdb) mon version
Black Magic Probe (Firmware v1.6.1) (Hardware Version 3)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
```

jcw, 2018-12-28:
I'm using the `swlink` branch, which is a BMP running **ON** a Blue Pill (independent of target choice).

I do have an (older) "real" BMP, but I tend to keep many BMP's around so I can leave them in-circuit.

The differences are unfortunate. Obviously, Black Sphere Technologies is entitled to do as they please and promote the real thing, with a shop, sales, revenues, praise, etc. But the fact that the BMP code also works on a Blue Pill makes it hard to resist the temptation to flash several of 'em for convenience.

jcw, 2019-05-22:
Gentle bump to ask if there is any progress on this? - I'm a bit lost right now as to which version of BMP is currently the best choice on a Blue Pill. I'm ok with fixing the 1.5 kΩ pull-up, if needed. There are more options, but for me, the BP is such a simple and convenient board to use for this sort of task ...

UweBonnes, 2019-05-22:
This PR has been hickjacked by other problems so it is hard to decipher what your setup is. I think, a F407 and an Bluepill/swlink is involved. You seem to flash the device and than to hard disconnect without issuing a quit run or detach. Test ypur procedures in pure gdb. "tar ext ...; mon s; att 1; load; quit" and confirm that the problem persists with that sequence.

jcw, 2019-05-22:
Hm, I see what you mean. Ok, so here's an upload to an F407, using a BP flashed with BMP:

```
$ arm-none-eabi-gdb .pioenvs/f407/firmware.elf
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git
[...]
(No debugging symbols found in .pioenvs/f407/firmware.elf)
(gdb) tar ext /dev/cu.usbmodemDFEAA5E71
Remote debugging using /dev/cu.usbmodemDFEAA5E71
(gdb) mon s
Target voltage: ABSENT!
Available Targets:
No. Att Driver
 1      STM32F40x
(gdb) at 1
Attaching to program: /Users/jcw/Code/jcw/stm32x/diymore-f407vg/blink/.pioenvs/f407/firmware.elf, Remote target
0x08000444 in enableSysTick(unsigned long)::{lambda()#1}::_FUN() ()
(gdb) load
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x5b4 lma 0x8000188
Loading section .rodata, size 0x74 lma 0x800073c
Loading section .init_array, size 0x4 lma 0x80007b0
Loading section .fini_array, size 0x4 lma 0x80007b4
Start address 0x8000630, load size 1976
Transfer rate: 8 KB/sec, 329 bytes/write.
(gdb) quit
A debugging session is active.

	Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) y
Detaching from program: /Users/jcw/Code/jcw/stm32x/diymore-f407vg/blink/.pioenvs/f407/firmware.elf, Remote target
[Inferior 1 (Remote target) detached]
```

At this point, the uploaded blink/uart test is not running. Target reset button does not work. Power cycling the target does. The behaviour I'm used to is that detach or quit will leave the target running.

UweBonnes, 2019-05-22:
"tar ext ...; mon s; att 1; load; quit"  leaves me hanging too. Running "tar ext ...; mon s; att 1; load; run ; ^C, quit"  gets me going. On detach, https://github.com/UweBonnes/blackmagic/tree/detach always sets the CPU to running. Please comment.

jcw, 2019-05-22:
Detach iso quit is not enough. Run will work, but then ^C stops it (of course), and I'm back to not being able to leave gdb with the target running. I'm going to try you latest commit now.

jcw, 2019-05-22:
Ok, I can confirm that your "detach" branch works perfectly on a freshly-assembled Blue Pill, re-using the SWD pins as described in the `swlink/README.md`.

I hope this change can end up in the master branch, and eventually in the BMP master repo - because to be honest, the number of forks and branches has had me running around pulling hairs for a while ...

jcw, 2019-05-22:
A _biiiiig_  thank you, Uwe!

PR #477 (2019-05-23): gdb_main: Always release CPU when detaching.
Fixes #413, was PR #477
closed 2019-05-23 by UweBonnes