blackmagic-debug/blackmagic #416: USB config error Docker-Machine / VirtualBox

lmapii, 2018-12-04:
Hello,

I've just received my BMP which I want to use for batch-programming in production. The programming software (includes end-of-line test) is running within docker, which is working finde for comparable openocd debuggers (e.g., olimex).

While the BMP shows up correctly on the host and is working just fine there it is not recognized within the virtual-machine used by docker: This is essentially simply a virtualbox image running linux:
`Linux default 4.4.119-boot2docker #1 SMP Wed Feb 28 17:44:37 UTC 2018 x86_64 GNU/Linux`

I get the following error message from the box:
```
usb 1-1: new full-speed USB device number 5 using xhci_hcd
usb 1-1: ep 0x82 - rounding interval to 1024 microframes, ep desc says 2040 microframes
usb 1-1: ep 0x84 - rounding interval to 1024 microframes, ep desc says 2040 microframes
usb 1-1: can't set config #1, error -110
```

If I connect the device to a normal linux machine I don't get this error. Is there anything I can do to make this work within the VirtualBox?

Thanks and BR,
Martin

UweBonnes, 2018-12-04:
Did you google for that error? Maybe you run on a USB 3 port and your docker linux can't handle it?

lmapii, 2018-12-04:
Yes I did google it. The only reference I've found is that this error is typically generated if the USB port doesn't provide enough current. In VirtualBox I've enabled the USB 3.0 hub and so far it was never complaining. 

I am currently setting up different versions of the `boot2docker` images. Sadly the latest one is broken (cannot even get normal USB devices within the host), so I'm back to setting up the machines release by release. I'll let you know If I'm lucky.

UweBonnes, 2018-12-04:
Does the host have a USB2 host? Perhaps try that port.

But I don't think it is a BMP issue. So perhaps close the issue.

lmapii, 2018-12-04:
The host has USB 3.0 yes. This is definitely a docker-machine / boot2docker image issue. A few hours and something that felt like 200 virtual-machines later I found an image that lists the BMP. For anybody who might run into the same issue:

```
$ docker --version
Docker version 18.09.0, build 4d60db4
$ docker-machine --version
docker-machine version 0.16.0, build 702c267f
$ docker@default:~$ uname -a
Linux default 4.9.93-boot2docker #1 SMP Thu Jul 19 18:29:50 UTC 2018 x86_64 GNU/Linux
```

Used release:
https://github.com/boot2docker/boot2docker/releases/tag/v18.06.1-ce

Created using
```
$ docker-machine create -d virtualbox --virtualbox-boot2docker-url https://github.com/boot2docker/boot2docker/releases/download/v18.06.1-ce/boot2docker.iso default
```

VirtualBox Version 5.2.22 r126460

alexwhittemore, 2019-03-03:
For anyone who comes here looking, I'm running into the same issue using macOS in a Parallels VM (trying to get my configuration functional before doing it on my actual laptop, and making a mess of the whole neighborhood in the process of getting it wrong). Connected to the host, BMP works fine. Patched through to the guest, the device appears in system info, but in console.app I see 

`AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002d6)`

and no serial ports show up.

Still don't have a resolution.

alexwhittemore, 2019-03-15:
Just to eliminate variables, I picked up a 1BitSquared version BMP and tried that - enumerates reliably and mounts serial devices at /dev/cu.* on the macOS host; enumerates but fails to mount ports on guest, or when letting Parallels hand-over control (i.e. if I 'disconnect' the device from the guest with software, without replugging). Seems like maybe there's some kind of timing issue.

yoberi, 2019-11-22:
Adding a comment to a closed issue is probably not the right approach, yet here I am.

I'm also trying to get the BMP working inside a guest vm, with no luck. I can pass through a device with the STLink2-1 firmware into a guest and use as expected (both debugger and cdc UART), but I can not get the BMP 2.1 device to work. I have used the same BMP device on an install of the same linux OS running on baremetal and all is working as expected. But when passing it into a guest VM with vmware, the CDC devices enumerate but don't work as needed. So, I'm seeing the same issue that @alexwhittemore has observed above and [here](https://community.platformio.org/t/debugging-with-black-magic-probe-on-vscode-macos/6795/7).

Please consider reopening. Feel free to reach out for more details.

UweBonnes, 2019-11-22:
Wild guessing. BMP with delivery firmware is far behing git head. Can you try to upgrade to git head and retry?

yoberi, 2019-11-25:
Hi @UweBonnes I tested with both the installed release tag and the latest 8a07f44 before posting, with the same results.

esden, 2021-12-15:
Is this still an issue? Does it need additional testers? Or can we close this, as this issue is becoming a bit stale.

alexwhittemore, 2021-12-15:
I never resolved the problem, myself. I just switched to a different debugger for that project and haven't had occasion to come back. At the time, I was quite certain my firmware was latest.

esden, 2021-12-15:
Ok I have vmware that I can test it on. I will see if I can get it to work.

lmapii, 2021-12-16:
`docker-machine` is now deprecated so the original issue I had with does no longer apply.

esden, 2021-12-16:
Ok thank you very much for the update. In that case I will close this issue. If some other docker system or vmware needs to be investigated we should open dedicated issues for the specific platform so we can focus on the specific problem. :)

closed 2021-12-16 by esden

lmapii, 2021-12-16:
Yep, in case of trouble I will do that. Thanks!