blackmagic-debug/blackmagic #417: STM32F205: BMP may create and report to gdb a faulty memory map

DerMeisteRR, 2018-12-13:
Hi, blacksphere!
I have flashed an STM32F103C8 with the BMP firmware and am quite fascinated by your project!
When trying to flash and debug a STM32F205RB with the BMP, i ran into some problems:
GDB connects to the BMP, discovers the target and attaches to it, as expected. But when i try to do "(gdb) load" gdb freezes. I run Ubuntu and have to use monitor to kill gdb in order to recover.
I had a closer look at what BMP does when attaching to the STM32F205 and noticed that it creates a strange - looking memory map: after discovering and attaching, the command "(gdb) info mem" reveals:
Num Enb Low Addr   High Addr  Attrs 
0   y  	0x08000000 0x08010000 flash blocksize 0x4000 nocache 
1   y  	0x08010000 0x08020000 flash blocksize 0x10000 nocache 
2   y  	0x08020000 0x08020000 flash blocksize 0x10000 nocache 
3   y  	0x20000000 0x20050000 rw nocache 
This is strange because my Chip has 128 kByte Flash and  64k RAM. Not 320k RAM! 
Also, observe that the reported end address of bank 2 is the same as its start address. I think this is what crashes gdb.

In the BMP code, i found in target/stm32f4.c the following code in static bool stm32f4_attach(target *t):
...
	if (is_f7) {
....
		}
	} else {
		if (has_ccmram)
			target_add_ram(t, 0x10000000, 0x10000); /* 64 k CCM Ram*/
		target_add_ram(t, 0x20000000, 0x50000);     /* 320 k RAM */

This causes BMP to wrongly report 320k RAM for the STM32F205. As a hot fix, i changed the code thus:
	if (is_f7) {
		target_add_ram(t, 0x00000000, 0x4000);  /* 16 k ITCM Ram */
		target_add_ram(t, 0x20000000, 0x20000); /* 128 k DTCM Ram */
		target_add_ram(t, 0x20020000, 0x60000); /* 384 k Ram */
		if (dual_bank) {
			uint32_t optcr;
			optcr = target_mem_read32(t, FLASH_OPTCR);
			use_dual_bank =  !(optcr & FLASH_OPTCR_nDBANK);
		}
	} else {
		if (has_ccmram)
			target_add_ram(t, 0x10000000, 0x10000); /* 64 k CCM Ram*/

		/* check if this is a STM32F205 */
		if (t->idcode == ID_STM32F20X){
			DEBUG("flash size is: %ld kByte. \n",flashsize);
			if(flashsize > 256){
				/*This chip has 128 k RAM */
				target_add_ram(t, 0x20000000, 0x20000);
			}
			else if (flashsize > 128){
				/* This chip has 96 k RAM */
				target_add_ram(t,0x20000000, 96*1024);
			}
			else{
				/* This chip has 64 k RAM */
				target_add_ram(t,0x20000000, 64*1024);
			}
		}
		else {
			target_add_ram(t, 0x20000000, 0x50000);     /* 320 k RAM */
		}
This is probably very crude, but lets the BMP and GDB report the correct RAM size - for my chip, that is 64kB RAM.
Regarding the reported Flash Bank with size 0, In the same file in the same function, there is this code beginning with line 322:
`	} else {
		uint32_t remains = banksize - 0x20000; /* 128 k in small sectors.*/
		if (is_f7) {
			stm32f4_add_flash(t, ITCM_BASE, 0x10000,  0x4000,  0, split);
			stm32f4_add_flash(t, 0x0210000, 0x10000, 0x10000,  4, split);
			stm32f4_add_flash(t, 0x0220000, remains, 0x20000,  5, split);
		}
		stm32f4_add_flash(t, 0x8000000, 0x10000,  0x4000,  0, split);
		stm32f4_add_flash(t, 0x8010000, 0x10000, 0x10000,  4, split);
		stm32f4_add_flash(t, 0x8020000, remains, 0x20000,  5, split);
Since my chip has 128kB Ram, "remains" has the value 0. This makes BMP report the strange flash bank in the  memory map.
I changed it thus:
	} else {
		uint32_t remains = banksize - 0x20000; /* 128 k in small sectors.*/
		if (is_f7) {
			stm32f4_add_flash(t, ITCM_BASE, 0x10000,  0x4000,  0, split);
			stm32f4_add_flash(t, 0x0210000, 0x10000, 0x10000,  4, split);
			stm32f4_add_flash(t, 0x0220000, remains, 0x20000,  5, split);
		}
		if(t->idcode==ID_STM32F20X){
			DEBUG("adding STM32F2 flash banks to memory table. Banksize is: %ld.\n",banksize);
/*
			Flash Layout of STM32F2 according to RM0033 Rev. 8
			Name Block base addresses Size
			Sector 0 0x0800 0000 - 0x0800 3FFF 16 Kbyte
			Sector 1 0x0800 4000 - 0x0800 7FFF 16 Kbyte
			Sector 2 0x0800 8000 - 0x0800 BFFF 16 Kbyte
			Sector 3 0x0800 C000 - 0x0800 FFFF 16 Kbyte
			Sector 4 0x0801 0000 - 0x0801 FFFF 64 Kbyte
			Sector 5 ... 	...		...		..128 Kbyte
			...
*/
			stm32f4_add_flash(t, 0x8000000, 0x10000,  0x4000,  0, split);
			stm32f4_add_flash(t, 0x8010000, 0x10000, 0x10000,  4, split);
			/* STM32F2 may have more or less than 128 k Flash. */
			if (remains > 0){
				stm32f4_add_flash(t, 0x8020000, remains, 0x20000,  5, split);
			}
		}
		else{

			/* use flash memory map for STM32F4.. */
			stm32f4_add_flash(t, 0x8000000, 0x10000,  0x4000,  0, split);
			stm32f4_add_flash(t, 0x8010000, 0x10000, 0x10000,  4, split);
			stm32f4_add_flash(t, 0x8020000, remains, 0x20000,  5, split);
		}`

This is probably overkill - i think it might suffice to always check the value of "remains" independent of the target.
May i ask that you kiindly share your thoughts on this with me? I will gladly help to create a nicer, more efficient fix for this if so desired. For completeness, i have attached my modified stm32f4.c to this issue report.

best regards,
Jochen

[stm32f4.zip](https://github.com/blacksphere/blackmagic/files/2676504/stm32f4.zip)

DerMeisteRR, 2018-12-13:
Hi, Again!

I have improved the fixes i suggested above and attached the stm32f4.c file, again.
With the attached file, i built and ran the BMP firmware. It correctly detects the STM32F2, reports a correct memory map for my target (stm32f2RB) and it erases the flash. loading the program took two attempts, though.

please let me know your thoughts on the new approach to fix this issue.

cheers,
Jochen

[stm32f4_more_integrated_fix.zip](https://github.com/blacksphere/blackmagic/files/2676640/stm32f4_more_integrated_fix.zip)

UweBonnes, 2018-12-13:
There is no need to save flash by omitting the bootloader. STM32F103x8 and STM32F103xB are the same chip. And there has been no report so long that flashing above 64k on a STM32F103x8 failed. Use scripts/stm32mem to flash via the boot loader and it will check what was written.

For reporting RAM, we think that is is no error to report more. That spares us from a lot of code guessing what RAM is available.

stm32f4_add_flash() should return without creating a flash target, That should resolve your wrong reported target.
  <
diff --git a/src/target/stm32f4.c b/src/target/stm32f4.c
index a7848fc..12d5153 100644
--- a/src/target/stm32f4.c
+++ b/src/target/stm32f4.c
@@ -134,6 +134,7 @@ static void stm32f4_add_flash(target *t,
                               uint32_t addr, size_t length, size_t blocksize,
                               unsigned int base_sector, int split)
 {
+       if (length == 0) return;
        struct stm32f4_flash *sf = calloc(1, sizeof(*sf));
        struct target_flash *f = &sf->f;
        f->start = addr;
>
 Please test and report. But again, STM32F20XxB and STM32F20XxC are the same chip and the flash above 128 k was produced, but perhaps disabled by some fuse, not production tested or  tested bad in production

Your problem seems to be caused by something different. Did you set the right parallelism for your target voltage? Do other programmers succeed?

DerMeisteRR, 2018-12-13:
Hi, Uwe!
Thank you for the hint regarding the statistics with STM32F103x8 chips. I will try the script you pointed out gladly - that would allow me to re-include the targets i had to exclude from build to reduce fw size to < 64kB.
Regarding RAM: reporting more than the chip acutally has will probably not interfere with flashing and debugging but will confuse users (like me) that query the amount of RAM with gdb and BMP. Is there a hint in the documentation that i missed regarding the "oversized" RAM bank reported?
I have just tested the fix in stm32f4_add_flash() that you suggested. It works: GDB reports a correct memory table and i can erase flash, load a program etc.
Buut.. there is one STM32F4, namely the STM32F410, that comes with 64kB RAM. Never mind if the actual chip has more flash - it will report 64kB. This may become a problem in the future when somebody wants to add support for this chip: The calculation "remains = banksize - 128*1024" will return some veeery large number because uint32_t remains cannot hold negative numbers.
Regarding the STM32F205xB having more actual flash: for the issue i reported, the acutal flash size of the target is irrelevant because the setup of the memory map bases on the flash size the chip _reports_.

The problem i noticed was indeed caused by the flash bank with size 0 in the memory map: by compiling the code this or that way, i can "switch on and off" if GDB crashes when i try to use `(gdb) load` to flash the target. Also i was able to program the very same target with a ARM-USB-TINY-H and OpenOCD or the "st-link" on a value-line discovery board and OpenOCD without problems. Nevertheless, your hint regarding target voltage is well-noted because i did note some issues with the power supply when i started this little project: i must plug USB into the target eval board or BMP looses the SWD connection erratically. Apparently, the voltage controller on the st-link clone i turned into a BMP is not strong enough for both chips.

UweBonnes, 2018-12-13:
Add another test 
<
diff --git a/src/target/stm32f4.c b/src/target/stm32f4.c
index a7848fc..3974256 100644
--- a/src/target/stm32f4.c
+++ b/src/target/stm32f4.c
@@ -134,6 +134,7 @@ static void stm32f4_add_flash(target *t,
                               uint32_t addr, size_t length, size_t blocksize,
                               unsigned int base_sector, int split)
 {
+       if (length == 0) return;
        struct stm32f4_flash *sf = calloc(1, sizeof(*sf));
        struct target_flash *f = &sf->f;
        f->start = addr;
@@ -320,7 +321,9 @@ static bool stm32f4_attach(target *t)
                stm32f4_add_flash(t, 0x8020000, 0x20000, 0x20000, 4, split);
                stm32f4_add_flash(t, 0x8040000, remains, 0x40000, 5, split);
        } else {
-               uint32_t remains = banksize - 0x20000; /* 128 k in small sectors.*/
+               uint32_t remains = 0;
+               if (banksize > 0x20000)
+                       remains = banksize - 0x20000; /* 128 k in small sectors.*/
                if (is_f7) {
                        stm32f4_add_flash(t, ITCM_BASE, 0x10000,  0x4000,  0, split);
                        stm32f4_add_flash(t, 0x0210000, 0x10000, 0x10000,  4, split);
>

UweBonnes, 2018-12-13:
Argh, how to post a diff?

DerMeisteRR, 2018-12-13:
Hmm, maybe put it into a text file and attach? - or post the new code instead of a diff. I can merge that into my stuff here for testing.

UweBonnes, 2018-12-13:
Instead of 
 uint32_t remains = banksize - 0x20000; /* 128 k in small sectors.*/ 
do
 uint32_t remains = 0;
 if (banksize > 0x20000) remains = banksize - 0x20000; /* 128 k in small sectors.*/

DerMeisteRR, 2018-12-13:
ok, i just tested this and it seems to work: BMP finds the STM32F205RB, attaches to it and i can run and stop the program.
Regarding the python flash script: for some strange reason, i see erase timeouts when i try it. Had the same problem with dfu-util. Do you have any suggestions?

DerMeisteRR, 2018-12-13:
This is what happens:

> dermeisterr@i7-4790K:~/ARM-Env/blackmagicprobe/scripts$ python stm32_mem.py ../src/blackmagic.bin -a 0x8002000

USB Device Firmware Upgrade - Host Utility -- version 1.2
Copyright (C) 2011  Black Sphere Technologies
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

Device ID:	 1d50:6018
Manufacturer:	 Black Sphere Technologies
Product:	 Black Magic Probe (STLINK), (Firmware v1.6.1-224-g14bedcc-dirty)
Serial:		 C0BFB1F2
Failed to read device state! Assuming APP_IDLE
Invoking DFU Device
Device ID:	 1d50:6017
Manufacturer:	 Black Sphere Technologies
Product:	 Black Magic (Upgrade) for STLink/Discovery, (Firmware v1.6.1-224-g14bedcc-dirty)
Serial:		 C0BFB1F2
Programming memory at 0x08002000
Erase Timed out

All operations complete!

dermeisterr@i7-4790K:~/ARM-Env/blackmagicprobe/scripts$

DerMeisteRR, 2018-12-13:
And this when i tried again - but before the error list appeared, the script made nice progress with programming some 3 or 4 kBytes of blackmagic.bin.

> dermeisterr@i7-4790K:~/ARM-Env/blackmagicprobe/scripts$ python stm32_mem.py ../src/blackmagic.bin -a 0x8002000

USB Device Firmware Upgrade - Host Utility -- version 1.2
Copyright (C) 2011  Black Sphere Technologies
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

Device ID:	 1d50:6017
Manufacturer:	 Black Sphere Technologies
Product:	 Black Magic (Upgrade) for STLink/Discovery, (Firmware v1.6.1-224-g14bedcc-dirty)
Serial:		 C0BFB1F2
Traceback (most recent call last):
  File "stm32_mem.py", line 223, in <module>
    stm32_write(dfudev, bin[:1024])
  File "stm32_mem.py", line 56, in stm32_write
    status = dev.get_status()
  File "/home/dermeisterr/ARM-Env/blackmagicprobe/scripts/dfu.py", line 107, in get_status
    6, index=self.index) 
  File "/usr/local/lib/python2.7/dist-packages/usb/legacy.py", line 211, in controlMsg
    timeout = timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
dermeisterr@i7-4790K:~/ARM-Env/blackmagicprobe/scripts$

UweBonnes, 2018-12-13:
Some  "-a xxxx" argument is for special cases. I did not test.  

Check if there are some strange option bits set in your host cpu. Check with another CPU. Check that you address the right CPU, so unplug all similar devices from USB...

DerMeisteRR, 2018-12-13:
Ok, good hint! I had been somewhat suspicious of the USB hardware of my red stlink clone. Just took a closer look (literally) and noticed that there are two 50 Ohm series resistors in the signal path of USB D+ and USB D- from the USB plug to the MCU. I replaced them by 10 Ohm resistors and now the python script programs the device like a charm. Thnx for making this available! :+1:

DerMeisteRR, 2018-12-13:
Good to know! I will check right away.
My STM32F103C8 host did flash and verify 78500 bytes - i have support for all the targets, now. Thanks for pointing this out to me! 
Regarding the changes to the code we identified together - will they go into the repository? If so, how? (meaning: Do you commit them, or do you want me to do this somehow? And if you want me to, how do i do that?) Sorry for asking such noob-ish questions: i have much more experience with SVN than git due to my employer wanting me to use svn.

DerMeisteRR, 2018-12-13:
Just checked the pullup on USB D+: Both the Discovery VL and the steval-pcc009v2 schematics show 1.5k but on the red stlink clone is 2.7k. Works though, so for now i will not change the resistor.

UweBonnes, 2018-12-13:
Onlu Gareth and Esden have repo access. Esden did no check-in until now and is busy until end of the year. No contact to Gareth since more than two weeks. I will put the change into staging in https://github.com/UweBonnes/blackmagic/branches

DerMeisteRR, 2018-12-13:
Ok, thank you! I will take a peek every now and then - i am really happy i was able to contribute something and i have learned muuuch. Thank you! At the moment, i have the STM32F205RB and a STM32F100RB target at hand. Let me know if you want me to test something.

UweBonnes, 2018-12-13:
Just for curiosity: Fake BMP to accept the STM32F205RB as a STM32F205RC and try to programm above 128k . I don't have any such device.

DerMeisteRR, 2018-12-14:
Hey, Uwe!
After a good night's sleep, i have now faked the BMP to think my STM32F205RB has 256 kB Flash by this:
`		uint32_t remains = 0;`
`		if (banksize > 0x20000) remains = banksize - 0x20000; /* 128 k in small sectors.*/`
`		remains = 128 * 1024;`
I have flashed this to the BMP and will test in a moment what happens when i try to flash a bmp image into the controller, read it back again and compare the two pictures.

DerMeisteRR, 2018-12-14:
GDB reports the following memory map, as expected:
`Using memory regions provided by the target.`
`Num Enb Low Addr   High Addr  Attrs `
`0   y  	0x08000000 0x08010000 flash blocksize 0x4000 nocache `
`1   y  	0x08010000 0x08020000 flash blocksize 0x10000 nocache `
`2   y  	0x08020000 0x08040000 flash blocksize 0x20000 nocache `
`3   y  	0x20000000 0x20050000 rw nocache `

DerMeisteRR, 2018-12-14:
Ok, i have tried for some hours now to make gdb write a picture of 255 kByte size into the target's memory, but no luck. How do i make gdb write a binary? `(gdb) restore` did not work.

DerMeisteRR, 2018-12-14:
Got it! :) I used arm objcopy to convert the .bmp into an object file:
`arm-none-eabi-objcopy -O elf32-littlearm -B arm --rename-section .data=.text -I binary board.bmp board.o`
Then i told GDB to load this file. Since the object file is relative, i need to tell gdb explicitly where to load the .text section in the file:
`(gdb) load ~/ARM-Env/blackmagicprobe/pictures_PCB/board.o 0x8000000`
Then i used "dump" to read the content of the flash:
`(gdb) dump binary memory ~/ARM-Env/blackmagicprobe/pictures_PCB/readback2.bmp 0x08000000 0x0803de06`
I stole the end adress from the gdb's response to the load command, btw.
The result: a nice b/w picture of a flower. Thus, my STM32F205RB has 256 kB of working flash - cooool! :) :)

[pictures.zip](https://github.com/blacksphere/blackmagic/files/2680246/pictures.zip)

DerMeisteRR, 2018-12-14:
One thought: How do you think about adding a target-specific command to BMP that allows to fake-set the flash-size? something like "monitor flashsize_fake <size in kb>"?

UweBonnes, 2018-12-14:
For comparing in gdb, use "compare-section", see also the BMP wiki and hints pages.

Probably using the flashsize register is no good idea at all . The compiler and linker has to care for available memory, BMP should only set up regions in a way that gdb can handle. Maybe we shoud use the device specific maximum sizes and not the flashsize register sizes.

So it looks like also the F205xb has flash above 128 k not somehow disabled. Good find.

DerMeisteRR, 2018-12-15:
Hi, Uwe! 
Maybe i was somewhat unprecise in my previous post - i do not suggest to (ab)use the flashsize register in the target. 
What i had on mind was: create a target-specific command like `(gdb) monitor flashsize_fake <fake-size>` that lets BMP create a memory map that shows not what the device itself has reported but a memory map for that device but according to what the command parameter was. My thoughts for this:
-delete the existing memory table
-create a new memory map following the flash bank setup of the target device family and for the flash size given with the command
However, i understand your  that this is making trouble in terms of coordinating with GDB. Is it possible for the BMP to force GDB to detach from the target?

UweBonnes, 2019-02-21:
Fixed with #428

closed 2019-02-21 by UweBonnes