blackmagic-debug/blackmagic #419: Process of programming EBYTE NRF52 devices

navado, 2018-12-17:
I open this issue as some support request. Not sure the root cause of the problem, probably just documentation. I promise to write wiki section with results :)

The setup consists of BMP 2.1 `Black Magic Probe (Firmware v1.6.1-224-g14bedcc) (Hardware Version 3)` as programming device and NRF52840 (on EBYTE E73 board) as a target.

1.  there was a problem to scan for device. The root cause was enabling srst `mon assert_srst scan`. After this configuration the device is recognized.
```
(gdb) mon swdp_scan
Target voltage: 3.3V
Ignoring packet error, continuing...
Available Targets:
Available Targets:
Available Targets:
Available Targets:
No. Att Driver
No. Att Driver
 1      ARM Cortex-M
 2      Nordic nRF52 Access Port
 3      ARM Cortex-M
```
There is still suspicious message `Ignoring packet error, continuing...`. Bonus question - how to look for the problematic message?
2. There is a problem to attach to device.
2.1. 
```
(gdb) attach 1
Attaching to Remote target
Attaching to Remote target failed
```
2.2. This seems like succeded, but it looks like not a debug port
```
(gdb) attach 2
Attaching to Remote target
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring
PC not available
PC register is not available
PC not available
```
3.1. Loading firmware. For target 1 and 3 the result is the same
```
$ arm-none-eabi-gdb -nx --batch \
-ex 'target extended-remote /dev/ttyACM0' 
-ex 'mon assert_srst scan' \
-ex 'mon tpwr enable'  \
-ex 'monitor swdp_scan'   \
-ex 'attach 2' \
-ex 'file /opt/nordic/nRF5_SDK_15.2.0_9412b96/examples/ble_peripheral/ble_app_pwr_profiling/pca10056/s140/armgcc/_build/nrf52840_xxaa.out' \
-ex 'load'  \
-ex 'compare-sections'  \
-ex 'kill'

Assert SRST until scan
Target voltage: 3.3V
Ignoring packet error, continuing...
Available Targets:
Available Targets:
Available Targets:
Available Targets:
No. Att Driver
 1      ARM Cortex-M
 2      Nordic nRF52 Access Port
 3      ARM Cortex-M
Attaching to Remote target failed
Loading section .text, size 0x3b60 lma 0x26000
Load failed
Section .text, range 0x26000 -- 0x29b60: matched.
Section .sdh_soc_observers, range 0x29b60 -- 0x29b68: matched.
Section .sdh_ble_observers, range 0x29b68 -- 0x29b80: matched.
Section .sdh_stack_observers, range 0x29b80 -- 0x29b90: matched.
Section .sdh_state_observers, range 0x29b90 -- 0x29b98: matched.
Section .ARM.exidx, range 0x29b98 -- 0x29ba0: matched.
Section .data, range 0x29ba0 -- 0x29c28: MIS-MATCHED!
warning: One or more sections of the target image do not match
the loaded file

The program is not being run.

```
3.2. For the target 2 it's a bit different
```
$ arm-none-eabi-gdb -nx --batch \
-ex 'target extended-remote /dev/ttyACM0' 
-ex 'mon assert_srst scan' \
-ex 'mon tpwr enable'  \
-ex 'monitor swdp_scan'   \
-ex 'attach 2' \
-ex 'file /opt/nordic/nRF5_SDK_15.2.0_9412b96/examples/ble_peripheral/ble_app_pwr_profiling/pca10056/s140/armgcc/_build/nrf52840_xxaa.out' \
-ex 'load'  \
-ex 'compare-sections'  \
-ex 'kill'

Assert SRST until scan
Target voltage: 3.3V
Ignoring packet error, continuing...
Available Targets:
Available Targets:
Available Targets:
Available Targets:
No. Att Driver
 1      ARM Cortex-M
 2      Nordic nRF52 Access Port
 3      ARM Cortex-M
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring
PC register is not available
PC not available
warning: while parsing target memory map (at line 1): Required element <memory> is missing
Loading section .text, size 0x3b60 lma 0x26000
Load failed
Section .text, range 0x26000 -- 0x29b60: MIS-MATCHED!
Section .sdh_soc_observers, range 0x29b60 -- 0x29b68: MIS-MATCHED!
Section .sdh_ble_observers, range 0x29b68 -- 0x29b80: MIS-MATCHED!
Section .sdh_stack_observers, range 0x29b80 -- 0x29b90: MIS-MATCHED!
Section .sdh_state_observers, range 0x29b90 -- 0x29b98: MIS-MATCHED!
Section .ARM.exidx, range 0x29b98 -- 0x29ba0: MIS-MATCHED!
Section .data, range 0x29ba0 -- 0x29c28: MIS-MATCHED!
warning: One or more sections of the target image do not match
the loaded file

Kill the program being debugged? (y or n) [answered Y; input not from terminal]
```

Any ideas appreciated.

UweBonnes, 2018-12-17:
Please Ttry #399. Perhaps it will reduces the detection to 2 devices.

navado, 2018-12-17:
Indeed. It works. The error during the scan process disappeared too.
However still cannot attach or program the device.

UweBonnes, 2018-12-18:
It seems your device is not yet known to BMP, so you can not attach. Find out the device id, e.g. as discussed om gitter or in some issue and add that number.

UweBonnes, 2019-03-22:
Probably solved with #434Reopen in case errors persist.

closed 2019-03-22 by UweBonnes