blackmagic-debug/blackmagic #422: Latest FW build not working

jhnbyrn, 2018-12-27:
I built the latest version of the BMP firmware using the instructions in the Hacking section of the wiki. Whether I update the device with the upgrade program or with dfu-util, I get the following in the dmesg log:

```
[10136.885706] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10136.885710] usb 1-1: Product: Black Magic Probe (Upgrade)
[10136.885713] usb 1-1: Manufacturer: Black Sphere Technologies
[10136.885716] usb 1-1: SerialNumber: C1DDA10C
[10170.912646] usb 1-1: USB disconnect, device number 24
[10171.267441] usb 1-1: new full-speed USB device number 25 using xhci_hcd
[10171.390749] usb 1-1: device descriptor read/64, error -71
[10171.620804] usb 1-1: device descriptor read/64, error -71
[10171.850730] usb 1-1: new full-speed USB device number 26 using xhci_hcd
[10171.974060] usb 1-1: device descriptor read/64, error -71
[10172.204075] usb 1-1: device descriptor read/64, error -71
[10172.310777] usb usb1-port1: attempt power cycle
[10172.954012] usb 1-1: new full-speed USB device number 27 using xhci_hcd
...

```

At this point, dfu-util no longer worked but I was able to get it to work again back by holding down the button while plugging it in to get it into bootloader mode. 

Any suggestions on how to debug this? Thanks!

UweBonnes, 2018-12-27:
What BMP platform? Did you perhaps compile for a wrong platform?

jhnbyrn, 2018-12-27:
> What BMP platform? Did you perhaps compile for a wrong platform?

I'm not 100% sure if I understand what you mean by platform, but if you mean the hardware, it's the hardware I got from 1 Bit Squared, along with the 1Bitsy board. I didn't change any options when I did the build - I just ran make from the top level directory. After that I built the updater and tried that. When that failed I tried dfu-util with the blackmagic.bin file from the build, but that resulted in the same issue. 

The only other thing I can think of that I did is I used arm-none-eabi-gcc-7.3 instead of the latest version. The reason I did this is that it's the latest build for my linux (arch) that doesn't have the problem described in issue #408 (-Werror=cast-function-type). That issue was preventing me from building it with 8.2 and 8.1.

UweBonnes, 2018-12-27:
BMP runs on the different platorms, e.g. the native 1 Bit Squared,on ST Links board, on bluepills, on FTDI board etc. etc. Using the binary for a wrong platform could result in the messagre above. It could also happen if you load the hex file with some update programm expecting a bin file.  The error looks very like an error on your side.

What do you mean by "updater"?  I think, the best way is to update is to use ../scripts/stm32mem.py <binfile>

I compiled with different gcc versions and, errors like #408 aside, things always worked. My staging branch should fix such and other problems.

What is the length of your binfile? Perhaps upload it somewhere...

UweBonnes, 2018-12-27:
Please use scripts/stm32mem.py. I never used  blackmagic_upgrade and it was not touched for ages.
I will also try to append a bmp/native bin file here  that I just compiled... Argh, upload always aborts!

UweBonnes, 2018-12-28:
Try artifacts in https://github.com/UweBonnes/blackmagic/tree/staging_artifacts

jhnbyrn, 2018-12-28:
It looks like the issue is the version of arm-none-eabi-gcc that I'm using. With arm-none-eabi-gcc-7.3, I'm getting the same results with either master or the staging branch. 

I tried reinstalling 8.2 for the staging branch ,in case #408 was fixed there. I still got the function cast error, but then I just tried working around it by removing the -Wextra option. This time, the bin file I got worked fine! It's the same with the master branch. Are there any issues with removing this option?

UweBonnes, 2018-12-28:
I use gcc version 8.2.1 20181213. Staging (d1c9c68f67ef4def0678d3221ea8547458ff15e9) fixes #408.

jhnbyrn, 2018-12-28:
> I use gcc version 8.2.1 20181213. Staging ([d1c9c68](https://github.com/blacksphere/blackmagic/commit/d1c9c68f67ef4def0678d3221ea8547458ff15e9)) fixes #408.

Got it...I was mistakenly referring to the repo you linked to previously as 'staging'. Thanks for all the help with this, I really appreciate it.

UweBonnes, 2018-12-28:
Please wrap up. What does work together, what not?

jhnbyrn, 2018-12-28:
arm-none-eabi-gcc 8.2.0 works, with the fix for #408. 

arm-none-eabi-gcc 7.3 does not work. It doesn't complain about the #408 issue but it produces a blackmagic.bin that does not work.

closed 2018-12-28 by jhnbyrn

jhnbyrn, 2018-12-29:
> So _your_ arm-none-eabi-gcc 7.3 is broken! I used gcc version 7.3.1 20180622 for half a year.
> Where did you get your arm-none-eabi-gcc 7.3 from?

I got it from the Arch Linux package archive [here](https://archive.archlinux.org/packages/a/arm-none-eabi-gcc/).