blackmagic-debug/blackmagic #424: GDB unresponsive after using n command

jhnbyrn, 2018-12-31:
I'm using arm-none-eabi-gdb 8.2 and the latest BMP firmware built from the master branch, on native hardware. I had this issue when I got the BMP, and I've also tried 1.6.1, with no improvement. The target is the 1Bitsy development board. 

In general, GDB works fine, but if I have a while loop with no statements inside it, for example:

`while (a < b);`

...then I cannot step through this using the 'n' command. If I use 'n' on this line, GDB becomes unresponsive, every time. All I can do is ctrl+z and kill the process. 

I've built the FW with debugging, and enabled the extra packet debug code in gdb_getpacket. Either GDB or the device seems to be stuck in an endless loop when the issue occurs. I'm including a snippet of what it logs via the debug tty below.

I seem to be able to step using 's' with no issues. If I put a dummy statement into the while loop, for example:

```
while (a < b) {
    int nop = 0;
}

```
...then I can step with 'n' and I don't get the issue. The empty while loop is the only way I've been able to reproduce the issue so far. 

I don't think there's anything wrong with my code because it works fine as long as I'm not trying to step with 'n' on this particular line. When the issue happens, I can tell that execution it definitely not continuing (my code blinks the led when it's running). 

Here's the log from /dev/ttyACM1- this should be enough to include the complete pattern, and it just goes on logging this repeatedly. 

```
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : m80001e8,4
m packet: addr = 80001e8, len = 4
gdb_putpacket : 90b400af
gdb_getpacket : m800024a,4
m packet: addr = 800024a, len = 4
gdb_putpacket : 0a460146
gdb_getpacket : m800024a,2
m packet: addr = 800024a, len = 2
gdb_putpacket : 0a46
gdb_getpacket : Z1,800024a,2
gdb_putpacket : OK
gdb_getpacket : c
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : m800024a,4
m packet: addr = 800024a, len = 4
gdb_putpacket : 0a460146
gdb_getpacket : z1,800024a,2
gdb_putpacket : OK
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : s
gdb_putpacket : T05
gdb_getpacket : g
00000000000000000000000000000000000000000000000000000000
gdb_getpacket : s
gdb_putpacket : T05
...
...
```

After I kill GDB, I can relaunch it, and re-attach to the target, and then resume debugging normally (as long as I don't use 'n' on this problem line). 

I tried using the assembly view in GDB - using 'n' worked fine until I used it on this instruction (0x80001f2), then it locked up as usual:

```
  |0x80001e8 <millis>              push   {r4, r7}
  |0x80001ea <millis+2>            add    r7, sp, #0
  |0x80001ec <millis+4>            ldr    r3, [pc, #12]   ; (0x80001fc <millis+20>)
  |0x80001ee <millis+6>            ldrd   r3, r4, [r3]
>|0x80001f2 <millis+10>           mov    r0, r3
  |0x80001f4 <millis+12>           mov    r1, r4
```

Any ideas what else I can do to debug it?

UweBonnes, 2019-01-01:
It is n known problem that tht debugger can not break into tight loops.  I do not know if that is a BMP problem or a problem with the cortex debugging hardware. How do other debuggers behave in that situation?

sidprice, 2019-06-16:
This is a common debugger issue and is not related to BMP at all. I see it with JLink and STLink, it is caused by the way the the debugger tries to step to the next line, an empty "while" loop I believe the debugger figures out the next line, in this case the same as the current line, it then decides that it does not need to step at all because it is already on the right line. 

Why do you need to single step over this line, can't you use "until 'location'"

UweBonnes, 2019-06-25:
As Sid stated: "This is a common debugger issue and is not related to BMP at all"

closed 2019-06-25 by UweBonnes

markrages, 2020-03-31:
I wasted some time on this issue again recently.

I am interested in learning more about the problem.

Seeing as this is a "common debugger issue", would you happen to have a link to the upstream gdb issue or to other debugger issues?

markrages, 2020-03-31:
Replying to myself.

Would this be the issue? https://sourceware.org/bugzilla/show_bug.cgi?id=21221

UweBonnes, 2020-03-31:
Please state what OS you use. For me and many others, working on Linux is fine. MacOS however seems to have a problem,

UweBonnes, 2020-03-31:
Yes, 21221 is the problem I talk about.

B.t.w, can one step in the situation with "ni" in the assembler code?

markrages, 2020-03-31:
Chip is STM32F301.

Debugging with PROBE_HOST=stlink on bluepill.

Using `gdb-multiarch` on Ubuntu. `GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git`

Code compiled `-Og` by `arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620`

Contents of mainloop:
```
  while (1)
  {
          uint32_t i;
          for (i=0; i<1000000l; i++) __asm("nop");
          HAL_GPIO_TogglePin(LED1_GPIO_Port, LED1_Pin);
  }
```
Stepping by `ni` doesn't hang:

```Available Targets:
No. Att Driver
 1      STM32F3
0x08000618 in main () at Src/main.c:116
116		  for (i=0; i<1000000l; i++) __asm("nop");
(gdb) ni
Note: automatically using hardware breakpoints for read-only addresses.
0x0800061a	116		  for (i=0; i<1000000l; i++) __asm("nop");
(gdb) 
116		  for (i=0; i<1000000l; i++) __asm("nop");
(gdb) 
0x08000614	116		  for (i=0; i<1000000l; i++) __asm("nop");
(gdb) 
0x08000616	116		  for (i=0; i<1000000l; i++) __asm("nop");
(gdb) 
0x08000618	116		  for (i=0; i<1000000l; i++) __asm("nop");
(gdb) 
0x0800061a	116		  for (i=0; i<1000000l; i++) __asm("nop");
(gdb) 
116		  for (i=0; i<1000000l; i++) __asm("nop");
(gdb) 
```
and so forth.

UweBonnes, 2020-03-31:
I rechecked and with BMP/stlink on the F4 Disco board, I can step over something  like
for (i=0; i<1000000l; i++) __asm("nop");.

> arm-none-eabi-gdb --version
GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git

markrages, 2020-03-31:
Ah, that's interesting.

1. What compiler version and flags are you using?
2. Could you attach the ELF file?

UweBonnes, 2020-03-31:
> arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

from
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm

arm-none-eabi-gcc -c -I/devel/en_apps/gpio/f4_discovery/nutbld/include -I/devel/ethernut/nut/include -I/devel/ethernut/nut/include/contrib  -DBOARDNAME=\"F4_DISCOVERY\" -DBOARD="<arch/cm3/board/f4_discovery.h>" -DSTM32DIE=413 -DVENDOR="<arch/cm3/stm/vendor/stm32f407xx.h>" -DPINMUXHEADER="<arch/cm3/stm/generated/stm32f417_pinmux.h>" -DDMACHANNELHEADER="<arch/cm3/stm/generated/stm32f417_dma.h>" -DIRQn_MAX=82 -MD -MP -mcpu=cortex-m4 -mthumb -D__CORTEX__ -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffunction-sections -fdata-sections -fomit-frame-pointer  -Og -g3 -Wall -Wstrict-prototypes -Wa,-a=gpio.lst -DIRQn_MAX=82  -o gpio.o /devel/en_apps/gpio/f4_discovery/../gpio.c
arm-none-eabi-gcc gpio.o -Wl,--defsym=flash0_length=1024K -Wl,--defsym=ccm_length=64K -Wl,--defsym=ram0_length=112K -Wl,--defsym=ram1_length=16K -Wl,--defsym=ram2_length=0 -mcpu=cortex-m4 -mthumb -D__CORTEX__ -mfloat-abi=hard -mfpu=fpv4-sp-d16 -nostartfiles -L/devel/ethernut/nut/arch/cm3/ldscripts -Tstm32f4_ccm_flash.ld -Wl,-Map=gpio.map,--cref,--gc-sections -L/devel/en_apps/gpio/f4_discovery/nutbld/lib -Wl,--start-group /devel/en_apps/gpio/f4_discovery/nutbld/lib/nutinit.o -lnutos -lnutarch -lnutdev -lnutcrt -lnutc  -Wl,--end-group -o gpio.elf

markrages, 2020-03-31:
OK, I tried with gdb-1.9, and `step` still locks up.

Next I will try different compiler.

markrages, 2020-03-31:
Ok, built with `arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]` and added `-g3` to `-Og` optimization.

And gdb-9.1 compiled from source.

Still not working.

```
0x080004c0 in main () at Src/main.c:116
116		  for (i=0; i<1000000l; i++) __asm("nop");
(gdb) step
Note: automatically using hardware breakpoints for read-only addresses.
^C^C^C
```
(I have to kill gdb externally.)