blackmagic-debug/blackmagic #436: EFM32 driver doesn't support recovering locked parts

richardeoin, 2019-01-18:
Parts are locked by clearing the Debug Lock Word (DLW)

perigoso, 2022-07-07:
should be closed as per
5943552a6b1f54ab9add80a87f64945db00b7769
and
#567

perigoso, 2022-07-07:
thats because I deleted it, this one is actually not implemetned (doesnt look like it to me at least), sorry for the confusion

dragonmux, 2022-07-07:
Oh, huh - indeed, the linked PR doesn't look to fix this. Re-opening till we can otherwise prove it's implemented

perigoso, 2022-07-08:
according to Silabs, the correct method to clear the debug lock word is to issue a mass erase through the AAP, which is in fact implemented by `efm32_aap_cmd_device_erase`, so this too should be closed.

dragonmux, 2022-07-08:
Thank you for checking this. Closing.

closed 2022-07-08 by dragonmux