blackmagic-debug/blackmagic #439: blue pill jtag_scan failed

Thompson-NO3, 2019-01-24:
I use a blue pill to test BMP 2.1
command monitor swdp_scan succeeded with 1 cortex m4
but command monitor jtag_scan failed.
after research for some time, I found the problem.
This is because I use fly wire to debug the target board and the default GPIO speed is 50 MHZ
It's too fast for the signals to transport.
So I just want to inform the maintainer: you can test the command yourself and adjust the freq of GPIO to let us scan the jtag cores and can use it to debug some target board. 
Thank you.

UweBonnes, 2019-02-21:
I used BMP a lot with flying wires. Please assert that your problems are not caused by some other error.

esden, 2021-12-15:
As this can't be reproduced and the issue is stale I am closing this.

closed 2021-12-15 by esden