blackmagic-debug/blackmagic #440: Missing support for Microchip ATSAMV7x and other recent CM7 devices

pavel-kirienko, 2019-01-29:
Are there plans to support the ATSAMV7x family? If not, is it because of lack of interest, lack of resources, or technical infeasibility?

Thanks.

UweBonnes, 2019-01-29:
BMP is a open source project. Things needs to be  contributed by interested people. Test hardware is also needed. When there are already similar chips, technical feasibility is normally not so hard. So think about trying yourself. Ask questions on Gitter. Be persistant when then trying to get code into the main brain. Look at the "offical" codings style.

UweBonnes, 2019-01-29:
brain->branch

kubiaj, 2020-09-25:
Hi,
how it looks like with the support? I might be interested to give helping hand here. But I would need some guidance here. I have checked the implementation and the sam3x seems to use EEFC which is also implemented in the SAME(V) series MCU. So if only cloning/updating sam3x is needed then I can give a try.  

Is the CM7 debugging (or whatever is needed for CM7) already in BMP?

kubiaj, 2020-09-26:
Hi
well, I have adopted the sam3x source code to samx7x devices (quite easy job). 

However I should start from the beginning -  I cannot connect to my target (SAME70 CM7):

GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor tpwr enable
Enabling target power
(gdb) monitor jtag_scan
Target voltage: 3.3V
JTAG device scan failed!

It always end with JTAG device scan failed! I have tested my modified FW and the latest build. Im using BMP probe (native).  If I use another programmer then everything is OK. The BMP is OK, because I can programm nRF52 with it using the same cables.

I need some advanced help here.

kubiaj, 2020-09-26:
OK ... my fault ... It would be better to use swdp_scan instead of  jtag_scan for SWD connected devices.  However now Im receiving:

(gdb) monitor swdp_scan
Target voltage: 3.3V
Exception: SWDP invalid ACK
SW-DP scan failed!

or 

Target voltage: 3.3V
Timeout during scan. Is target stuck in WFI?
SW-DP scan failed!

UweBonnes, 2020-09-28:
Problems that early in startup point to a problem on your setup:
- Device remaps SWJ pins and connection does not happen under reset
- Devices is sleeping for long time. Again connect under reset will help
- Flaky wiring, long cables, bad ground connections
- Inconsistant logic level.

Otherwise, BMP supports CM7 debugging.

esden, 2021-12-15:
@kubiaj any updates on this? Were you able to resolve the issue?

kubiaj, 2021-12-20:
@esden 
Hi sorry, it didn't end up well so I give up and I have moved to DAPlink implementation which is working well.

kubiaj, 2021-12-20:
@UweBonnes
I think that the DAPLink implementation will be a cool feature for BPM as it will be compatible with any MCU that openocd supports.  

I have find my BMP so I can give a try.

UweBonnes, 2021-12-20:
Sorry if my proposal was not clear enough. I wanted to say:  way: Peek at the DAPLINK implementation for ATSAMv7x and implement the ATSAMv7x  also in BMP. Is this what you understood too?

kubiaj, 2021-12-20:
OK my understanding was different 8-).  My idea was to implement DAPLink into BPM.

Im not sure if it possible to peak the implementation as the DAPLink it is something like SWD bridge between target MCU and openocd.  As far I remember the BMP contains the "whole stack" inside but the DAPLink is just the bridge. It contains a very little intelligence - and that makes it slower, I think that we have made in past some PoC with BPM compatible device (Chinese one) and the upload speed was too slow - some think like  8 kB/s but maybe Im wrong.

UweBonnes, 2021-12-20:
I meant the peeking at the implementation in OOcd. I did now have a peek, and it seems to be an implementation that uses register access, not a binary blob.That is good,  i.m.h.o.. But it is not simple and without actual part access and without some deeper understanding of the ATSAMV7 flashing, I retract my offer.

dragonmux, 2022-06-04:
Support for several ATSAMx7x parts was recently added, does that solve this issue and can it now be closed?

closed 2022-06-04 by pavel-kirienko

perigoso, 2022-06-04:
I completely missed this before *facepalm*, the recent PR should support this, if it works is another story