blackmagic-debug/blackmagic #441: NXP RT1052 & RT1062 support

eco747, 2019-02-04:
Hi,
Any chance to see RT105X & RT106X support ?

Regards

UweBonnes, 2019-02-04:
BMP is a community project.. Interested People extend existing Code or add new code. So  no predictions can be made.

btashton, 2019-08-13:
I'm looking at possibly adding support for the RT1050.  I am trying to figure out what to do here with the flash though since it is external to the part.  There is a special configuration area at the very start of the flash that contains configuration data, the ROM bootloader reads this using "safe" (slow) settings and then reconfigures the flexspi device with whatever was in there.  This also contains information about the size of the attached flash. 

One thought I had was to read this area of flash during the probe phase and it if failed to read, fall back to the "safe" settings that the ROM would use.  Then if a write was ever issued on this area I would reconfigure the flash including the regions based on this area.

Also all of the flash routines for the flexspi hardware are available from NXP under a BSD cause 3 license, it would be a lot more convenient if we could just leverage them... Is this doable?

UweBonnes, 2020-12-09:
The RT10xx series is now recognized by cdd07544d5df183520d23ef41420ae02c78aac2e.I'd appreciate if you implement the routines ;-)

dragonmux, 2023-01-30:
An update to this issue, we have been working on support for this in feature/iMXRT-support - please let us know how you get on with this branch, which is slated for inclusion in v1.10

dragonmux, 2023-04-23:
There is a PR open now with support for this processor series. Please give it a go and let us know how you get on

PR #1468 (2023-04-24): Feature: i.MXRT support

## Detailed description

This PR provides initial support for the i.MXRT10xx series processors. This includes complete RAM and Flash address maps, full debug support, and support for the SPI Flash boot device via FlexSPI1.

What this PR does not do is add support for other boot devices such as eMMC Flash, or Parallel Flash.

Via ORBTrace running at 10MHz over SWD, we are able to achieve 6kiB/s transfer speeds reliably.

Stubs for other boot sources are provided with the intent of handling expansion of supported boot devices in the future.

## Your checklist for this pu [... cut]
closed 2023-04-24 by esden