blackmagic-debug/blackmagic #444: t->idcode uninitialized in stm32l4.c

newbrain, 2019-02-20:
In the stm32l4.c option bytes handling a check is made on `t->idcode` to determine the size of the option area:
```C
static bool stm32l4_cmd_option(target *t, int argc, char *argv[])
{
	[...]
	if (t->idcode == 0x435) {/* L43x */
		len = 5;
	} else if (t->idcode == 0x460) {/* G07x */
		i2offset = g0_i2offset;
		len = 7;
	} else {
		len = 9;
	}
```
However, the `target` struct member `idcode` is never initialized: only a local `idcode` variable is used in the probe function.
```C
	uint32_t idcode = target_mem_read32(t, idcode_reg) & 0xfff;
	switch(idcode) {
```
and that variable is never assigned to `t->idcode`.
This might be a minor error, as the final effect will be writing to non-existent registers in the flash peripheral.
I see no checks for that part of the code, so the errors generated by  `target_mem_write32` might be ignored in practice (I have no way to check with my hardware).
Still, it's not correct handling.

UweBonnes, 2019-02-20:
#432 should fix that issue. Please report.

newbrain, 2019-02-20:
PR #432 seems to correct the issue.
Given its wide impacts, I'll add some comment there re: STM32WB55.

UweBonnes, 2019-02-21:
Closed by #445/#432.

closed 2019-02-21 by UweBonnes