blackmagic-debug/blackmagic #447: Corruption observed when loading to ram.

hydra, 2019-02-23:
Hi again guys,  first - thanks for your continued great work on BMP.

Recently I have been working on a new project and have been using the gdb `load` command to load .elf files which populate memory sections, however I noticed that there was some corruption at the end of the section that was transferred.

corruption:

```
    0x2407DFAE  DEB90000 DEB9DEB9 DEB9DEB9 DEB9DEB9  ..¹Þ¹Þ¹Þ¹Þ¹Þ¹Þ¹Þ
    0x2407DFBE  000000B9 00000000 00000000 00000000  ¹...............
```
should be:
```
    0x2407DFAE  DEB9DEB9 DEB9DEB9 DEB9DEB9 DEB9DEB9  ¹Þ¹Þ¹Þ¹Þ¹Þ¹Þ¹Þ¹Þ
    0x2407DFBE  00000000 00000000 00000000 00000000  ................
```

The marker was generated using a linker script `section`:
```
  .debug_marker :
  {
    . = ALIGN(16);
    LONG(0xDEB9DEB9)
    LONG(0xDEB9DEB9)
    LONG(0xDEB9DEB9)
    LONG(0xDEB9DEB9)
    . = ALIGN(16);
  } >CODE_RAM
```

the .debug_marker section is placed so that it is placed at the end of the file.  I added it only as a workaround to the corruption problem I faced in BMP and so that I could get some insight onto exactly what was being corrupted when my RAM -> FILE comparisions failed MD5 checksums.

However...

When i upgraded to latest master code and reflashed my probe the issue _seems_ to be gone!

I would just like to ask if this was a known issue that was fixed intentionally, and if so can you point me at the commit/code that was fixed?

UweBonnes, 2019-02-23:
This was probably fixed by using buffered write everywhere. Before it was hard to write the right number of bytes when length of the write and size of the flash write unit did not match. B.t.w. it is good habit to give much more context, like platform used, versions used etc when reporting problems.

hydra, 2019-02-23:
@UweBonnes ok, yes sorry for lack of version information in the question. slipped my mind.

I'll keep my eye on this issue and if it resurfaces I'll re-open this.  Previous code I was using was master from 14bedcc + baite patches from #274

closed 2019-02-23 by hydra