blackmagic-debug/blackmagic #453: Blackpill support

knopserl, 2019-03-01:


knopserl, 2019-03-01:
Thanks for this great BMP project!

I have flashed a couple of bluepills and they work fine using the PROBE_HOST=swlink 
I have now also a couple of blackpills, which I also want to use.
The blackpill has a few advantages like the correct USB pull-up (1.5kB) and also mounting holes.
But it has some few differences like the LED (PB12 instead of PC13).
I have changed the source in the platform/swlink dir in several files and it seem to use now the PB12 LED.
However, I have not really found out, where to configure the UART pins, which seem to be PB7 and PB8.
Could anybody tell me where this is defined?
It would be nice if the pinout could be defined in one place (e.g. platform.h) or if the blackpill could be supported by default.
Pin PB7 and PB8 is not ideal for the blackpill, as they are far from each other on different sides of the PCB.

I would e.g. like to use B10, B11, B0, B1 (two of them) for the UART port (terminal, serial output) or anything which is close to the flash pins and alose to each othe r(whatever you propose). But need to know where to configure.
Thanks a lot! 

Here are infos for the backpill
http://wiki.stm32duino.com/index.php?title=Black_Pill
http://wiki.stm32duino.com/index.php?title=File:Black_Pill_PCB.png
https://de.aliexpress.com/item/5pcs-STM32F103C8T6-ARM-STM32-Minimum-System-Development-Board-Module-For-arduino/32653839788.html?spm=2114.13010208.99999999.270.wL97wi

UweBonnes, 2019-03-02:
Changing Pin position unconditionally requires creating a new platform. We should try to avoid that. We need a way to detect the difference  between blue- and blackpin programatically. I can thing of 3 ways:
 1.) require some PIN of the black pin to be pulled low or high with this pin floating on bluepill/swlink.
 2.)  Require the bluepill to have some marker in flash, perhaps at highest address
-3.) Test what happens when you pull GPIO of the LEDs (PB12 vs PC13)  low and read back the pin. When unconnected you will read low. With the LED connected maybe we read high. But diode drop on LED may inhibit reading high or read high not reliable.

When you can recognise the blackpill, you can change device allocation conditionally. What device used is done in swlink/platform.h, the pin used are the standard pin or the remapped pins. Have a look at the datasheet and the rference manual.

knopserl, 2019-03-02:
@UweBonnes thanks for your time, repsonse and explanations.
I understand, that you/we don't want to make and maintain another platform (I did not dare to request it even). I just wanted to know if/how I can reassign some of the not matching pins between bluepill and blackpill (or the one blackpill I have mentioned).
I could do that even on my own (not disturbing the project), and I would/could do that with a new platform or even a new project folder (copy the blackshere repository).
I just wanted to know where the LED Pin and the second UART (RX/TX which can be used in the sketch for serial output) can be set/redefined, as I have not found a good/since place where pins can be easily remapped (some of them are in the platform.h of swlink folder)  
The differences are unfortunately not only the one programmable LED (PB12 instead of PC13) but also the other pins. I would be already happy to find the LED and the UART pins.
I have found the places where the LED PC13 is set nut not where the UART pins are selecetd. 
I found that out by testing that it was PB7 and PB8 on blue and blackpill but PB7/8 is not really ideal for the blackpill, it would be rather PB10/11 or PB0/1 or A2/A3 or something similar.
I not deep enough the code to make a conditional board detection. I would just rather do s static config (new platform) for the bluepill and blackpill.

Thanks anyway for your support so far.

UweBonnes, 2019-03-02:
I would love to see  proper support for more boards without  more platforms.

For UART pin selection: Look at rm0008, 9.3.8 USART alternate function remapping

UweBonnes, 2019-03-20:
Please have a look at https://github.com/UweBonnes/blackmagic/tree/blackpill and comment

knopserl, 2019-04-09:
@UweBonnes thanks a lot for your effort and support ans sorry for my late response. I was quite busy the last weeks but have seen that you have added blackpill support in/via the swlink platform right?
I did a build with: heinz@mint:~/blackmagic/src$ make PROBE_HOST=swlink
but I get compile errors if I use swlink as probe host.
I can compile with default (native) or stlink but then the pinout and led does not work (only the serial A2/A3). 
Have I understood something wrong or how should the make command look like?

Below is the compile error:

  GIT     include/version.h
  CC      target/adiv5.c
  CC      target/adiv5_jtagdp.c
  CC      target/adiv5_swdp.c
  CC      command.c
  CC      target/cortexa.c
  CC      target/cortexm.c
  CC      crc32.c
  CC      target/efm32.c
  CC      exception.c
make: *** Keine Regel vorhanden, um das Ziel „gdb_if.o“, 
  benötigt von „blackmagic“, zu erstellen.  Schluss.

UweBonnes, 2019-04-10:
There is no support for for the blackpill yet. Somebody has to contribute it. It could be done by adding it  conditional to the files in the swlink directory.

OpenUAS, 2023-01-29:
Close maybe in the event of a 1.9 release?

dragonmux, 2023-01-29:
We've left this issue open intentionally as BMP supports Blackpill v2, *not* the original Blackpill, meaning this issue is still valid and outstanding

closed 2023-05-11 by dragonmux