blackmagic-debug/blackmagic #455: PR #445 introduces attach-detach fault

newbrain, 2019-03-02:
Pull request #445 introduces a serious problem when repeatedly attaching and detaching from a target.
As the memory information is stored again in the attach function, gdb will see multiple overlapping areas and ignore them.

Software used:
- latest master (da62cba), stlink platform
- Windows 10 1809
- arm-none-eabi-gdb version 8.01

Hardware:
- STM32l476 Nucleo as target
- STM32WB55 dongle (on my dev branch)
- ST-Link V2.1 taken from another Nucleo.

Log:
```
PS C:\Users\newbrain> C:\SysGCC\arm-eabi\bin\arm-eabi-gdb.exe
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
[----------8<--snip some gdb messages--8<---------]
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-eabi".
[----------8<--snip more gdb messages--8<---------]
(gdb) target extended-remote COM4
Remote debugging using COM4
(gdb) monitor swdp_scan
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32L47x
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xfffffffe in ?? ()
(gdb) info mem
Using memory regions provided by the target.
Num Enb Low Addr   High Addr  Attrs
0   y   0x08000000 0x08080000 flash blocksize 0x800 nocache
1   y   0x08080000 0x08100000 flash blocksize 0x800 nocache
2   y   0x10000000 0x10008000 rw nocache
3   y   0x20000000 0x20018000 rw nocache
(gdb) detach
Detaching from program: , Remote target
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
warning: Overlapping regions in memory map: ignoring          <=== Here is the rub
0xfffffffe in ?? ()
(gdb) info mem
Using memory regions provided by the target.
There are no memory regions defined.
(gdb)
```

The problem is not limited to an explicit attach detach sequence, also interrupting a running program with Ctrl-C has the same outcome:
```
(gdb) file c:/Users/newbrain/Git/l476BMtest/VisualGDB/Debug/l476BMtest
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from c:/Users/newbrain/Git/l476BMtest/VisualGDB/Debug/l476BMtest...done.
(gdb) load
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x2df0 lma 0x8000188
Loading section .exidx, size 0x8 lma 0x8002f78
Loading section .init_array, size 0x4 lma 0x8002f80
Loading section .fini_array, size 0x4 lma 0x8002f84
Loading section .data, size 0x4004 lma 0x8002f88
Start address 0x8002ecc, load size 28556
Transfer rate: 35 KB/sec, 839 bytes/write.
(gdb) info mem
Using memory regions provided by the target.
Num Enb Low Addr   High Addr  Attrs
0   y   0x08000000 0x08080000 flash blocksize 0x800 nocache
1   y   0x08080000 0x08100000 flash blocksize 0x800 nocache
2   y   0x10000000 0x10008000 rw nocache
3   y   0x20000000 0x20018000 rw nocache
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: c:\Users\newbrain\Git\l476BMtest\VisualGDB\Debug\l476BMtest
[hit Ctrl-C]
Program received signal SIGINT, Interrupt.
warning: Overlapping regions in memory map: ignoring
0x08000a6e in HAL_GetTick ()
    at W:/STM32Repo/STM32Cube_FW_L4_V1.13.0/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal.c:337
337     {
(gdb) info mem
Using memory regions provided by the target.
There are no memory regions defined.
(gdb)
```
This has impact on all later operations, e.g. `load` will seem complete correctly (at a very high speed), but, as the memory is no longer characterized as flash, it will have no effect.
```
(gdb) monitor erase_mass
(gdb) x 0x08000000
0x8000000 <g_pfnVectors>:       0xffffffff
(gdb) detach
Detaching from program: c:\Users\newbrain\Git\l476BMtest\VisualGDB\Debug\l476BMtest, Remote target
(gdb) attach 1
Attaching to program: c:\Users\newbrain\Git\l476BMtest\VisualGDB\Debug\l476BMtest, Remote target
warning: Overlapping regions in memory map: ignoring
0x08000aae in HAL_Delay (Delay=500)
    at W:/STM32Repo/STM32Cube_FW_L4_V1.13.0/Drivers/STM32L4xx_HAL_Driver/Src/stm32l4xx_hal.c:363
363       while((HAL_GetTick() - tickstart) < wait)
(gdb) load
Loading section .isr_vector, size 0x188 lma 0x8000000
Loading section .text, size 0x2df0 lma 0x8000188
Loading section .exidx, size 0x8 lma 0x8002f78
Loading section .init_array, size 0x4 lma 0x8002f80
Loading section .fini_array, size 0x4 lma 0x8002f84
Loading section .data, size 0x4004 lma 0x8002f88
Start address 0x8002ecc, load size 28556
Transfer rate: 99 KB/sec, 839 bytes/write.
(gdb) x 0x08000000
0x8000000 <g_pfnVectors>:       0xffffffff
(gdb)
```

Possible solution: make sure the memory information is sent only once between two probings of the target (to be verified).

UweBonnes, 2019-03-02:
Argh, I did not check the original PR good enough. Newbrain, can you please restructure the code as in stm32f4.c?

newbrain, 2019-03-02:
I had a _very_ quick look: it seems the only meaningful difference is the use of `target_mem_map_free()` in the attach function. I'll try to have a PR asap, possibly by tonight/tomorrow night... depending on PRs (Public Relations ;-) )

UweBonnes, 2019-03-02:
No, the difference is bigger. For the working code in F4, Memory is only added in the attach function, after first releasing all remainders, while L4 adds memory in the probe function. I guess L4 must only add memory in the attach function.

newbrain, 2019-03-02:
Yes, absolutely. I was fooled as I add all the memory in attach in my WB branch.
Should be able to have the PR tonight, if I find no stumbling blocks.

UweBonnes, 2019-03-03:
Fixed by #456

closed 2019-03-03 by UweBonnes