blackmagic-debug/blackmagic #459: udev rules don't seem quite right

JamesHagerman, 2019-03-12:
After banging my head against the existing udev rules, I realized that they weren't actually targeting the devices correctly.

Adding a `GROUP="plugdev"` directive to the rules then running `ls -al /dev/ttyACM*` should show the 2 tty devices on the BMP with a group other than `dialout`. However, the current rules don't allow this change.

Here are updated udev rules that seem to target the leaf nodes correctly:

```
# Black Magic Probe
# there are two connections, one for GDB and one for UART debugging
# copy this to /etc/udev/rules.d/99-blackmagic.rules
SUBSYSTEM=="tty", ACTION=="add", ATTRS{interface}=="Black Magic GDB Server", SYMLINK+="ttyBmpGdb"
SUBSYSTEM=="tty", ACTION=="add", ATTRS{interface}=="Black Magic UART Port", SYMLINK+="ttyBmpTarg"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6017", MODE="0666", GROUP="plugdev"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6018", MODE="0666", GROUP="plugdev" 
```

Honestly, I'm not that great with udev, but changing the group and mode associated with devices should be pretty simple.

If you agree that this is an issue, or have other opinions, let me know and I'll open a PR with an agreed upon change.

dragonmux, 2022-06-05:
Having reviewed this, we can confirm this is in fact a bug. We will be patching it in v1.9 and providing a variant of the rules that uses 'uucp' instead of 'plugdev' for those distros that use that instead. Also, such a patch must also include `TAG+=uaccess` for distros that use neither group.

dragonmux, 2022-06-06:
@JamesHagerman we have written some new udev rules. Please grab the ones suitable for your distro and give them a try. We're using the UUCP ones here successfully.

PR #1039 (2022-06-12): Better udev rules
This PR seeks to address our existing udev rules being both outdated and not entirely correct.

To do this we have introduced: A set of updated rules for systems that use 'plugdev' as the removable device group; A set of updated rules for systems that use 'uucp' as the removable device group; support in both cases for the uaccess tag which also serves the same function.

This closes #459
closed 2022-06-12 by esden