blackmagic-debug/blackmagic #461: NRF52840 is not recognized

EeuweQ, 2019-03-19:
Hi,

When i connect the NRF52840 to the blackmagic probe i get the following GDB output

(gdb) target extended-remote /dev/cu.usbmodem7BAA69B81
Remote debugging using /dev/cu.usbmodem7BAA69B81
(gdb) monitor swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      ARM Cortex-M
(gdb) attach 1
Attaching to program: armgcc/out.hex, Remote target
warning: while parsing target memory map (at line 1): Required element <memory> is missing
0xfffffffe in ?? ()

I think it should recognize the NRF52840 and know the memory layout?

UweBonnes, 2019-03-19:
Please try #434  and report back.

EeuweQ, 2019-03-19:
I did try, but bricked my bmp..
Dont think the compile part worked correctly (size to small)

:blackmagic eeuwequerido$ sudo dfu-util -d 1d50:6018,:6017 -s 0x08002000:leave -D src/blackmagic_dfu.bin 
Password:
Sorry, try again.
Password:
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1d50:6018
Run-time device DFU version 011a
Claiming USB DFU Runtime Interface...
Determining device status: state = appIDLE, status = 0
Device really in Runtime Mode, send DFU detach request...
Device will detach and reattach...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08002000, size = 7252
Download	[=========================] 100%         7252 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state

Is it solvable or can i order a new one with firmware for NRF52840 in it?

UweBonnes, 2019-03-19:
Have a look at https://github.com/blacksphere/blackmagic/wiki/Frequently-Asked-Questions about upgrading and recovering and report back.

UweBonnes, 2019-03-19:
You must also flash blackmagic.bin. blackmagic_dfu.bin is the bootloader itself...

EeuweQ, 2019-03-19:
Thanks for your help, but.. :-)

I can still find the device with:

MacBook-Pro-van-Eeuwe:blackmagic eeuwequerido$ sudo dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found Runtime: [05ac:8290] ver=0166, devnum=5, cfg=1, intf=5, path="20-8", alt=0, name="UNKNOWN", serial="UNKNOWN"

But averything i try to change (even download) failes:

sudo dfu-util -v -d 1d50:6018,:6017 -s 0x08002000:leave -D ~/Downloads/blackmagic-native-1.6.1.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

And with the stm32_mem.py script:

sudo scripts/stm32_mem.py -a 0x08000000 src/blackmagic_dfu.bin -m
USB Device Firmware Upgrade - Host Utility -- version 1.2
Copyright (C) 2011  Black Sphere Technologies
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

No compatible device found

Any ideas? I am working on an Apple laptop, but getting a ubuntu laptop next week, is it easier on a device like that?

UweBonnes, 2019-03-19:
Your BMP is not in bootloader mode and not visible. 1d50:6018 vs Found Runtime: [05ac:8290]. Force bootloader entry:
/platforms/native> grep Force *
platform.h: * Force DFU mode button: PB12

EeuweQ, 2019-03-19:
Hi UweBonnes!

That was the trick, i can also update the firmware of my NRF52840 now!!! (with the new patches)

Thank you very much 👍

closed 2019-03-19 by EeuweQ

UweBonnes, 2019-03-19:
Please comment #434