blackmagic-debug/blackmagic #468: Windows drivers fail with "The system cannot find the specified file"

adamcrossley, 2019-04-24:
When attempting to install the inf driver files for Windows, the installation fails repeatedly with the error message "The system cannot find the specified file".

Looking at the inf files, there are some referenced files (like blackmagic.cab) which don't exist anywhere in the distribution. Is there another source to get the missing files from to complete the driver installation?

sidprice, 2019-04-25:
What version of Windows are you using?

 

Did you use the instructions here -> https://github.com/blacksphere/blackmagic/wiki/Getting-Started

Sid

 

Sid Price

Desktop Application and Embedded System Design

adamcrossley, 2019-04-25:
Hi there, I'm using Windows 10.

I can see from the .inf files that it's using libusb-win32 as the underlying driver, and from the libusb-win32 page it doesn't list Windows 10 as a target platform, so, maybe that's the issue?

I saw that guide you linked, but I didn't know which one was out of date, because installing the libusbk drivers conflicts with installing the provided inf (libusb-win32) drivers, i.e. you can only install one or the other.

Since the inf files are still in the current release and they're specific to each usb device (updater vs tracer), it would make sense that we should be installing the .inf files over using zadig to install generic drivers. I don't really know which is meant to be the latest, most recommended solution, but the libusbk driver at least works to allow the devices to be recognized under Windows.

sidprice, 2019-04-25:
First, on Windows 10 no driver installation is required when debugging with BMP. If you want to update the firmware then you would use zadig. Disregard the ZIP file of drivers, the WiKi page I linked to is the correct information for Windows 10 as far as I am aware.

adamcrossley, 2019-04-26:
Ok, good to know. The debugger seemed to be able to connect without installing the driver, but the upgrade utils like dfu-util would not update without the driver. But after using zadig the upgrade is working, so thanks for the help.

closed 2019-04-26 by adamcrossley