blackmagic-debug/blackmagic #470: Cannot build on windows

hierophect, 2019-05-07:
Can the libopencm3 snapshot please be updated to the latest version that avoids the use of shell find, so that blackmagic will compile on windows by default? 

Libopencm3 issue referenced here:
https://github.com/libopencm3/libopencm3/issues/828
Issue has been run into before for this repo, but not resolved:
fatal error: libopencm3/stm32/f0/nvic.h: No such file or directory

I'm not intimately familiar with windows shell shenanigans so I'm struggling with this find thing. If someone can explain what I actually need to do to get even a basic build of this working, I'd appreciate it.

hierophect, 2019-05-08:
Well, it's a bit confusing that this issue only happens for this project when it's been solved for libopencm3 as a whole since 2017, as per my link. I don't know how many windows users encounter the error, but for those that do, they need to recognize that this issue is caused by an old snapshot and then either update themselves or muck around in their windows path to rearrange the two conflicting `finds`.

sidprice, 2019-05-08:
Since each BMP platform only needs libopencm3 for one MCU I decided that I would build the required sub-module using a CygWin installation, I describe that process here -> http://www.sidprice.com/2018/05/23/cortex-m-debugging-probe/

The above blog also describes building BMP, however, recently, I have been working towards removing the need for CygWin when building BMP itself and will publish that once I have a little time to write it up and publish my repo.

Since once the libopencm3 library only needs building once I chose not to expend any more effort on getting that sub-module to build on Windows. Is there some strong reason to update the libopencm3 library from where it is in BMP?

sidprice, 2019-05-08:
I understand your point-of-view, as I said the libopencm3 sub-module only needs to be built once and to expend effort regression testing a new snapshot of it doesn’t make time-sense to me. Much more important is to be able to edit-compile-debug BMP itself. However, if this is really important to you could always submit a pull request with an updated sub-module, that would be outstanding and remove a potential barrier to entry for Windows users.

hierophect, 2019-05-08:
I might be able to get that set up as a part of an article I'm doing.

hierophect, 2019-05-10:
Since my ability to do this is up in the air and it doesn't appear to be high priority to anyone else I'm closing the issue. I'll return with a fix if possible.

closed 2019-05-10 by hierophect