blackmagic-debug/blackmagic #471: Clarify Flashsize_f103

hierophect, 2019-05-10:
Since using bluepills seems to be a popular issue, and they will not be receiving a separate platform as per https://github.com/blacksphere/blackmagic/pull/354, I'd like to make some clarifications to:
https://github.com/blacksphere/blackmagic/blob/master/src/platforms/stlink/Flashsize_F103
Which documents how to force larger flash sizes onto the F103C8T6

The objectives and implementation on this page is not very clear. 
```
script/stm32_mem.py does not care for the announced size:
 > ../scripts/stm32_mem.py blackmagic.bin
 ...
 USB Device Firmware Upgrade - Host Utility -- version 1.2
 ...
 Programming memory at 0x08010800
 All operations complete!

Get length of binary
  > ls -l blackmagic.bin
 -rwxr-xr-x 1 bon users 59712 21. Sep 22:47 blackmagic.bin
Actual file size may differ!

Upload binary from flash with the exact size
  > dfu-util -s 0x08002000:leave:force:59712 -U blackmagic.bin.1

Compare
  > diff blackmagic.bin*
No differences should get reported!
```

What is the distinction between the python script and the later use of dfu-util? What is the blackmagic.bin.1 file and how is it generated? I think this page would greatly benefit from a short paragraph describing purpose of the steps and why they occur in this order. 

Lastly, I notice this file is in the st-link, as opposed to swlink which was the target of https://github.com/blacksphere/blackmagic/pull/395. Why is this? 

I'm writing an article on using Black Magic on the bluepill/other targets, so I'm hoping to clear this up. There are a LOT of old article/tutorials out there that simply don't work because they assume outdated smaller flash sizes and I'd like to clean up that misinformation.

UweBonnes, 2019-05-12:
Please comment if src/platforms/Flashsize_F103  in https://github.com/UweBonnes/blackmagic is clearer.

UweBonnes, 2019-05-12:
"Lastly, I notice this file is in the st-link" : There are a lot STM boards out there with a STM32F103C8. They used to work with BMP up to some version, but as BMP grew over the 64 k limit, reflashing was difficult. Propose what to add to the swlink directory!

hierophect, 2019-05-14:
I don't see any differences. The current file is presumably just yours merged in - the text you've recommended is identical to what I posted in my issue. 

I'd love to reword this but I need to actually understand why the steps are what they are, which is unclear. 
- Do the python script and dfu-util both serve to flash the binary? If so, why use both? 
- What is blackmagic.bin.1? Does the python script generate this? 
- Why is the `diff` required? What is the purpose of this comparison, AFTER the binary has been flashed?
- Why is the bluepill only referred to on the Readme's for swlink (STlink V1), but the instructions for flashing it are only in the st-link (V2, presumably) folder, where it is not mentioned in the Readme at all?

There is very little I can do to improve the instructions or document them in my article without understanding these 4 items.

UweBonnes, 2019-05-14:
Argh, I meant https://github.com/UweBonnes/blackmagic/tree/doc

hierophect, 2019-05-14:
I'm sorry, but that commit still only has the 2 year old file? Can you link to it directly?

hierophect, 2019-05-14:
Perhaps I'm navigating to it wrong? You meant src/platforms/stlink/Flashsize_F103, correct?  /platforms doesn't contain any documentation files.

UweBonnes, 2019-05-14:
Argh,  https://github.com/UweBonnes/blackmagic/tree/doc now witth real changes

hierophect, 2019-05-14:
So to summarize the changes it removes the dfu-util steps and replaces them with
```
Stm32_mem compares binary to flash and flashed binary and
reports mismatch.
```
I guess this implies that the comparison steps are actually occurring in the python sketch. But I'm not sure this solves the ambiguity about why the comparison is actually necessary or what the resulting outcomes are. 
I'll try and get this working with the python sketch by itself tomorrow.

UweBonnes, 2019-05-15:
As I tried to explain: There is no guarantee that there is working good FLASH above 64k on a STM32F103x8. You can use that flash on your own risk. By comparing the result after flashing , you minimize that risk.

hierophect, 2019-05-17:
I cannot get the stm32_mem script to work on either my windows or mac machines. Is it linux only? I only receive this message:
"No compatible device found"

I believe I understand the steps correctly now, at least. In the original instructions, the purpose of all steps after the initial use of the python script is simply to check whether the binary has correctly uploaded and the memory is intact, correct? That's reasonable, when explained: why is your suggested change removing these steps?

UweBonnes, 2019-05-17:
"No compatible device found"
Did you announce the both BMP DFU device with ZADIG?  The devie normally visible when BMP is running? And the device that appears in DFU mode?

The additional checks for integrity are removed as stm32mem does the checks already.

hierophect, 2019-05-17:
My mistake, issue was a Bluepill with an uncorrected USB resistor (they ship with a defect, I forgot to repair the one I was using). 

I've reproduced all steps. My suggestion for expanding the documentation would be:

```
F103C8 and F103CB share their chip architecture and most users report that undocumented upper flash exists on the F103C8. However, this flash may have been tested bad, or not have been tested at all, or, in the best case, was tested good but market requirements made STM sell it as the F103C8.

Ignoring the chip markings and using an F103C8 blindly as a F103Cb has been done already with few problems on many china boards (e.g. blue pill). This second approach will work for many of the older STLinks. 

However, The F103C8 will report its own flash size as 64kb, preventing dfu-util from using it, as it will not allow upload past announced flash. 
 > dfu-util -S E4D078EA -s 0x08002000:leave -D blackmagic.bin
 dfu-util 0.9
 ...
 dfu-util: Last page at 0x0801093f is not writeable

To flash above the announced size of the chip, use the dfu bootloader and the included stm32_mem.py script. This script does not care about the announced size and will attempt to flash the full binary regardless of whether the upper flash is functional or not. It will then read back the chip flash and compare to the original on the host machine, to confirm whether the entire binary was successfully uploaded or whether problematic upper flash corrupted the upload. 

First, you must flash the blackmagic_dfu.bin using a separate blackmagic, stm32loader, stlink, or some other method. This binary is small and will easily fit in the Bluepill's announced flash. 

Once the blackmagic_dfu.bin is flashed, attach to the board via USB. Navigate to the src folder, and run the following python script to flash the blackmagic.bin binary to the chip. This script requires libusb and the pyusb python library. 
 > ../scripts/stm32_mem.py blackmagic.bin
 ...
 USB Device Firmware Upgrade - Host Utility -- version 1.2
 ...
 Programming memory at 0x08013C00
 Verifying memory at 0x08013C00
 Verified!
 All operations complete!

The verification steps indicate whether the flashed binary, when read back, is identical to the original. If the binaries do not match, this indicates the upper flash on the target F103C8 is not viable and another chip should be used. 
```

jcw, 2019-05-24:
Might I suggest a TL;DR section at the start of all this? In the end it comes down to:

* some F103's have more flash than the 64K they report
* current BMP builds need more than 64K flash
* the `dfu_util` code refuses to upload firmware which exceeds the reported size
* use `../scripts/stm32_mem.py blackmagic.bin` instead of dfu_util to get around this

hierophect, 2019-05-28:
That sounds reasonable to me. Is a documentation change like this appropriate for merge request or is it better implemented by one of the more frequent curators alongside a wider update?

UweBonnes, 2019-05-28:
A PR is welcome. Feedback on https://github.com/UweBonnes/blackmagic/commits/451 (based on #451) to complile a C file fior upgrade is welcome too.

philpem, 2019-08-25:
@hierophect - May I suggest adding something to this effect to your PR:

> Some clone ST-Link/V2 boards have appeared which use a CKS CKS32F103C8T6 microcontroller. This part is an unlicensed pin-compatible clone of the STM32F103C8T6. Unlike the official ST part, this device only seems to contain 64K of Flash. If stm32_mem.py is used, there will be a verification error at address 0x2000 due to the flash write wrapping around.

I was going to suggest a commit-id which produced a <64K bin file, but I haven't been able to find one (gcc 6.3.1). The easiest fix for one of these boards might be to fit a genuine STM32F103CBT6.

UweBonnes, 2019-08-25:
All of the STM32F103x8 are the same chip as STM32F103xB, with flash above 64 k available, but not tested. It is always worth to try to flash above 64k and if verify succeeds, all is fine. Only in case verify fails, which I never heared of, a stripped down binary would come handy.

hierophect, 2019-08-28:
@philpem I don't really see a reason to include that, as that's the "expected" behavior for a F103. This is basically a hack enabled by an ST manufacturing defect, which I think will be pretty clear to anyone using it, and naturally won't apply to clones who are just matching the datasheet specs.

hierophect, 2019-08-28:
That said I totally lost track of this PR amongst other things. @UweBonnes are there further steps I should take for the PR to get merged?

philpem, 2019-08-29:
@hierophect That's a fair point, I was thinking that some people (like me) might assume the Chinese chip has the same quirks as the ST one. As for ST, its probably more yield management (like what Intel did with the 486SX/DX back in the day).

Though it might be an idea to make the "verify error at 0x2000" note a little more prominent - as a reminder that if you see that error, you need to swap the STM32 chip. At that point you may as well fit an F103RB (128k flash) and avoid problems.

Sorry for tacking this on your thread!

UweBonnes, 2019-08-29:
Find in #507 another tltr.

git fetch upstream pull/507/head:507
git checkout 507

Please comment

commit 80a9fd51 (2019-09-01): Clarify STM32F103x8 Flashsize handling.

Should finally fix #471.
closed 2019-09-01 by UweBonnes