blackmagic-debug/blackmagic #478: Cannot attach to EFM32ZG

Enzeys, 2019-05-23:
I cannot seem to attach to an EFM32ZG108F32.
When I run the swdp_scan command I see it listed as an 'ARM Cortex-M' processor.
When trying to attach to it via 'att 1' it says the attaching failed.
I can program the chip normally through its bootloader however.
[image]

UweBonnes, 2019-05-23:
If only "ARM Cortex-M" is recognized by "scan .." that means your chip is not yet recognized. In the easy case, when a similar chip with same programming procedure is already know to BMP, adding the MCUID to the appropriate file is enough. If programming algos differ, more work is needed.

As probably no other user of BMP uses this chip, best is if _you_ try to contribute to get the chip supported.

closed 2019-05-23 by UweBonnes

Enzeys, 2019-05-23:
Thank you. What got me confused is that the EFM32ZG is already listed in the devices struct in efm32.c. I thought that meant that it supported it?
How can I start contributing aka how can I start modifying the source code to get it recognised? I'm not new to programming, however I'm new to contributing & modifying others' code so I'm quite lost ^^

UweBonnes, 2019-05-23:
If you have the firmware corresponding to the sources on your stick, there is no mess up.

Find out the MCUID of your chip from datasheets/reference/user-manual. Check if hat this MCUID is already recogniced in src/target/efm32.c. If yes, problems are deeper. If no, check datasheets/reference/user-manual if there are differences between your chips and the already recognized chips. If yes, problems are deeper. If now, add you MCUID and set the needed flash sizes.