blackmagic-debug/blackmagic #479: Cortex-M33?

nickray, 2019-05-24:
Hi, is anyone working on or planning to work on Cortex-M33 support? Given it's Cortex-M4 + extras, maybe the necessary adjustments aren't huge. There's for instance a nice "LPC55S69-EVK: LPCXpresso55S69" dev board available.

UweBonnes, 2019-05-24:
When STM32L5 is available, I will sure have a look. What about trying yourself? Compare flash algo of the  LPC55S69 with already implemented algos for LPC. But wait, doesn't have the LPC IAP calls for programming? Are they the same? Then a lot of the work is already done.

Furthermore I guess that LPC55S69 has the newer ADIv5 interface where you can read out the MCUID from the AP. No need for the IAP call when probing. This probing is ugly when done on other CPUs...

What BMP capable hardware do you have? With an STlinkV2, you can try https://github.com/UweBonnes/blackmagic/tree/stlinkv2. As with the libftdi adapters, BMP runs on the PC and has much better debug output capabilities and no need for BMP firmware upload. That makes debugging easier. When done, program your "real" BMP.

nickray, 2019-05-24:
Cool, good to know, particularly about your PC branch! Who knows when the L5 will actually be out, the NXP is (a dual-core monster...) available today :)

Intent was definitely just to ask about status; re. implementing it myself will have to learn the chip (via on-board J-Link) a bit first.

UweBonnes, 2019-07-28:
ST did not meat there promise of general availability of the L2 in Q2 2019...

UweBonnes, 2019-07-28:
L2->L5

JJJollyjim, 2019-08-17:
Also keen on using this with a Cortex M23 (also an armv8 target) :+1:

mayanez, 2020-03-10:
Has there been any update on this? I'm also interested in this 😃

UweBonnes, 2020-03-10:
With #507, the infrastructure to read the romtable is there. Probably targets/stm32l4 can be made to read and program the L5 with not to much effort. L5 is nit high on my agenda. Contributions welcome!

mean00, 2020-04-08:
In case it helps, this is the romtable dump of a STM32L552

Leaving MASS Mode
Enter SWD
Read Core ID: 0x0be12477
RESET_SEQ succeeded
TARGETID 00000001
AP   0: IDR=14770015 CFG=00000000 BASE=e00fe003 CSW=0b800040
ROM: Table BASE=0xe00fe000 SYSMEM=0x1, PIDR 0x00000a0472
ROM: Table BASE=0xe00ff000 SYSMEM=0x1, PIDR 0x04000bb4c9
 0 0xe000e000: Debug component - Cortex-M33 () (PIDR = 0x04000bbd21)
 1 0xe0001000: Debug component - Cortex-M33 () (PIDR = 0x04000bbd21)
 2 0xe0002000: Debug component - Cortex-M33 () (PIDR = 0x04000bbd21)
 3 0xe0000000: Debug component - Cortex-M33 () (PIDR = 0x04000bbd21)
 4 Entry 0xfff41002 -> Not present
 5 0xe0041000: Debug component - Cortex-M33 () (PIDR = 0x04002bbd21)
 6 0xe0042000: Debug component - Cortex-M33 () (PIDR = 0x04000bbd21)
 7 Entry 0xfff44002 -> Not present
 ROM: Table END
1 0xe0040000: Debug component - Cortex-M33 () (PIDR = 0x04000bbd21)
2 Entry 0x1ff02002 -> Not present
3 Entry 0x1ff02002 -> Not present
ROM: Table END

UweBonnes, 2020-04-08:
Someone needs to write code to decode the CortexM33 romtable pidr/cidr bits!

themadinventor, 2020-07-07:
I'm making some progress on this. The same PIDR is used for each of the SCS, DWT, FPB and ITM blocks:
>CoreSight permits that two or more functionally different components are permitted to share the same Part number,
so long as they have different values of the DDEVTYPE or DDEVARCH registers.

So the approach I've come up with is to, after finding `PIDR 0x04000bbd21`, checking the `ARCHPART` bits in `DEVARCH`. Those should be `0xA04` on a v8M SCS. If that's the case, just proceed with `cortexm_probe`.

This seems to work fine on the STM32L5. I haven't looked into the flash driver yet.

I'm personally more interested in the LPC55S69, but it's tricker to get working, due to the dual cores and debug authentication.

Anyway, I'll try to clean up my (simple) patch for generic v8M support and file a PR in the coming days :)

UweBonnes, 2020-07-07:
Cortex-M33 is ADIv6 vs ADIv5 for M0... M7. The Decoding is different and best would be to implement that decoding. Perhaps a look at openocd and pyocd might give you some ideas.

UweBonnes, 2020-07-07:
For the NUCLEO-L552, "pyocd gdb" dumps:
0001235:INFO:board:Target type is cortex_m
0001247:INFO:dap:DP IDR = 0x0be12477 (v2 MINDP rev0)
0001252:INFO:ap:AHB5-AP#0 IDR = 0x14770015 (AHB5-AP var1 rev1)
0001262:INFO:rom_table:AHB5-AP#0 Class 0x1 ROM table #0 @ 0xe00fe000 (designer=020 part=472)
0001265:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c9>
0001265:INFO:rom_table:  AHB5-AP#0 Class 0x1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c9)
0001269:INFO:rom_table:  [0]<e000e000:SCS-M33 class=9 designer=43b part=d21 devtype=00 archid=2a04 devid=0:0:0>
0001271:INFO:rom_table:  [1]<e0001000:DWT class=9 designer=43b part=d21 devtype=00 archid=1a02 devid=0:0:0>
0001273:INFO:rom_table:  [2]<e0002000:BPU class=9 designer=43b part=d21 devtype=00 archid=1a03 devid=0:0:0>
0001275:INFO:rom_table:  [3]<e0000000:ITM class=9 designer=43b part=d21 devtype=43 archid=1a01 devid=0:0:0>
0001277:INFO:rom_table:  [5]<e0041000:ETM-M33 class=9 designer=43b part=d21 devtype=13 archid=4a13 devid=0:0:0>
0001279:INFO:rom_table:  [6]<e0042000:CTI class=9 designer=43b part=d21 devtype=14 archid=1a14 devid=40800:0:0>
0001281:INFO:rom_table:[1]<e0040000:TPIU-M33 class=9 designer=43b part=d21 devtype=11 archid=0000 devid=ca1:0:0>
0001284:INFO:cortex_m_v8m:CPU core #0 is Cortex-M33 r0p2
0001290:INFO:cortex_m:FPU present: FPv5-SP
0001294:INFO:dwt:4 hardware watchpoints
0001301:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0001320:INFO:server:Semihost server started on port 4444 (core 0)
0001320:INFO:gdbserver:GDB server started on port 3333 (core 0)

themadinventor, 2020-07-12:
I took a look at how pyOCD handles it and updated the patch. I've filed PR #690 where we can discuss the solution :)

UweBonnes, 2020-07-12:
Perhaps give a short reference to the sources you used in the patch description. Otherwise : Nice work! It adds however about 1 k to our flash, but I think that is worth it.

UweBonnes, 2020-07-12:
https://github.com/UweBonnes/blackmagic/tree/stm32l5 adds L5 detecion, memory map and flash r/w. Option handling is missing.

themadinventor, 2020-07-12:
Yeah, the `pidr_pn_bits` table is a beast. I'm thinking that lines that are `aa_nosupport` could be grouped together and wrapped with a `#ifdef ENABLE_DEBUG`, because they are of no use in a non-debug build anyway. That would save approx 400 bytes.

UweBonnes, 2020-07-12:
Good idea!

UweBonnes, 2020-12-09:
Cloes with #690

closed 2020-12-09 by UweBonnes

lucagessi, 2022-01-19:
Hi.
Is cortex-m33 actually supported?
I read in cortexm.h header that there is some kind of reference to cortex-m33.
I have a MGM210P0 module but blackmagic says that the chip in unknown during swd scan.

UweBonnes, 2022-01-19:
CortexM2|33 is already recognized. Debug will work, as long as you do not use M2|33 specific registers.  How to map secure and unsecure regions needs to be evaluated, as handling other M2|33  specific acpects.

MGM210P0 specific handling like memory map and flashing support needs to be done by somebody with the part in hand and some understanding of the specific implementation. Often new chips are a variant of already supported chips, so few additions may get you going.

Please run git head compiled for hosted as "blackmagic -t" and post the output of "blackmagic -t" as a new issue to request , or perhaps even as a start to implement MGM210P0  support.

lucagessi, 2022-01-19:
Thank you for the response.
I really would like to try to implement a new chip but actually I don't know where to start. Do you have some kind of documentation for porting new chip?

Apart from this below the ouput of blackmagic -t:

```
BMP hosted (BMP Only) v1.7.1-335-g5c07d61
Using:
 Black Magic Probe (Unknown)  V2__v1.7
Running in Test Mode
Target voltage: 2.37V Volt
Speed set to  3.2727 MHz for SWD
Trying old JTAG to SWD sequence
TARGETID 01c016e7
DPIDR 0x6ba02477 (v2 rev6)
RESET_SEQ failed
AP   0: IDR=84770001 CFG=00000000 BASE=e00fe003 CSW=43800040 (AHB-AP var0 rev8
Halt via DHCSR: success 00130003 after 4ms
ROM: Table BASE=0xe00fe000 SYSMEM=0x00000001, designer 673 Partno c01
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 43b Partno 4c9
 0 0xe000e000: Debug component - Cortex-M33 (System Control Space) (PIDR = 0x04000bbd21  DEVTYPE = 0x00 ARCHID = 0x2a04) -> cortexm_probe
CPUID 0x410fd213 (M33 var 0 rev 3)
 1 0xe0001000: Debug component - Cortex-M33 (Data Watchpoint and Trace) (PIDR = 0x04000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a02)
 2 0xe0002000: Debug component - Cortex-M33 (Breakpoint Unit) (PIDR = 0x04000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a03)
 3 0xe0000000: Debug component - Cortex-M33 (Instrumentation Trace Macrocell) (PIDR = 0x04000bbd21  DEVTYPE = 0x43 ARCHID = 0x1a01)
 4 Entry 0xfff41002 -> Not present
 5 0xe0041000: Debug component - Cortex-M33 (Embedded Trace) (PIDR = 0x04002bbd21  DEVTYPE = 0x13 ARCHID = 0x4a13)
 6 0xe0042000: Debug component - Cortex-M33 (Cross Trigger) (PIDR = 0x04000bbd21  DEVTYPE = 0x14 ARCHID = 0x1a14)
 7 Entry 0xfff44002 -> Not present
 ROM: Table END
1 0xe0040000: Debug component - Cortex-M33 (Trace Port Interface Unit) (PIDR = 0x04000bbd21  DEVTYPE = 0x11 ARCHID = 0x0000)
remote_ap_mem_read returned REMOTE_RESP_ERR at apsel 0, addr: 0xe00fdff0
remote_ap_mem_read returned REMOTE_RESP_ERR at apsel 0, addr: 0xe00fdff4
remote_ap_mem_read returned REMOTE_RESP_ERR at apsel 0, addr: 0xe00fdff8
remote_ap_mem_read returned REMOTE_RESP_ERR at apsel 0, addr: 0xe00fdffc
CIDR read timeout on AP0, aborting.
ROM: Table END
AP   1: IDR=54770002 CFG=00000000 BASE=00000003 CSW=80000040 (AHB-AP var0 rev5
AP   2: IDR=84770001 CFG=00000000 BASE=f0000003 CSW=43800000 (AHB-AP var0 rev8
remote_ap_mem_read returned REMOTE_RESP_ERR at apsel 2, addr: 0xf0000ff0
remote_ap_mem_read returned REMOTE_RESP_ERR at apsel 2, addr: 0xf0000ff4
remote_ap_mem_read returned REMOTE_RESP_ERR at apsel 2, addr: 0xf0000ff8
remote_ap_mem_read returned REMOTE_RESP_ERR at apsel 2, addr: 0xf0000ffc
CIDR read timeout on AP2, aborting.
remote_ap_mem_write_sized returned REMOTE_RESP_ERR at apsel 0, addr: 0xe000ef54
remote_ap_mem_write_sized returned REMOTE_RESP_ERR at apsel 0, addr: 0xe000edf4
*** 1 Unknown ARM Cortex-M Designer 673 Partno c01 M33
```
`