blackmagic-debug/blackmagic #480: bug in swlink/usbdfu.c ?

jcw, 2019-05-24:
There's a bug in the code, which looks to be a remnant of the stlink/usbdfu.c platform:

https://github.com/UweBonnes/blackmagic/blob/master/src/platforms/swlink/usbdfu.c#L75

I think this should be:

	if(normal_boot)

I'm trying to port to a "HyTiny" board, which has the LED on PA1. I don't understand what this check on the configuration setting of PA1 does. Removing the check made the port work.

UweBonnes, 2019-05-24:
See platforms/swlink/platform_common.c:
`       uint32_t crl = GPIOA_CRL;
        /* Assert bootloader marker.
         * Enable Pull on GPIOA1. We don't rely on the external pin
         * really pulled, but only on the value of the CNF register
         * changed from the reset value
         */
`
If you need PA1, you need another boot marker. Best is to  make it configurable with platform.h

Using some pattern in RAM would be a better idea i.m.h.o. As RAM content is not reset by warm start, this also allows real rest for bootloader entry and not only core reset as now. But there is a lot of hardware with a bootloader using this setup, so keep it as legacy if you use an existing platform.

jcw, 2019-05-24:
Aha, murky stuff. What would you suggest for now, given that I do indeed need PA1?

jcw, 2019-05-24:
FWIW, I have a modified swlink working - for [this board](https://www.hotmcu.com/stm32f103tb-arm-cortex-m3-development-board-p-222.html) - I differentiate on reported flash size, since this one has an STM32F103TB with real 128K. So this build now works on Blue Pill as well as HyTiny boards. It's reported as h/w "rev 2", with tweaks to deal with the differences (such as PA0 USB pull-up enable).

jcw, 2019-05-24:
Perhaps move this "boot marker" to PA11? That's one of the USB pins and it's overridden when USB is enabled, so the GPIO config bits are not really ever going to be used.

UweBonnes, 2019-07-19:
Moving the boot marker is no option for legacy reasons

closed 2020-12-09 by UweBonnes