blackmagic-debug/blackmagic #481: led_error_port & led_error_pin

jcw, 2019-05-24:
These two variables are set in `platform.c` but never used?

I'm asking, because I don't see where the LED_PORT is being initialised as an output. On my attempt to port to the HyTiny STM32F103, the LED is not blinking.

jcw, 2019-05-24:
Solved as follows in `platform.c`:

```
	gpio_set_mode(TDO_PORT, GPIO_MODE_INPUT,
			GPIO_CNF_INPUT_FLOAT, TDO_PIN);

	switch (rev) {
	case 0:
		/* LED GPIO already set in detect_rev()*/
		led_error_port = GPIOA;
		led_error_pin = GPIO8;
		adc_init();
		break;
	case 1:
		led_error_port = GPIOC;
		led_error_pin = GPIO13;
		/* Enable MCO Out on PA8*/
		RCC_CFGR &= ~(0xf << 24);
		RCC_CFGR |= (RCC_CFGR_MCO_HSE << 24);
		gpio_set_mode(GPIOA, GPIO_MODE_OUTPUT_50_MHZ,
					  GPIO_CNF_OUTPUT_ALTFN_PUSHPULL, GPIO8);
		break;
    case 2:
		led_error_port = GPIOA;
		led_error_pin = GPIO1;
		/* Enable MCO Out on PA8*/
		RCC_CFGR &= ~(0xf << 24);
		RCC_CFGR |= (RCC_CFGR_MCO_HSE << 24);
		gpio_set_mode(GPIOA, GPIO_MODE_OUTPUT_50_MHZ,
					  GPIO_CNF_OUTPUT_ALTFN_PUSHPULL, GPIO8);
		break;
	}

    gpio_set_mode(led_error_port, GPIO_MODE_OUTPUT_2_MHZ,
                  GPIO_CNF_OUTPUT_PUSHPULL, led_error_pin);
    set_idle_state(0);

	platform_srst_set_val(false);
```

Now the activity light works - on while in gdb, off while the target is running.

esden, 2021-12-15:
As far as I understand you found a workaround for you that worked. So I am closing this issue. If you think there should be some other changes made please feel free to reopen this issue.

closed 2021-12-15 by esden