blackmagic-debug/blackmagic #486: Will there ever be another stable release

jrozner, 2019-06-22:
It’s been over two years. Is there another one coming?

UweBonnes, 2019-06-24:
As there will get new devices added as they pop up, what is the advantage of a "stable" version"?

jrozner, 2019-06-25:
So it's clear what feature set is available, whether bugs are fixed without having to track down a specific commit, understanding compatibility, and whether or not the code at a specific point is stable enough to depend on? This is a pretty standard concept in software development.

sidprice, 2019-06-25:
From my perspective, the way this repository is managed, I consider the master branch to be the "stable" version. The nature of the software is that no single person has the facilities, i.e. the hardware needed, to fully validate the whole package.

As to knowing what bugs have been fixed, surely that is the use of the "Issues" TAB of the repository. If one has a problem the first place to check is the Issues list. If it has already been identified, then the Issues list will tell what the current status is.

For a community developed project this to me seems a good approach. Feature requests and bug reports are public for all to see and for all to make contributions to add and/or fix.

Thanks to @UweBonnes for his continued coordination and work on this valuable project!

esden, 2019-08-20:
Hey everyone. 

The last stable version of BMP firmware is the V1.6. You can find it in the [V1.6 branch](https://github.com/blacksphere/blackmagic/tree/v1.6-branch).

As you can imagine stabilizing a new version takes a few weeks of work. It is not like a normal software project where you run your test bench and call it good. ;)

The last stabilization process took about four weeks of me and Gareth testing piles of dev-boards with the BMP firmware. Fixing a bunch of bugs and going back and forth. This is why all Black Magic Probe still ship with the V1.6.

If you are missing features or feel like there is a bug in that stable version that causes issues for you specifically, please go ahead and use the current master. It is pretty stable to and I know all the devs put a lot of effort into keeping it working and not introducing regressions.

That said, yes a new stable version will be there. It is on my never ending todo list. I will create the milestone and branch for V1.7 soon. (before the end of the year) And we can start chipping away on testing all the supported targets. When that is done we will hopefully have another stable release. Maybe this time more people will get involved in the testing of targets and we can be even more confident that the release is working for everyone. :)

I do agree that we should be doing this way more often. At least twice a year. But as you might now understand it takes more effort than you think. :D 

I hope it explains the situation a bit.

Cheers,
Piotr

TDHolmes, 2020-05-23:
Checking in on this, is v1.7 still planned at some point? I think a new stable release would be nice, but I also understand the work required is quite high. Thanks all for an awesome product & continued updates on master.

UweBonnes, 2020-05-24:
https://github.com/UweBonnes/blackmagic/tree/multi  unifies all pc-hosted BMPs and increases  BMP/remote speed substantial by using high level commands. To allow fallback for older BMPs, I will bump up revision as 1.6.2. Early feedback is welcome. When that is done, the next step is to merge cortexm romtable. I consider making that step as 1.7

UweBonnes, 2020-12-09:
At least V1.7.2 is out

closed 2020-12-09 by UweBonnes