blackmagic-debug/blackmagic #489: [SAM3X8C] Error writing data to flash

Zagitta, 2019-07-17:
Hello,
We're using your excellent blackmagic v2.1 probes to flash and debug a SAM3X8C chip however we're running into some issues with flashing that seems related to the code size.
For example everything works fine when we upload the code build with -Os resulting in around 144Kb binary:
```
(gdb) target extended-remote \\.\COM3
Remote debugging using \\.\COM3
(gdb) monitor jtag_scan
Target voltage: 0.0V
JTAG device scan failed!
(gdb) monitor jtag_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Atmel SAM3X
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x000ac7d2 in ?? ()
(gdb) file firmware.elf
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from firmware.elf...
(gdb) info target
Symbols from "D:\code\Valhalla\build\firmware.elf".
Extended remote serial target in gdb-specific protocol:
Debugging a target over a serial line.
        While running this, GDB does not access memory from...
Local exec file:
        `D:\code\Valhalla\build\firmware.elf', file type elf32-littlearm.
        Entry point: 0x944fc
        0x00080000 - 0x000a3628 is .text
        0x20070000 - 0x20070308 is .relocate
        0x20070308 - 0x2007571c is .bss
(gdb) load
Loading section .text, size 0x23628 lma 0x80000
Loading section .relocate, size 0x308 lma 0xa3628
Start address 0x944fc, load size 145712
Transfer rate: 36 KB/sec, 971 bytes/write.
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: D:\code\Valhalla\build\firmware.elf
```
However when we build with -O0 to improve debugging the binary reaches ~260KB and fails to flash:
```
(gdb) target extended-remote \\.\COM3
Remote debugging using \\.\COM3
(gdb) monitor jtag_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Atmel SAM3X
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x0009456e in ?? ()
(gdb) file firmware.elf
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from firmware.elf...
(gdb) info target
Symbols from "D:\code\Valhalla\build_debug\firmware.elf".
Extended remote serial target in gdb-specific protocol:
Debugging a target over a serial line.
        While running this, GDB does not access memory from...
Local exec file:
        `D:\code\Valhalla\build_debug\firmware.elf', file type elf32-littlearm.
        Entry point: 0xac728
        0x00080000 - 0x000c0a04 is .text
        0x20070000 - 0x200703d8 is .relocate
        0x200703d8 - 0x200758e4 is .bss
(gdb) load
Loading section .text, size 0x40a04 lma 0x80000
[Initially some LED activity for ~8 seconds then +15 minutes hang without any LED activity]
Error writing data to flash
```

I suspect it has something to do with the dual flash bank design because our chip has 512KB and uploading seems to break above 256KB however a cursory glance of the source code reveals this is already handled if I'm not mistaken: https://github.com/blacksphere/blackmagic/blob/02b2fdb2aef56f08070504d915876bad38e61603/src/target/sam3x.c#L208
I tried updating the firmware to the daily build from 16/7/2019 however that didn't make any difference.

Any ideas what this could possibly be caused by and how to get past it?

UweBonnes, 2019-07-17:
Well, you have the diagnosis: Dual bank parts not handled. The cure: Somebody has to implement that.  The problem: Few people have that device to test the changes.

What about taking a try yourself?

B.t.w; where is the source of the daily build?

Zagitta, 2019-07-17:
Further investigation reveals the following in the debug output of the probe for a successful flash:
```
...
Flash Write 000A30C0 000003E0
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000230
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000231
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000232
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000233
Flash Write 000A34A0 00000188
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000234
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000235
Flash Write 000A3628 00000308
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000236
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000237
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000238
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x000239
*** Unsupported packet: Pf=fc440900
````
Where as the the failed flash hangs with the following output:
```
...
Flash Write 000BF8D0 000003E0
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x0003F8
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x0003F9
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x0003FA
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x0003FB
Flash Write 000BFCB0 00000350
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x0003FC
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x0003FD
sam3x_flash_cmd: base = 0x400e0a00 cmd = 0x03, arg = 0x0003FE
Flash Write 000C0000 000003E0
sam3x_flash_cmd: base = 0x400e0e00 cmd = 0x03, arg = 0x000000
```

The [SAM3X datasheet](http://ww1.microchip.com/downloads/en/devicedoc/atmel-11057-32-bit-cortex-m3-microcontroller-sam3x-sam3a_datasheet.pdf#G1.1333004) says: 

> HALF_FLASHSIZE address: 
> - 512 KB products: **0x000C0000**
> - 256 KB products: 0x000A0000
> - 128 KB products: 0x00090000

So I'm guessing it's writing to the second bank that does screw up. 
Could you guide me in the direction on how to fix this? I'm no expert in all of this but I can give it a shot at making a PR.

UweBonnes, 2019-07-17:
Tell me your environment. From the above, I see that you have windows. I guess you have a arm compiler. Do you also have "make" an "git"? For the BMP un unix it then boils down to:
 - git clone https://github.com/blacksphere/blackmagic.git
 -  cd blackmagic/
 - git submodule init
 - make
 - reflash BMP
Have a look at the code in openOCD, pyOCD and pyStlink if and how they handle these difference in the SAM3X products. Peak also at BPM STM32 code how dual banks and differences in devices are handled.
If you perhaps already have code for in-circuit programming these parts have a look there too.

The cycle 
- change code
- make
- reflash BMP
can be shortened with some LIBDTDI or STLINKV2 adapter as firmware upload is not needed. BMP running on the PC also eases debug output.

Good luck

PR #490 (2019-07-17): Fixed SAM3X8C large file flashing.
The datasheet specifies the EEFC_BASE(0) is at `0x400E0A00` and EEFC_BASE(1) is at `0x400E0C00` which means they're spaced 0x200 bytes apart rather than 0x400.
This fixes #489.
closed 2019-07-17 by UweBonnes