blackmagic-debug/blackmagic #513: CORTEXM_DWT_MASK_WORD should be (1<<2)?

braiden, 2019-09-01:
Should this line:
https://github.com/blacksphere/blackmagic/blob/e6504e149b1f974a2620ea2c3c2fc42fbdbd1f09/src/target/cortexm.h#L150
be: 
```c
#define CORTEXM_DWT_MASK_WORD		(2 << 0)
```

Source: 
http://infocenter.arm.com/help/topic/com.arm.doc.ddi0337e/ch11s05s01.html#BABCJIFD

*Mask on data address when matching against `COMP`. This is the size of the ignore mask. So, `~0<<MASK` forms the mask against the address to use. That is, `DWT` matching is performed as: `(ADDR & (~0 << MASK)) == COMP`*

For context;
I was seeing my write watch point trigger inside an interrupt when an adjacent address was written. gdb seemed to try to resume after checking that the watched expression didn't change, but that takes too long. After this change the watch triggers only when indented.

UweBonnes, 2019-09-02:
I have added the required change to #507. Feedback welcome.

UweBonnes, 2019-09-04:
Fixed by #516

closed 2019-09-04 by UweBonnes