blackmagic-debug/blackmagic #518: pc-stlinkv2: Dropped broken connection

EliasOenal, 2019-09-05:
As suggested in #319 I tried to setup pc-stlinkv2. Sadly I couldn't even manage to connect GDB successfully. Are there any known bugs with this? Do I have to downgrade my GDB version? (It works fine with a regular BMPV1.)

```
$ arm-none-eabi-gdb
GNU gdb (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.0.20190703-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote :2000
Remote debugging using :2000
Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
(gdb)
```
```
$ ./src/blackmagic_stlinkv2 
STLINKV3E serial 333433333333333009
Config 1
V3J3M2B0S0
MASS Mode
Listening on TCP: 2000
Got connection
*** Unsupported packet: vMustReplyEmpty
*** Unsupported packet: Hg0
*** Unsupported packet: qTStatus
Dropped broken connection
```

UweBonnes, 2019-09-05:
You seem to be on darwin. I tested with (Opensuse) Linux. 
Can you perhaps first test on Linux? If it works on Linux, perhaps the server code needs to get some care for darwin.

This care would probably be up to you.

UweBonnes, 2019-10-15:
Pc-Hosted BMP vatiants should now work on OSX/

UweBonnes, 2019-11-10:
Closing as no feedback was given

closed 2019-11-10 by UweBonnes

EliasOenal, 2020-02-06:
Sorry for the late reply. I have just tested this again and it works, thanks!